fritz-smh/yi-hack #2: Pas d'accès à la caméra

Wifsimster, 2016-03-22:
D'abord merci pour ce hack :)

J'ai bien suivie la procédure, la caméra se connecte bien au wifi (et je la vois sur le réseau) mais la page html n'est pas accessible  via navigateur (Chrome/IE).

La caméra enregistre bien les mouvements détectés par contre.

J'ai activé le mode debug mais rien de notable à première vue (log_http & log_rtsp sont vides).

windumasta, 2016-03-22:
Same Issue

fritz-smh, 2016-03-22:
Pouvez-vous donner le fichier test/log.txt de la carte sd ici ?
Can you join the test/log.txt file from the sd card here please ?

Merci/thanks

MrClemGit, 2016-03-22:
[log.txt](https://github.com/fritz-smh/yi-hack/files/184773/log.txt)

bizarre, ça se connecte au WIFI mais pas de mise à jour NTP pourtant l'accès internet est OK.
et donc pas d'accès au http.

windumasta, 2016-03-22:
[log.txt](https://github.com/fritz-smh/yi-hack/files/184803/log.txt)
idem problème avec le ntp
l'accès à internet ok (ping google.com)
la commande date me retourne bient : Tue Mar 22 20:35:48 GMT 2016

fritz-smh, 2016-03-22:
Mes premières impressions : on a déjà un souci de firmware : dans un cas, firmware J au lieu de M. Dans l'autre... pas d'info du firmware...

Je pense que les binaires ne sont pas compatibles avec le firmware. Je regarde et je vous dis quand j'ai poussé un truc de neuf pour retester.

windumasta, 2016-03-22:
Pour le mien normalement j'était en N il faudrait peut être que je downgrade avant ?

MrClemGit, 2016-03-22:
Dans mon hack de la version J, j'ai le binaire du serveur rtsp et web si ça peut aider
[server.tar.gz](https://github.com/fritz-smh/yi-hack/files/184983/server.tar.gz)

[rtspsvr.tar.gz](https://github.com/fritz-smh/yi-hack/files/184981/rtspsvr.tar.gz)

fritz-smh, 2016-03-22:
@windumasta pour le N, je vais voir pour trouver les binaires qui vont bien

@MrClemGit : merci, je vais les intégrer. Je devrais pousser un truc testable pour J d'ici à 30 min

@tous : j'ai l'impression que le firmware 'home' posé sur ma carte SD ne remplace pas celui d'origine... je cherche encore pourquoi... en attendant, je vais voir si ajouter et utiliser les binaires des autres firmware fera l'affaire pour vous dans un premier temps.

windumasta, 2016-03-22:
je crois que le hack est valable par un exploit dans le firmware M du coup je fais downgrad en M et refaire un essaie

fritz-smh, 2016-03-22:
Si tu arrives à faire le downgrade, je veux bien que tu me passes ton fichier home, que je teste avec...
Je viens de vérifier et j'ai 2 fichiers home différents qui viennent de 2 firmware M différents (et  2 rtsp différents...)

fritz-smh, 2016-03-22:
Nouvelle version poussée.
Rien de neuf pour N, mais pour J ça devrait être mieux.

Merci de faire vos retours avec le fichier test/log.txt (il a plus d'informations désormais)

limbogithub, 2016-03-22:
Hi everyone, is it possible a small summary in english please? I want to join the discussion ;)

MrClemGit, 2016-03-22:
Je ne vais pas pouvoir faire de test avec la J d'ici Vendredi : la cam a été remis en place

fritz-smh, 2016-03-22:
@MrClemGit : ah zut

@imbogithub : it seems that the home file in my repository is not upgrading the yi camera firmwares. So http and rtsp binaries that were executed were not starting because not compliant with the M firmware.

In the last push (a few minutes ago), I improved the startup script to put more usefull logs in the test/log.txt logfile and I also put some new binaries for I,J,K,M (rtsp) and J,M (http). The script is now able to select the appropriate binary depending on the firmware (assuming there is no surprise with firmware detection).

I will in a few minutes push another upgrade to add the hardware information in test/log.txt

MrClemGit, 2016-03-22:
bon si j'y pense je l'embarque au boulot pour faire des tests.

fritz-smh, 2016-03-22:
@MrClemGit Les tests intéressants à faire si le dernier push ne passe pas : 
M'envoyer ici le contenu du log.txt
Tester éventuellement les binaires http et rtspsrv des archives dispos ici : http://xiaoyi.querex.be/
Pour faire ça, récupère les binaires rtspsrv et server et suffixe les avec un truc clair pour que tu puisses retrouver leur source.
Uploade les via ftp dans /tmp
Connecte toi en telnet
Vérifie que le processus ne tourne pas : "ps | grep rtsp" ou "ps | grep server". Si besoin kill le (mais si tu le vois c'est que ça devrait fonctionner.... ;) )
Va dans /tmp
Lance les différents binaires chacun leur tour et voit ce que ça donne

windumasta, 2016-03-22:
Il y a un moyen de lire le log.txt depuis telnet?

fritz-smh, 2016-03-22:
bien sûr
vi /home/hd1/test/log.txt

La carte SD est montée dans /home/hd1/

MrClemGit, 2016-03-22:
Je n'ai pas fait les phases suivantes : 

How to flash another firmware

    Disconnect cam
    Remove micro SD card
    **Power on and long press reset button**
    Disconnect cam
    Put "home" firmware file on SD card
    Insert micro SD card an power on ...

peut être que ça peut être utile pour la mise à jour firmware.
Je passerai en M si vraiment ça ne fonctionne pas.

windumasta, 2016-03-22:
bon bon bon 
pour le downgrad c'est simple : http://en.miui.com/thread-165550-1-1.html

I manage to solve my problem and unbrick the camera, following the below procedure
Please try and let me know if works for you

* Turn off the camera 
* Pull out the memory card
* format the sd in FAT32
* extract and copy the firmware file to the memory card home (link for the firmware I use)
http://xiaoyi.querex.be/
* Turn on the camera and hold down the reset button for 6-8 seconds
* Turn off the camera 
* Insert the memory card
* Turn on the camera 
* Wait for about 5 minutes 
(will flash orange, then turn orange, then speak Chinese and then will flash orange) 
* Connect the camera via the camera app

ça à marché j'ai recommencé à entendre la madame qui parle trop.

Next step récupéré ton update et reconfiguré. 
C'est fait mes toujours pas d'interface web 
le message est le suivant :
Ce site est inaccessible

192.168.20.21 n'autorise pas la connexion.
ERR_CONNECTION_REFUSED

# ps | grep rtsp
11149 root      1704 S    grep rtsp
# ps | grep server
  686 root       984 S    /home/log_server
11297 root      1704 S    grep server

fritz-smh, 2016-03-22:
@windumasta 
Tu es bien passé en M... et tu as la même version que moi : 
```
Starting to log...
Hardware version informations :
    hwplatform=2
Software version informations :
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
```

Peux-tu vérifier la taille de tes fichiers ? Moi j'ai : 
```
# ls -l /home/hd1/home 
-rwxr-xr-x    1 root     root       7971912 Dec  4 20:33 /home/hd1/home
# ls -l /home/hd1/test/rtspsvrM
-rwxr-xr-x    1 root     root         39580 Dec  8 00:00 /home/hd1/test/rtspsvrM
# ls -l /home/hd1/test/http/serverM
-rwxr-xr-x    1 root     root        144912 Jan  1  1980 /home/hd1/test/http/serverM
```

Et peux-tu faire (exemple pour moi) :
```
# cd /home/hd1/test/
# ./rtspsvrM

 Set 0 UID = DF899HNEJMAFVNPPWFXT 

RTSP v2 MP77V 4pda.ru

[56f1c797][RTSP]  call get_sharemem return ok!
[56f1c798][RTSP] LwlibOpenMsgQueue: mq_open ok!
[56f1c798][RTSP]  rtspsvr_init ok! 

[56f1c798][RTSP]  getBufferSize : 16384, 16 KB
[56f1c798][RTSP]  increaseBufferTo: req size 51200, 50 KB
[56f1c798][RTSP]  getBufferSize : 102400, 100 KB
[56f1c798][RTSP]  setallSocket listen port 554 ok, sock=13 

[56f1c798][RTSP]  getBufferSize : 16384, 16 KB
[56f1c798][RTSP]  increaseBufferTo: req size 51200, 50 KB
[56f1c798][RTSP]  getBufferSize : 102400, 100 KB
[56f1c798][RTSP]  setallSocket listen port 8554 ok, sock=14 

[56f1c798][RTSP]  getBufferSize : 16384, 16 KB
[56f1c798][RTSP]  increaseBufferTo: req size 51200, 50 KB
[56f1c798][RTSP]  getBufferSize : 102400, 100 KB
[56f1c798][RTSP]  setallSocket listen port 18554 ok, sock=15 

[56f1c798][RTSP] chn 0 cmd: renice -n 1 -p 23899 ok!
[56f1c798][RTSP] chn 3 cmd: renice -n 1 -p 23899 ok!
[56f1c798][RTSP] chn 1 cmd: renice -n 1 -p 23899 ok!
```

Merci,

windumasta, 2016-03-22:
alors srverM n'esxiste pas en faite 

Test1
# ls -l /home/hd1/home
-rwxr-xr-x    1 root     root       7971758 Mar 22 15:50 /home/hd1/home
# ls -l /home/hd1/test/rtspsvrM
-rwxr-xr-x    1 root     root         39571 Mar 22 15:50 /home/hd1/test/rtspsvrM
# # ls -l /home/hd1/test/http/serverM
#

Test2
# ./rtspsvrM
Segmentation fault

Test3

# ./serverM
Segmentation fault

fritz-smh, 2016-03-22:
2 choses

1. j'ai modifié encore le script de démarrage pour mieux vérifier si les process sont up ou pas...

2. @windumasta : tu n'as pas pris les dernières modificiations vu la taille et les dates.
Fait bien un "git pull" dans le repo git et vérifie que tu as bien les derniers fichiers. Puis mets à jour ta carte SD.

windumasta, 2016-03-23:
Salut bonjour!
tout les répertoir sont créer ce coup si 
Mais les tailles ne sont pas identique mais l'heure à l'ère d'être bonne.
Par contre je n'utilise pas de git je télécharge depuis la page 

`# ls -l /home/hd1/home
-rwxr-xr-x    1 root     root       7971758 Mar 22 16:59 /home/hd1/home
# ls -l /home/hd1/test/rtspsvrM
-rwxr-xr-x    1 root     root         39571 Mar 22 16:59 /home/hd1/test/rtspsvrM
# ls -l /home/hd1/test/http/serverM
-rwxr-xr-x    1 root     root        144885 Mar 22 16:59 /home/hd1/test/http/serverM
`

et 

`
# ./rtspsvrM
Segmentation fault
# ./serverM
Segmentation fault

`

byanou69, 2016-03-23:
Hello la communauté. Merci pour ce très bon boulot !
J'ai dû upgrader mon firmware en version 1.8.5.1M_201512011815 avant d'obtenir un accrochage sur le réseau de ma cam.
Mais pb idem @windumasta, je n'ai que l'accès TELNET qui fonctionne :-(
Même résultats de commande :
./rtspsvrM
Segmentation fault
./serverM
Segmentation fault
Dommage pour l'instant ce firmware ne m'est d'aucune utilité :-( J'espère une fix, si besoin...

fritz-smh, 2016-03-23:
@byanou69 : peux-tu m'indiquer la source où tu as récupéré ton firmware ?
Je commence à me demande si le firmware M que j'ai installé n'est pas
quelque peu exotique, je vais tenter de reflasher avec un firmware plus
commun ce soir pour vérifier si j'ai le même comportement que vous et
trouver les binaires adéquats

byanou69, 2016-03-23:
@fritz-smh Oui bien sûr, ici :http://xiaoyi.querex.be/

J'ai même tenté avec le firmware 1.8.5.1N_201512212009 mais rien n'y fait...

fritz-smh, 2016-03-23:
Merci, je regarde ce soir!

byanou69, 2016-03-23:
Le log (j'ai tenté de dupliquer serverM en serverN et rtspsvrM en rtspsvrN pour gruger mais ça ne marche pas) : 

`Starting to log...
Hardware version informations :
    hwplatform=2
Software version informations :
    [sys]
    version=1.8.5.1N_201512212009
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'N'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Mar 22  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144885 Mar 22  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        144885 Mar 22  2016 /home/hd1/test/http/serverN
    -rwxr-xr-x    1 root     root        121673 Mar 22  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root        121673 Mar 22  2016 /home/hd1/test/rtspsvrJ
    -rwxr-xr-x    1 root     root         39561 Mar 22  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39571 Mar 22  2016 /home/hd1/test/rtspsvrM
    -rwxr-xr-x    1 root     root         39571 Mar 22  2016 /home/hd1/test/rtspsvrN
The blue led is currently blinking
Debug mode = yes
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...

Status for wifi configuration=0  (0 is ok)
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:24:88:A4
              inet addr:192.168.1.247  Bcast:192.168.1.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe24:88a4/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:2 errors:0 dropped:0 overruns:0 frame:0
              TX packets:2 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000
              RX bytes:284 (284.0 B)  TX bytes:200 (200.0 B)

Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
Previous datetime is Thu Jan  1 01:00:17 GMT 1970
Done
New datetime is Thu Jan  1 01:00:19 GMT 1970
Start blue led on
Start ftp server...
Check for ftp process :
 1150 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverN...
Check for http server process :
Start rtsp server : rtspsvrN...
Check for rtsp process :
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  620 root         0 SWN  [jffs2_gcd_mtd4]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  680 root      1704 S    telnetd
  683 root       984 S    /home/log_server
  795 root         0 SW   [RtmpTimerTask]
  796 root         0 SW   [RtmpMlmeTask]
  797 root         0 SW   [RtmpCmdQTask]
  798 root         0 SW   [RtmpWscTask]
  928 root         0 SW   [hidog]
  966 root      1164 S    ./peripheral
  967 root      1144 S <  ./dispatch
  968 root      1060 S    ./exnet
 1071 root      3128 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1144 root     97524 S    ./rmm
 1150 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1211 root      3360 S    ./record_event
 1212 root     14728 S    ./mp4record 60
 1213 root      1708 S    /bin/sh /home/hd1/test/check_motion.sh
 1229 root      1700 S    sleep 30
 1347 root       904 S    /home/led_ctl -boff -bon
 1350 root      1708 R    ps
#
`

P0uet, 2016-03-23:
J'ai fait les mêmes tests et je n'ai que l'accès telnet.

     ls -l /home/hd1/home
     -rwxr-xr-x    1 root     root       7971758 Mar 22  2016 /home/hd1/home
     ls -l /home/hd1/test/rtspsvrM
     -rwxr-xr-x    1 root     root         39571 Mar 22  2016 /home/hd1/test/rtspsvrM
     ls -l /home/hd1/test/http/serverM
     -rwxr-xr-x    1 root     root        144885 Mar 22  2016 /home/hd1/test/http/serverM
     ./rtspsvrM
     Segmentation fault

[log.txt](https://github.com/fritz-smh/yi-hack/files/185843/log.txt)

J'avais resert ma camera (YHS-113) avec le firmware M disponible ici: [http://www.dealabs.com/forums/le-bar/le-laboratoire-high-tech--informatique/camra-ip-xiaomi-xiaoyi-720p/11335](url) et téléchargeable ici: [http://www.dealabs.com/url/?e=OwSjeUmmteXw4ZA%2Fzw7YueLC6gOAV47Ar77sCe4Moq%2B%2B6XyY6lUpQ4HnE8sXNyQGNXXkjcZRZj0%3D&w=all&l=forum_thread](url)

windumasta, 2016-03-23:
@byanou69 
tu es en firmware "N" tu devrais dowloader!

byanou69, 2016-03-23:
@windumasta C'était un test, je suis revenu en 1.8.5.1M_201512011815

Pacit0, 2016-03-23:
J'ai pu récupérer le flux rtsp.
Toujours sur le site de dealabs.
J'ai remplacé les éxécutable rtspvrI-K-M par ceux de cette archive [http://xiaoyi.querex.be/1.8.5.1M_test-rtspfix-4.zip](url)
Toujours pas d'accès en html ni de synchronisation de l'heure...

Alpilles49, 2016-03-23:
Bonjour à tous,
Jusque là cette caméra c'était du chinois pour moi.
J'ai donc suivi la procédure proposée.
Résultat :
- La caméra est bien connectée sur mon réseau.
- Ping ok.
- Telnet ok.
- Au vu du log ci-joint le firmware est M.
malgré cela impossible d'accéder à la caméra via Safari.
Si quelqu'un pouvait m'aider à trouver le moyen d'accéder à la caméra en html ce serait chouette.
[log.txt](https://github.com/fritz-smh/yi-hack/files/186090/log.txt)

remylpat, 2016-03-23:
Salut J'ai essayé la version d'hier soir et suis avec grand interet vos evolutions.
Je me demandais l'effort que cela pourrait etre de rajouter une fonctionalité de faire une picture sur requete HTTP. En gros declencher une photo en appelant une URL. J'avais l'habitude de faire ca avec une vieille foscam et je pense que l'option MOTION est interressante mais pourrait peut etre etre completée? Qu'en dites vous ?

Dans tous les cas super boulot je retest ce soir vos versions et vous tient au courant.

./R

windumasta, 2016-03-23:
étrange ...
le dossier "home" et devenue un fichier
ls -lrt /home/hd1/
drwxr-xr-x    3 root     root         32768 Jan  1  1980 test
drwxr-xr-x    2 root     root         32768 Jan  1  1980 record_sub
-rwxr-xr-x    1 root     root       7971758 Mar 22 16:59 home
drwxr-xr-x    2 root     root         32768 Mar 22 17:10 System Volume Information
-rwxr-xr-x    1 root     root            11 Mar 23 16:00 timesec1
-rwxr-xr-x    1 root     root            11 Mar 23 16:00 timesec
drwxr-xr-x   11 root     root         32768 Mar 23 16:22 record

MrClemGit, 2016-03-23:
moi je fais faire ça par mon synology qui récupère le flux et qui me crée une image toutes les secondes pour pouvoir l'intégrer dans Domoticz. mais c'est vrai que si ça pouvait me faire tout de suite dans le Yi ça serait bien.

remylpat, 2016-03-23:
Clem je suis preneur de ton script pour faire ca au travers du syno car j'ai pas reussi;

MrClemGit, 2016-03-23:
j'utilise l'API de Surveillance 

<?
// Configuration
$user = "USER";  // Synology username with rights to Surveillance station
$pass = "PSW";  // Password of the user entered above
$ip = "IP";  // IP-Adress of your Synology-NAS
$port = "5000";  // default port of Surveillance Station
$http = "http"; // Change to https if you use a secure connection
$cameraID = "X";  // ID of the camera for taking snapshots
$cameraStream = $_GET["stream"];
// You can change the stream by adding ?stream=0/1/2 to the url. For example: www.yourdomain.com/snapshot.php?stream=2
// 0: Live stream | 1: Recording stream | 2: Mobile stream
// Default value is stream 0
if ($cameraStream == NULL) {
        $cameraStream = "2";
        }
 
// Authenticate with Synology Surveillance Station WebAPI and get our SID
$json = file_get_contents($http.'://'.$ip.':'.$port.'/webapi/auth.cgi?api=SYNO.API.Auth&method=Login&version=3&account='.$user.'&passwd='.$pass.'&session=SurveillanceStation&format=sid');
$obj = json_decode($json, true);
@$sid = $obj["data"]["sid"];
 
// Setting the correct header so the PHP file will be recognised as a JPEG file
header('Content-Type: image/jpeg');
// Read the contents of the snapshot and output it directly without putting it in memory first
readfile($http.'://'.$ip.':'.$port.'/webapi/entry.cgi?camStm='.$cameraStream.'&version=3&cameraId='.$cameraID.'&api=SYNO.SurveillanceStation.Camera&preview=false&method=GetSnapshot&_sid='.$sid);
// Log out from Surveillance Station
file_get_contents($http.'://'.$ip.':'.$port.'/webapi/auth.cgi?api=SYNO.API.Auth&method=Logout&version=3&session=SurveillanceStation&_sid='.$sid);
?>

bigmacmo, 2016-03-23:
bonsoir,
j'ai bien les enregistrements mais pas d'accès par html 
En revanche, la cam est bien reconnue sur le réseau
en pj le log

merci
[log.txt](https://github.com/fritz-smh/yi-hack/files/186805/log.txt)

fritz-smh, 2016-03-23:
Bonsoir,

J'ai modifié quelques trucs sur les binaires et le firmware (en gros j'ai repris des fichiers pour partie de http://xiaoyi.querex.be/).

J'ai testé le downgrade en L de ma caméra puis le retour en M avec les fichiers poussés sur le dépôt.
Logiquement, ça devrait être bon désormais !

Merci de vos retours

bigmacmo, 2016-03-23:
c'est tout good !!!!!
bravo et merci

fritz-smh, 2016-03-23:
yes !
J'attends une journée pour voir le retour des autres avant de clore le ticket

bigmacmo, 2016-03-23:
oui car je ne suis pas un expert et d'ailleurs les liens stream channel ne fonctionnent pas ;(

fritz-smh, 2016-03-23:
ça veut dire quoi ne focntionnent pas ?
Pour résumé, peux-tu dire ce qui fonctionne et ce qui ne fonctionne pas ?

bigmacmo, 2016-03-23:
désolé ça déconne chez moi mais c'est pas ton fichier qui pour moi est ok
J'ai accès aux différents liens rtsp, aux records etc bref à tout
merci pour ton travail, c'est cool

Pacit0, 2016-03-23:
Downgrade avec le firmware du site belge et reset avec tes fichiers.
L interface hhtp fonctionne, le stream hd fonctionne via le lien mais quand je choisis le strem sd, il ne fonctionne pas et le hd est ensuite hors d'usage.
Ca ne me dérange pas, je n'utilise que le flux hd

Merci beaucoup pour les correctifs! c'est top

bigmacmo, 2016-03-23:
Sous imperihome les flux HD et SD fonctionnent (avec environ 8 sec de décalages mais fonctionnent)

fritz-smh, 2016-03-23:
Pour les streams qui ne fonctionnent pas, ça peut être lié au fait que la caméra capte mal, j'ai rencontré ça à un moment quand je la testais dans une zone mal couverte.

fritz-smh, 2016-03-23:
Pour le décalage vidéo, j'avais la même chose en h264 (via vlc). Je pense que c'est lié au décodeur.
A ce jour je transcode le h264 en mjpeg et j'ai un flux vidéo quasi instantané en mjpeg.

Mais ceci est une autre histoire ;)

Pacit0, 2016-03-23:
J'ai bougé ma camera et cela va mieux ;-) 
Comment transcode tu en mpeg, cela m'interesserait.
Merci

fritz-smh, 2016-03-23:
A coup de ffserver et ffmjpeg. Tu devrais trouver en 2/3 minutes sur google ;)
La difficulté surtout c'est de savoir gérer l'absence de la caméra lors du streaming pour reprendre le transcodage à la volée quand elle revient et faire en sorte de ne pas rester sur le flux figé mjpeg sur l'ihm. Je détaillerais ça sur domotique-info quand on aura stabilisé 2/3 trucs sur la version de dev de Domogik (qui apportera le supoprt des cameras mjpeg). Je ne détaillerais pas plus dans ce ticket, on va se disperser sinon ;)

windumasta, 2016-03-24:
salut bonsoir ! 
Hack Validé 
Telnet connection : OK
Wifi : OK
Http server OK
rtsp ok

byanou69, 2016-03-24:
yes top merci @fritz-smh  !
Pour ma part :
Wifi : OK
Telnet : OK
HTTP : OK
RTSP : OK
LED OFF : OK

NTP : KO
Continue Recording : KO

Firmware largement exploitable, manque plus que la synchro temps et l'enregistrement en continu à activer.

Super job !

fritz-smh, 2016-03-24:
@byanou69 :
C'est quoi qui est KO précisément sur le NTP ? Le client ne se lance pas,
la caméra n'est pas à l'heure (on voit l'heure dans le flux vidéo) ?
Qu'appelle-tu le "continue recording" ?

Wifsimster, 2016-03-24:
J'étais donc en version A, j'ai dû downgrader en M pour que cela fonctionne :)

closed 2016-03-24 by Wifsimster

byanou69, 2016-03-24:
@fritz-smh 
Quand je dis le NTP ne marche pas c'est en effet l'heure indiquée sur le flux video qui est au 1970/01/01
Pardon pour ma trad de merde :-) j'entends par "continue recording" le fait qu'avec le firmware classique la cam enregistre en boucle sur la carte SD. Ici ce n'est pas actif

fritz-smh, 2016-03-24:
@byanou69
Pour l'heure, peux-tu fournir ton fichier log.txt ?
Pour le continue recording, pour moi ça n'a jamais été censé fonctionner
ainsi. Elle n'enregistre la vidéo que si elle voit un mouvement.
A moins qu'il y ait une option pour changer ça quelque part...

byanou69, 2016-03-24:
@fritz-smh 
Si c'était bien le cas chez moi, le dossier records contenait de l'enregistrement continu. Mais t'embêtes pas avec ça je passe par un serveur en amont qui s'occupe du recording en central donc pas gênant.
Pour le NTP j'ai trouvé MA coquille : j'ai une règle de contrôle parental sur ma Freebox pour bloquer l'accès au Web de mes cams (pour que ça ne parte pas je ne sais où...) du coup le NTP était bloqué.
Est-il possible de définir son propre serveur NTP dans sa config ? Je voudrais utiliser ma box ou mon serveur.
Merci @fritz-smh pour ton taf !

fritz-smh, 2016-03-24:
Oui il est possible de faire une mise à jour pour rendre paramétrable le
serveur NTP. Tu peux créer un ticket dédié à ça ?

[2 more replies]