fritz-smh/yi-hack #3: Pas d'accès web telnet ok

stepe44, 2016-03-23:
Bonjour

Sur une version chinoise avant novembre 2015 et avec version  1.8.6.1A.201602241619

Connexion Telnet ok ping Internet ok date ok accès Web ko

fritz-smh, 2016-03-23:
Je n'ai pas pu mettre les binaires du A. Peux-tu regarder en telnet si il y
a des fichiers /home/rtsp* et /home/web/server* ou /home/server* ?
Si oui, peux-tu les attacher à ce ticket ?

Merci,

stepe44, 2016-03-23:
Bonjour

Merci pour ta réponse rapide comment puis je prendre les fichiers en ftp
c'est vide

J'ai trouvé en Telnet le dossier /home/web le fichier server j'ai pas de
/home/rstp* mais surtout je sais pas comment faire pour les copier merci
d'avance
Je n'ai pas pu mettre les binaires du A. Peux-tu regarder en telnet si il y
a des fichiers /home/rtsp* et /home/web/server* ou /home/server* ?
Si oui, peux-tu les attacher à ce ticket ?

Merci,

fritz-smh, 2016-03-23:
Utilises filezilla pour te connecter en ftp
Mets root pour le login et n'importe quoi pour le mot de passe

stepe44, 2016-03-23:
bonjour,

mon fichier de log.txt
les 2 fichiers de /home/web

autre chose pour vous aider ?

Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.6.1A_201602241619
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'A'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Mar 23  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144885 Mar 23  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121673 Mar 23  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root        121673 Mar 23  2016 /home/hd1/test/rtspsvrJ
    -rwxr-xr-x    1 root     root         39561 Mar 23  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39571 Mar 23  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...

Status for wifi configuration=0  (0 is ok)
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:1E:20:54  
              inet addr:192.168.0.250  Bcast:192.168.0.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe1e:2054/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:20 errors:0 dropped:0 overruns:0 frame:0
              TX packets:1 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:4524 (4.4 KiB)  TX bytes:300 (300.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
Previous datetime is Thu Jan  1 01:00:17 GMT 1970
Done
New datetime is Thu Jan  1 01:00:22 GMT 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1182 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverA...
Check for http server process : 
Start rtsp server : rtspsvrA...
Check for rtsp process : 
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  620 root         0 SWN  [jffs2_gcd_mtd4]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  680 root      1704 S    telnetd
  683 root       984 S    /home/log_server
  795 root         0 SW   [RtmpTimerTask]
  796 root         0 SW   [RtmpMlmeTask]
  797 root         0 SW   [RtmpCmdQTask]
  798 root         0 SW   [RtmpWscTask]
  928 root         0 SW   [hidog]
  966 root      1164 S    ./peripheral
  967 root      1144 S <  ./dispatch
  968 root      1060 S    ./exnet
 1069 root      3128 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1166 root     97568 S    ./rmm
 1182 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1243 root      3396 S    ./record_event
 1244 root     22988 S    ./mp4record 60
 1245 root      1708 S    /bin/sh /home/hd1/test/check_motion.sh
 1268 root      1700 S    sleep 30
 1386 root       904 S    /home/led_ctl -boff -bon
 1392 root      1708 R    ps

stepe44, 2016-03-23:
[server.zip](https://github.com/fritz-smh/yi-hack/files/185892/server.zip)

fritz-smh, 2016-03-23:
J'ai poussé des modifications, merci de retester :)

stepe44, 2016-03-24:
Bravo le serveur web fonctionne maintenant avec ma version de caméra bravo je vais la rajouter sur ipcam viewer maintenant.

closed 2016-03-28 by fritz-smh