fritz-smh/yi-hack #4: Pas d'accès web telnet ok

stepe44, 2016-03-23:
Bonjour

Sur une version chinoise avant novembre 2015 et avec version  1.8.6.1A.201602241619

Connexion Telnet ok ping Internet ok date ok accès Web ko

Voici mon log

stepe44, 2016-03-23:
cd /mnt
# 
ls
# 
ps
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  616 root         0 SW   [mmcqd/0]
  620 root         0 SWN  [jffs2_gcd_mtd4]
  636 root       872 S <  udevd --daemon
  656 root         0 SWN  [jffs2_gcd_mtd5]
  677 root         0 SW   [flush-179:0]
  680 root      1712 S    telnetd
  683 root       984 S    /home/log_server
  795 root         0 SW   [RtmpTimerTask]
  796 root         0 SW   [RtmpMlmeTask]
  797 root         0 SW   [RtmpCmdQTask]
  798 root         0 SW   [RtmpWscTask]
  928 root         0 SW   [hidog]
  966 root      1164 S    ./peripheral
  967 root      1144 S <  ./dispatch
  968 root      1060 S    ./exnet
 1069 root      3128 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1166 root     97568 S    ./rmm
 1182 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1243 root      3396 S    ./record_event
 1244 root     23064 S    ./mp4record 60
 1245 root      1708 S    /bin/sh /home/hd1/test/check_motion.sh
 1386 root       904 S    /home/led_ctl -boff -bon
 1402 root      1720 S    -sh
 1653 root      1720 S    -sh
 3939 root      1720 S    -sh
 4233 root         0 SW   [flush-mtd-unmap]
 5986 root      1700 S    sleep 30
 6163 root      1708 R    ps

fritz-smh, 2016-03-23:
duplica du #3 . Je clos

closed 2016-03-23 by fritz-smh