fritz-smh/yi-hack #6: ntp server problem and blue state led off

musicaways, 2016-03-27:
HI,
congratulations for your work.
I have a problem with the NTP server, brings me the wrong date (1970/01/01 /), can you help me?
Also can I turn off the blue LED by default?
Thanks.

fritz-smh, 2016-03-28:
About the NTP, can you put here the file test/log.txt ?
Can you also check you did not put any firewall rule somewhere in your network to avoid cameras to speak on internet ?

About the led, in the yi-hack.cfg file, just do : 
LED_WHEN_READY=-boff -yoff

musicaways, 2016-03-28:
here it is
`Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Mar 23  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Mar 23  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Mar 23  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Mar 23  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Mar 23  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...

Status for wifi configuration=0  (0 is ok)
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:6 errors:0 dropped:0 overruns:0 frame:0
              TX packets:6 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:444 (444.0 B)  TX bytes:444 (444.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:61:3A:EF  
              inet addr:192.168.1.99  Bcast:192.168.1.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe61:3aef/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:551 errors:0 dropped:0 overruns:0 frame:0
              TX packets:3 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:136879 (133.6 KiB)  TX bytes:400 (400.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
Previous datetime is Thu Jan  1 01:00:17 GMT 1970
Done
New datetime is Thu Jan  1 01:00:18 GMT 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1152 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1188 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1235 root      2660 S <  ./rtspsvrM
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  620 root         0 SWN  [jffs2_gcd_mtd4]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  681 root      1704 S    telnetd
  683 root       984 S    /home/log_server
  795 root         0 SW   [RtmpTimerTask]
  796 root         0 SW   [RtmpMlmeTask]
  797 root         0 SW   [RtmpCmdQTask]
  798 root         0 SW   [RtmpWscTask]
  928 root         0 SW   [hidog]
  966 root      1164 S    ./peripheral
  967 root      1144 S <  ./dispatch
  968 root      1060 S    ./exnet
 1084 root      3128 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1146 root     97488 S    ./rmm
 1152 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1188 root      1408 S    ./serverM 80
 1225 root      3360 S    ./record_event
 1226 root     14728 S    ./mp4record 60
 1227 root      1708 S    /bin/sh /home/hd1/test/check_motion.sh
 1235 root     27252 S N  ./rtspsvrM
 1239 root      1700 S    sleep 30
 1377 root       904 S    /home/led_ctl -boff -bon
 1385 root      1708 R    ps`

I don't think it's a firewall issue, because I have another xiaomi ip camera modified with your hack in the same network, and it has no problem with the NTP server.
Thanks for the support.

fritz-smh, 2016-03-28:
This is strange the other one works... I just did a push with more logs, and the ability to choose a custom ntp server in the config file. Can you check with this update ?

musicaways, 2016-03-30:
I upgraded the cam with 1.8.6.1A firmware and the new file you created, and everything works perfectly!
My cam is version 2.1, and your script allows you to update the firmware to the cam 1.8.6.1A bypassing the ban imposed by xiaomi.
Thank you so much great job!
I tried to test your script even with the latest firmware 1.8.6.1.B but unfortunately the cam does not start and get stuck with the yellow LED.
One last question, do you think it is possible to add user and password to access the rtsp stream?

fritz-smh, 2016-03-30:
For rtsp login/password, I didn't check yet.

fritz-smh, 2016-04-17:
Original issue seems solved. Closing

closed 2016-04-17 by fritz-smh