fritz-smh/yi-hack #7: probleme config

paulolopes87, 2016-03-28:
bonjour 
je n'arrive pas a trouver le ip de la ANTS nul part.

j'ai flasher la camera em firmware B et en suite fait le yi-hack.
mais impossible de la trouver dans mes router. 

le led bleu rest allumer elle capture les mouvements. mais je ne la trouve pas.

merci d'avance

fritz-smh, 2016-03-28:
Bonjour,

Que dit le fichier test/log.txt de la carte SD ?

++

paulolopes87, 2016-03-28:
> voila le fichier

> 
> 
> Bonjour,
> 
> Que dit le fichier test/log.txt de la carte SD ?
> 
> ++

Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.6.1B_201603181307
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'B'
The RTSP server binary version which will be used is the 'I'
The HTTP server binary version which will be used is the 'J'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Mar 23  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Mar 23  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Mar 23  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Mar 23  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Mar 23  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Successfully initialized wpa_supplicant Line 4: failed to parse ssid 'paulo_free'. Line 4: failed to parse ssid 'paulo_free'. Line 10: Invalid PSK 'deuxeuros'. Line 10: failed to parse psk 'deuxeuros'. Line 11: failed to parse network block. Failed to read or parse configuration '/home/wpa_supplicant.conf'.
Status for wifi configuration=0  (0 is ok)
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:63:C7:05  
              inet addr:192.168.1.121  Bcast:192.168.1.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe63:c705/64 Scope:Link
              UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
              RX packets:131 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:29941 (29.2 KiB)  TX bytes:0 (0.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
Previous datetime is Thu Jan  1 01:00:16 GMT 1970
Done
New datetime is Thu Jan  1 01:00:46 GMT 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1416 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverJ...
Check for http server process : 
Start rtsp server : rtspsvrI...
Check for rtsp process : 
 1493 root      2244 S <  ./rtspsvrI
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  620 root         0 SWN  [jffs2_gcd_mtd4]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  679 root      1704 S    telnetd
  683 root       984 S    /home/log_server
  795 root         0 SW   [RtmpTimerTask]
  796 root         0 SW   [RtmpMlmeTask]
  797 root         0 SW   [RtmpCmdQTask]
  798 root         0 SW   [RtmpWscTask]
  928 root         0 SW   [hidog]
  966 root      1164 S    ./peripheral
  967 root      1144 S <  ./dispatch
  968 root      1060 S    ./exnet
 1399 root     97496 S    ./rmm
 1416 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1477 root      3360 S    ./record_event
 1478 root     14732 S    ./mp4record 60
 1479 root      1708 S    /bin/sh /home/hd1/test/check_motion.sh
 1493 root     26836 S N  ./rtspsvrI
 1496 root      1700 S    sleep 30
 1624 root       904 S    /home/led_ctl -boff -bon
 1626 root      1708 R    ps

fritz-smh, 2016-03-28:
Je vois 2 soucis : 
1. le firmware M ne semble pas être pris en compte... C'est étrange.
2. il y a une erreur de syntaxe dans ton fichier de configuration wifi, cf : 

```
Successfully initialized wpa_supplicant Line 4: failed to parse ssid 'paulo_free'. Line 4: failed to parse ssid 'paulo_free'. Line 10: Invalid PSK 'deuxeuros'. Line 10: failed to parse psk 'deuxeuros'. Line 11: failed to parse network block. Failed to read or parse configuration '/home/wpa_supplicant.conf'.
```

Le point 2 explique que tu ne trouves pas ta caméra. Pour le 1er point, il faut voir si tu peux réaliser un downgrade en version M de manière manuelle...

paulolopes87, 2016-03-28:
pour le 1er point ou je peut trouve les bonnes infos .

normalment tout est bon. par ce que les infos que je introduit sur l'app chinoise sont ceux la et ca funcionne.

le 2eme point je suis en tain de la downgrader.
mai j'ai um souci .

elle ne veut pas downgrader, et apre me dit en anglais que elle peut etre utilizé q'en chine :D 
ou je pourait trouver le firmware adequate ?

fritz-smh, 2016-03-30:
Le problème doit venir de là. Tu as la fameuse dernière version dont tout le monde parle...
Est-ce que le firmware te permet à minima d'avoir le serveur telnet ?

fritz-smh, 2016-04-03:
Pas de réponse pour le moment. Je ferme.
Ticket à réouvrir si besoin ;)

closed 2016-04-03 by fritz-smh

paulolopes87, 2016-04-03:
J'ai pas de temps pour le moment pour détaillée ça.
Je vous remercie de votre aide

Je retourne vous voir quand j'aurai décortiquer le problème.

Merci dans tous les cas

fritz-smh, 2016-04-04:
Pas de souci :)

paulolopes87, 2016-04-09:
bonjour 

apres des heures passes au tour de la cam je n'arrive toujour pas a passer l'etape de connextion.

ja fait un reset a la freebox pour configure le wifi de racine mais ça change rien .

je vous laisse le log 
[log.txt](https://github.com/fritz-smh/yi-hack/files/211338/log.txt)

je tourne en rond depuis. 

je flash firmware je la connect a l'appli jusq'ici tout marche et jarrive a la voir dans les perifs.

je fait le hack pouf . plus rien . le log toujour la meme chose fail to pars ssid zrzrzrz.
merci d'avance

fritz-smh, 2016-04-10:
Salut.

L'erreur est pourtant claire. C'est le fichier wpa_supplicant.conf qui a une erreur de syntaxe.

++

<div>-------- Message d'origine --------</div><div>De : paulolopes87 <notifications@github.com> </div><div>Date :2016/04/09  18:09  (GMT+01:00) </div><div>À : fritz-smh/yi-hack <yi-hack@noreply.github.com> </div><div>Cc : Fritz SMH <fritz.smh@gmail.com> </div><div>Objet : Re: [fritz-smh/yi-hack] probleme config  (#7) </div><div>
</div>bonjour

apres des heures passes au tour de la cam je n'arrive toujour pas a passer l'etape de connextion.

ja fait un reset a la freebox pour configure le wifi de racine mais ça change rien .

je vous laisse le log 
log.txt

je tourne en rond depuis.

je flash firmware je la connect a l'appli jusq'ici tout marche et jarrive a la voir dans les perifs.

je fait le hack pouf . plus rien . le log toujour la meme chose fail to pars ssid zrzrzrz.
merci d'avance

paulolopes87, 2016-04-10:
Mais j'ai rien modifié à part le ssid et la pass

Que pensez-vous que ça pourrait être

paulolopes87, 2016-04-17:
Bonsoir. 

Je tourne en rond avec la Cam e mon réseau à la maison. 

Je n'arrive toujours pas à passer l'erreur du Wi-Fi. 
Toujours le même erreur au même endroit 

Une idée ? 
J'ai réinitialiser le réseau de la box paramétrage d'usine et même comme ça rien 

Merci e désolé

fritz-smh, 2016-04-17:
peux-tu mettre le contenu de ton fichier wap_supplicant.conf ici (remplace le mot de passe quand même :D mais ne touche pas aux guillemets)

paulolopes87, 2016-04-18:
voila. 

je le edit avet l'editeur de texte mac standard . 

ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid=freehome
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk=DEUX2euros
}

merci

fritz-smh, 2016-04-18:
Essaie avec des guillemets autour du ssid et du mot de passe.

ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="freehome"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="DEUX2euros"
}

paulolopes87, 2016-04-18:
Je vais pas pouvoir essayer tout de suite je suis en déplacement.

Week-end prochain je me penche sur le sujet.

Merci beaucoup pour la patience et le temps