fritz-smh/yi-hack #8: Region ban still an issue?

PJanisio, 2016-03-29:
Cameras produced after 11 Nov has a region ban while using firmware newer than "L" version.
I saw that you removed cloud feature so region ban will not happen after flashing with your firmware?

fritz-smh, 2016-03-29:
Hi,

I have no idea how this region ban works... So I would say "try" and tell
us ;)

++

PJanisio, 2016-03-29:
After succesfull flash of firware, camera is connecting https://api.xiaoyi.com and checks camera version. If chinese version is found (serial numer check vs wifi settings or domain or whatever) than it shut down with chinese message that this is hardware for China mainland only :)

I already figured out to have RTSP, telnet, ftp for firmware version "L" working like a charm so im aware to check with your scripts as it overwrites some files permamently. Maybe some other brave user will check that ;-)

d3bugg3r, 2016-04-24:
I can confirm that Regional ban is still an issue, the attached "M" version only works assuming you're using the US/EU version. 

i'm trying to fiddle around with it - trying to bypass this limit by either blocking Xiaomi's sites on host level or emulating the json respond with a localhost copy, if i'll have any actual success i'll make a pull request.

PJanisio, 2016-05-04:
@d3bugg3r keeping fingers crossed.

fritz-smh, 2016-05-11:
Nobody got a solution yet?

d3bugg3r, 2016-05-12:
@fritz-smh i've tried to block communication to "home" server via firewall or host but it didn't help - i might be missing callback addresses or barking the wrong tree..  if anyone has any other idea it would be great.

myounges, 2016-05-14:
Hello,

I happened to get working the camera I bought last week without it turning off automatically because of the region ban.
I blocked the following dns:
ot.io.mi.com
api.xiaoyi.com

I have 1.8.6.1B version installed.

The way I blocked it was pointing to an internal DNS server, pretty much as a local host file.
Once I did that I was able to use the camera again, as factory defaults, next step be able to enable RTSP, telnet and ftp... Will share any success.

PJanisio, 2016-05-14:
@myounges 

Good news indeed! Please share your script when its done. And can you say a litlle more about this DNS server? Did you managed to point to your DNS server to avoid asking xiaoyi servers? If so, than you needed to launch a DNS server on your router or server?

myounges, 2016-05-14:
@pjanisio

This was a quick and dirty test to confirm how the ban works, I started off checking the traffic that the camera generates once it is turned on until I get the english message saying that the camera is only for use in china and disconnects.

I happened to have a raspberry pi in my internao network which has pi-hole installed, this helps a lot to see the traffic that is being generated by the camera. I saw several dns requests for amazon web services and as you well mentioned api.xiaoyi.com and the other one. I first blocked the api but the camera kept turning off then I blocked the second dns (not sure if both need to be blocked), since it worked i left it as it is.

When I say block I really mean when ever the camera makes a dns query to my dns server (rpi) the raspberry will resolve the ip of the raspberry instead of the orginal ip address. I suppose the camera understands this as the site is up but has nothing to report back (luckly).

That is pretty much what i did, as soon as I have some time (which I don't lately) would like to try to adapt the scripts already in place ( thanks to fritz-smh!!!!) to suit our needs with this version. With earlier firmware versions it seems the network adapter is not found therefore it doesn't register an ip. 

Any ideas are always welcome.

PJanisio, 2016-05-15:
@myounges 

Thanks for explanation. As you said luckly it is enough to check by camera that the site is up and nothing has to be returned back.

Anyway I think the best solution would be to localize file, or string in the binary where serial number with "CN" is stored (or any flag that says it is chinese version) and replace it with international serial number. Than no blocking of traffic would be needed.

d--bisset, 2016-05-23:
See https://diy.2pmc.net/solved-xiaomi-xiao-yi-ant-home-camera-can-used-china/

All credit to the original author, looks as if modifying /home/cloud did the trick

p.s. Thanks for the great firmware

bendu45, 2016-05-28:
Hi,
Someone posted a reply to solve the problem on this page:

https://diy.2pmc.net/solved-xiaomi-xiao-yi-ant-home-camera-can-used-china/

fritz-smh, 2016-06-08:
I added your last link in the README.
https://github.com/fritz-smh/yi-hack/commit/cddc53daedb81d2c1a69e5e06ed705bdd0a44e57

As I have not this model, could someone summarize what can be done and if it works fine ?

VTCop, 2016-06-13:
I have the same problem but with a Smart Camera 2 {1080p) 

Could the DNS trick work?

Richardgriff, 2016-06-26:
Hi.

I have 1.8.6.1B installed and working with rtsp telnet FTP and http. To enable i did  the following.

Enable telnet and login.
Enter the following commands

# ps | grep /home/watch_process | grep -v "grep" | awk '{print $1}' | xargs kill -9
# ps | grep /home/cloud | grep -v "grep" | awk '{print $1}' | xargs kill -9
# sed -i  's|api.xiaoyi.com/v4/ipc/check_did|api.xiaoyi.cox/v4/ipc/check_did|g' /home/cloud
# reboot

Everything working

cenarturo, 2016-07-04:
Hi guys, I just bought this camera yesterday, and i still cant connect it to my phone after a lot of hours i put into it. i've tried upgrade and downgrade firmware from 1.8.5.1 B to 1.8.5.1 N, all firmware cant connect to my wireless router and i get this Log in my sd card saying pasword error.
i've tried changing my wireless pasword (in case its too long) to 8 digit, still give the same result.
i've tried using the latest firmware i could find, 1.8.6.1 C and voila, my camera finally can connect to my wireless router, but now it give me a new problem, which is the issue you guys talking about, region banned, its says this device can only be used in china. i keep digging how to fix that issue, and someone post this telnet think which i dont really understand. if someone can help me with my problem i would really apreciate it.. sorry for my bad english since its not my native tongue

magadan, 2016-07-27:
Hello pls, someone can create a video tutorial to remove ban region ?

ptaeck, 2016-09-02:
Just quick update - xiaomi moves on - from 17CN**160622 versions there is no way how to downgrade - seems like there is some kind of HW lock. And blocking the above DNS did not help either...

sumnerboy12, 2016-09-06:
Yep - I just received a CN camera and it is not possible to downgrade. Hoping someone figures it out but it is looking like a bricked device at this stage :(.

epinapala, 2016-09-18:
@sumnerboy12  May be you can give this method a shot? - http://en.miui.com/thread-226141-1-1.html .. ignore if you have already tried.

sumnerboy12, 2016-09-18:
Yes - I had come across that thread and tried it. The problem seems to be the latest version has disabled the ability to load new firmware (either upgrade or downgrade) from the SD card. So there is no way to alter the firmware on the camera.

hermitcrabslab, 2016-09-19:
Well, unless that specific firmware is not dumped or avail for download somewhere, nobody can do/say anything then guess...
I guess is again some new hardware inside, WiFi chip or whatever, and firmware is n00b proof, i.e. it does not allow user to brick the camera flashing wrong firmware...
Disabling the flashing feature would be the dumbest thing ever to do since is the most convenient way to service the camera.

sumnerboy12, 2016-09-19:
Yep - I have spent ages trying every different method I could find online to re-flash this one and it just won't even attempt to upgrade/downgrade. I had two earlier versions of this camera which I got working fine using the techniques described so something has definitely changed. 

I managed to brick one of the originals (which is why I bought this new one) buy stupidly disabling the `rss` process thinking it was something trying to _connect back to base_. I was unable to reflash anything on that camera after I broke it either :(.

hermitcrabslab, 2016-09-19:
Have a look here http://felipe.astroza.cl/hacking-hi3518-based-ip-camera/

DevinWatson, 2016-09-22:
I bought two of these and stupidly upgraded them to whatever the latest firmware is. None of the SD card methods seem to work. So I opened one up and lo and behold there is what appears to be a serial port.

[image]

I ordered some Micro 1.25mm T-1 4-Pin JST Connectors from [eBay](http://www.ebay.com/itm/10-Sets-Mini-Micro-1-25mm-T-1-4-Pin-JST-Connector-with-Wire-US-SHIPING-M316-/401159763755?hash=item5d66fc372b:g:LSoAAOSw--1Ws1Bh) and am currently tying to see if I can get root / bootloader access. If this works I have a older un-upgraded camera I can dump the root file system from and just copy that over. To be honest though I will probably just build something without Xiaomi's craptastic software.

On a completely unrelated noted Walmart has started stocking this [Momentum Camera](https://www.walmart.com/ip/Momentum-720p-WiFi-Video-Audio-Monitoring-Camera/47445937) for $36. Might grab one of these to see how it works, or I might just buy some Nest cams and be done fucking about with Chinese cameras.

heavenf, 2016-09-22:
17CN hack 
http://miui.vn/forum/threads/huong-dan-fix-loi-yi-camera-xiaomi-s-n-17cnxxxxxx.28203/

the number behind the qrcode is important to determine way to hack or version to be flash.

claylua, 2016-09-22:
very smart! i check the code a bit. looks like a bypass but gosh, how do you know what file to change!  @heavenf thanks!

hermitcrabslab, 2016-09-23:
@DevinWatson That is the way to go...
I didn't opened mine(no need yet) but is cool they provide a connector for serial!
If the "older un-upgraded camera" is not identical in hardware I don't advise to copy that over, 99.9% it wont work.
Try to find firmware from same model, mount it as described [here](https://blog.0x1115.org/2016/06/08/extracting-firmware-xiaomi-ants-xiaoyi-smart-camera/) and use that instead to recover.
I doubt anyone can compete with this "chinese crap" in terms of price/quality, Xiaomi is the best plus is fun to hack it, we know it and they know it as well ;)

some1563, 2016-10-14:
I have Xiaomi YI 12CH-version which has "can only be used in China" problem. I tried this hack, but it doesn't seem to work. The camera gets fixed blue light (WiFi connection) for a few seconds, but then it starts to flash yellow again.

I've tried the telnet trick above and several firmwares but I haven't got past this. Excellent hack though, only if it had worked on my device!

rebitti, 2016-10-14:
Hello @some1563  i have this problem too,  i solve with a ready-to-go firmware i downloaded 

Download here: http://www27.zippyshare.com/v/nXZP3NJY/file.html

Source comments of: http://diy.2pmc.net/solved-xiaomi-xiao-yi-ant-home-camera-can-used-china/

zonexo, 2016-11-27:
Hi fritz-smh, I tried adding:

_**ps | grep /home/watch_process | grep -v "grep" | awk '{print $1}' | xargs kill -9
ps | grep /home/cloud | grep -v "grep" | awk '{print $1}' | xargs kill -9
sed -i 's|api.xiaoyi.com/v4/ipc/check_did|api.xiaoyi.cox/v4/ipc/check_did|g' /home/cloud**_

into equip_test.sh file but it can't work. Maybe there's problem with the hack, or it could be because your file only works for EU/US ver.

Did you only edit the equip_test.sh? I did a file compare and your "home" is the same as that I downloaded before, so there must be no change.

I'm interested in the cloud feature hack. Can you explain how it was done or the script used? I hope to include it into 1.8.6.1Q or some other version. Thanks!

zonexo, 2016-11-27:
Btw, I forgot to add that I managed to get Yi http server screen on my browser. But there's error when I use vlc to play thru rtsp.

maysamsh, 2017-04-02:
Is there any working hack for 18CN upgraded cam? I've tried everything with no luck.

zonexo, 2017-04-02:
Sorry, mine is the old version.

andy2301, 2017-04-04:
@maysamsh For the region ban of 1080p, I did the following and it's working:
1. set up an OpenVPN server in China (a VM needed from cloud provider)
2. in my DD-WRT router, set up OpenVPN client, and used policy-based tunnel routing (specify the 18CN 1080p's ip address) to route all outgoing traffic of the 1080p camera through the OpenVPN tunnel
3. reset camera and go through the setup/paring process. Vola!

matteo-pica, 2017-06-21:
Hello everybody,
Is this hack compatible with Xiaomi Xiao Fang or is only for Xiaomi Yi Ants?
Regards