fritz-smh/yi-hack #11: Compatible Synology ?

jeromarsan, 2016-03-30:
Tout d'abord merci pour le boulot effectué !

Petite question, est ce que c'est compatible avec un Synology ?

Merci de ta réponse :)

fritz-smh, 2016-03-30:
Salut,

Qu'appelles-tu compatible ?

++

jeromarsan, 2016-03-30:
Est-ce que avec ton firmware, on peut connecter la Xiaomi Yi à Synology Surveillance Station ?

fritz-smh, 2016-03-30:
Je ne connais pas les softs Synology...
Si ça peut lire du h264 oui. Sinon... non.

jeromarsan, 2016-03-30:
Ca peut oui :)

jeromarsan, 2016-03-30:
Mais par contre il faut être sous le firmware 1.8.5.1B.

> Connecter avec le compte admin.
> 
> Puis créer un fichier de conf en putty à cette emplacement :
> 
> /volume1/@appstore/SurveillanceStation/device_pack/camera_support/Xiaomi.conf
> 
> éditer en vi
> echap :wq pour enregistrer
> 
> Voici le contenu du fichier a coller :
> 
> ---- start copy from below -----------
> [Xiaomi*Ants]
> api = custom
> resolution_h264 = 1280x720
> default_resolution_h264 = 1280x720
> fps_h264_640x480 = 5,10,15,20,25,30
> fps_h264_1280x720 = 5,10,15,20,25,30
> default_fps_h264_1280x720 = 10
> default_image-quality = 5
> 
> h264 = rtsp
> mpeg4 = rtsp
> mjpeg = rtsp
> motion = h264
> 
> motion_param = sensitivity,threshold
> rtsp_keepalive = none
> rtsp_protocol = auto,udp,tcp
> -----stop copy before this line -------
> 
> Pour certain il faut cette ligne pour motion : motion = h264,mpeg4,mjpeg
> 
> Redémarrer le syno
> 
> Ensuite ajouter la caméra sur SS
> avec l'ip et le port 554
> ne mettez pas de mot de passe ni de login.
> ne testez pas ça ne sert à rien
> enregistrez
> 
> Faite un export de la conf n'importe ou
> 
> Aller dans l'export et éditez le fichier
> ".ExpCam"
> remplacer les lignes path et live path par :
> 
> path = '/ch0_0.h264'
> live_path = '/ch0_0.h264'
> enregistrez
> 
> Faite un import de la configuration et ENJOY :)

fritz-smh, 2016-04-03:
Je n'ai pas bien compris pourquoi seulement avec le firmware B...

On peut clore le ticket ou pas du coup ?

closed 2016-04-17 by fritz-smh