fritz-smh/yi-hack #13: Start/Stop motion detection and Start/Stop recording

jimbo72, 2016-03-31:
Bonjour Fritz,

Excellent travail, tout fonctionne comme décrit.

J'aurais aimé avoir 2 fonctions en plus:
- start/stop la détection de mouvement via http par exemple (j'ai peur de faire mal à la carte SD, donc activer cette fonction uniquement quand personne à la maison)
- start/stop l'enregistrement: simple enregistrement sans détection

Qu'en penses-tu ?

fritz-smh, 2016-03-31:
Bonjour,

Le serveur web est basique, il ne peut pas servir d'API de commande.
Si tu veux faire ceci, il faudra passer par la modification d'un fichier
via ftp ou telnet + un redémarrage du composant (voir de la caméra pour
faire simple).

Concernant l'enregistrement en continu, il parait que certaines le font, il
doit y avoir un paramétrage mais je ne l'ai pas cherché.

++

jimbo72, 2016-03-31:
Bonjour,

Je te remercie sincèrement d'avoir le temps de me répondre.

Que veux-tu dire par modif d'un fichier via ftp ou telnet ? si tu veux dire
modifier equip_test.sh ou autre oui
je comprends, sinon non !

Pour l'enregistrement en continu, tu aurais des liens à me donner ?
Tu sembles avoir voir vu des hacks similaires ?

Merci encore

fritz-smh, 2016-04-03:
Le serveur web ne permet pas de réagir à l'appel d'une url (même si l'url motion fait penser le contraire).
Si tu veux exécuter un truc à distance sur la caméra, il faut prendre le contrôle à distance via telnet.

Pour l'enregistrement en continu, je n'ai pas de lien. Regarde du côté du fichier ui.conf (dans /etc de mémoire), il y a plein de paramètres dedans.

fritz-smh, 2016-04-17:
Tu as pu trouver pour l'enregistrement en continu ?

jimbo72, 2016-04-18:
Non et toi?

jimbo72, 2016-04-19:
Seul truc que j'ai trouvé c'est de tuer les process mp4record
et record_event pour arrêter l'enregistrement sur motion.
Et de les relancer quand je passe en mode "alarme active".
Je fais ça en telnet avec un shell qui m'était inconnu: expect

Je suis preneur de ton aide pour l'enregistrement en continue. Le fichier
ui.conf m'a fait le même effet qu'une poule à qui on donne un couteau !

Merci par avance

2016-04-18 7:08 GMT+02:00 Jimmy Lalande <lalande.jimmy@gmail.com>:

jimbo72, 2016-04-19:
Et j'oubliais, je cherche également à envoyer directement les record sur
une autre machine, ou vers du cloud (genre GDrive)
(J'ai peur de vite cramer la carte SD)
=> soit via un montage NFS tout bête ou autre
As-tu des pistes ?

fritz-smh, 2016-05-11:
Je n'ai pas eu le temps de chercher pour l'enregistrement en continu.
Sinon pour le script, j'en ai poussé un ce soir dans le dossier sd/test/scripts/

ephestione, 2016-07-13:
I wanted to request the same exact thing, being able to click something on the webserver to, for example, "pause recording for [input box] minutes", so that you don't need to physically unplug the camera for the said time.
From what I understand from french (too bad the thread wasn't in english, and right now I'm too lazy to google translate it) you cannot really use the webserver to act on system files, so it's not technically feasible.