fritz-smh/yi-hack #15: pas de flux video

samueldirn, 2016-04-01:
bonjour j'ai acheté le modèle chinois et suivi la procédure 
j'arrive bien sur la page web, mais ensuite quand je click sur un lien de streaming je suis redirigé vers : rtsp://192.168.1.167:554/ch0_0.h264 et je reçois un message d'erreur sous IE

sous Chrome, rien ne se passe je ne suis redirigé nul part....

Samuel

fritz-smh, 2016-04-01:
Salut,
Essaie d'ouvrir le lien avec vlc. Le h264 ne se lit pas dans un navigateur.
++

<div>-------- Message d'origine --------</div><div>De : samueldirn <notifications@github.com> </div><div>Date :2016/04/01  21:43  (GMT+01:00) </div><div>À : fritz-smh/yi-hack <yi-hack@noreply.github.com> </div><div>Objet : [fritz-smh/yi-hack] pas de flux video (#15) </div><div>
</div>bonjour j'ai acheté le modèle chinois et suivi la procédure 
j'arrive bien sur la page web, mais ensuite quand je click sur un lien de streaming je suis redirigé vers : rtsp://192.168.1.167:554/ch0_0.h264 et je reçois un message d'erreur sous IE

sous Chrome, rien ne se passe je ne suis redirigé nul part....

Samuel

samueldirn, 2016-04-01:
Salut ok je vais essayer. Y a t'il une chance devoir du mjpeg un jour car j'ai une box domotique qui ne supporte que cela Samuel

Envoyé d’Outlook Mobile

https://github.com/fritz-smh/yi-hack/issues/15#issuecomment-204544401

fritz-smh, 2016-04-02:
Directement sur la cam non. Mais tu peux faire une conversion avec ffmpeg à la volée.

samueldirn, 2016-04-02:
je ne connais pas je vais regarder mais le principe c’est un petit serveur a installé localement qui va lire et convertir c’est ça ?

fritz-smh, 2016-04-02:
grosse modo oui
Cherche ffmjpeg stream h264 to mjpeg sur un moteur de recherche

fritz-smh, 2016-04-03:
Sur la question de fond le sujet me semble clos, je clos le ticket.
Réouvrir si besoin !

closed 2016-04-03 by fritz-smh