fritz-smh/yi-hack #17: Cannot contact the camera

SeraphicRav, 2016-04-02:
Hi !

I tried to install the firmware. I placed the files on a micro sd, edited the 2 configuration files. The camera blinks and the stabilize.

I can see the camera with nmap but not with http, ftp or ping. I have no sound.

I attached the log file and the yi-hack.cfg file.

My camera were shipped on 09/08/2015.

Thank you !

[log.txt](https://github.com/fritz-smh/yi-hack/files/201067/log.txt)
[yi-hack.cfg.txt](https://github.com/fritz-smh/yi-hack/files/201069/yi-hack.cfg.txt)

fritz-smh, 2016-04-03:
Hi,

The downgrade did not work...
Is the home file on your sd card the good one ?

If so, can you try to downgrade with the M firmware from : http://xiaoyi.querex.be/

Thanks,

fritz-smh, 2016-04-17:
Any feedback ?

SeraphicRav, 2016-04-18:
Hi !

Just tested, almost working perfectly, the time is just 2 hours late. :)

closed 2016-04-18 by SeraphicRav