fritz-smh/yi-hack #18: Cam non visible sur réseau

smosse, 2016-04-03:
hello
je pense que j'ai un soucis niveau réseau.
Dans les logs le wifi semble pourtant ok.
mais on voit bien que le stp est ko puisque pas de mac de la date
j'imagine que j'ai loupé un truc.
voici mon log.txt

`Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Mar 28  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Mar 28  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Mar 28  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Mar 28  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Mar 28  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...

Status for wifi configuration=0  (0 is ok)
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:24:B1:BA  
              inet addr:192.168.1.121  Bcast:192.168.1.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe24:b1ba/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:417 errors:0 dropped:0 overruns:0 frame:0
              TX packets:2 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:86954 (84.9 KiB)  TX bytes:400 (400.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.fr.pool.ntp.org':
Previous datetime is Thu Jan  1 02:00:49 GMT 1970
Done
New datetime is Thu Jan  1 02:01:19 GMT 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1729 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1768 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1815 root      2644 D    ./rtspsvrM
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  620 root         0 SWN  [jffs2_gcd_mtd4]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  680 root      1704 S    telnetd
  686 root       984 S    /home/log_server
  798 root         0 SW   [RtmpTimerTask]
  799 root         0 SW   [RtmpMlmeTask]
  800 root         0 SW   [RtmpCmdQTask]
  801 root         0 SW   [RtmpWscTask]
  931 root         0 SW   [hidog]
  969 root      1164 S    ./peripheral
  970 root      1144 S <  ./dispatch
  971 root      1060 S    ./exnet
 1087 root      3172 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1715 root     97480 S    ./rmm
 1729 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1768 root      1408 S    ./serverM 80
 1805 root      3360 S    ./record_event
 1806 root     14728 S    ./mp4record 60
 1807 root      1708 S    /bin/sh /home/hd1/test/check_motion.sh
 1815 root     27252 S N  ./rtspsvrM
 1819 root      1700 S    sleep 30
 1952 root       904 S    /home/led_ctl -boff -bon
 1954 root      1708 R    ps
`

fritz-smh, 2016-04-03:
Ton réseau chez toi est en 192.168.1.xxx ?

smosse, 2016-04-03:
yes je suis derniere une Livebox Orange

fritz-smh, 2016-04-03:
Même le ping ne passe pas ?

smosse, 2016-04-03:
oui rien :( en fait j'ai un doute sur le fait que le wifi se connecte vraiment car je vois rien sur ma box. pourtant le log semble montrer que c'est ok ?

fritz-smh, 2016-04-03:
Oui c'est bizarre, tu as bien relu le fichier wpa_supplicant.conf pour vérifier le paramétrage ?

smosse, 2016-04-03:
> Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...

Status for wifi configuration=0  (0 is ok)
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:24:B1:BA  
              inet addr:192.168.1.121  Bcast:192.168.1.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe24:b1ba/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:542 errors:0 dropped:0 overruns:0 frame:0
              TX packets:2 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:98534 (96.2 KiB)  TX bytes:500 (500.0 B)
    
Do network configuration 2/2 (DNS)
Done

> 
 il semble pourtant connecté d'après le log

smosse, 2016-04-03:
bon c'est étrange la box voit la MAC ADRESS de la CAM..

fritz-smh, 2016-04-03:
Etrange en effet...
Tu n'as pas quelque-chose d'autre en tant que : 
192.168.1.121
192.168.1.128 (cette ip est montée en plus par la cam au démarrage, je dois encore virer cette verrue)

smosse, 2016-04-03:
OMG !! c'était mon wifi 2,4 qui était planté depuis tout ce temps !!! mon wifi 5ghz était up mais le 2,4 planté .. bon dsl et merci !
j'ai deux CAM ; l'adresse .128 va gêner ?

fritz-smh, 2016-04-03:
A priori ça ne gêne pas, je crois qu'une autre personne en a déjà 2.
On a dit qu'on me prêterait une seconde dans quelques semaines, au pire je pourrais corriger à ce moment là.
Je ferme le ticket.

closed 2016-04-03 by fritz-smh