fritz-smh/yi-hack #20: Cant connect to wifi...

Zozo1202, 2016-04-10:
Sorry about this but I've been trying a lot of times without success. 
I tried a few wifi broadcasts, and although I'm absolutely sure of the pswd etc, I can't managed to get the camera through. 
Among possible issues, I wondered about two possibilities : 
- do the passwords have to respect the (tedious) passport rules of Yi? Not likely, but I wonder. 
- do the gateway addresses have to end with 255 ? 
In fact my Gateway addresses are always ending by 1. (as given in line 8 of the yi-hack). Not by 255.  I wonder if this could be the issue. 
In fact when I look at the log indeed in the ra0 section, I read :
 ra0       Link encap:Ethernet  HWaddr B0:D5:9D:63:E3:5C  
              inet addr:192.168.8.121  Bcast:192.168.8.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe63:e35c/64 Scope:Link
              UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
              RX packets:343 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:78315 (76.4 KiB)  TX bytes:0 (0.0 B)
Why is this broadcast at 192.168.8.255? 
(the full log is attached if needed). 
Any suggestion welcome. 

[log.txt](https://github.com/fritz-smh/yi-hack/files/211988/log.txt)

Zozo1202, 2016-04-10:
Got it... 
Sorry to have posted for that, but I might as well post the solution as it seems I was not the only one to make the same mistake. I'm convinced paulolopes87 (issues now closed) had the same issues in fact.
In brief, in the wpa_supplicant.conf file, in the ssid and the psk lines, we must keep the " (quotes!!). 
Sorry if it was obvious, it was not to me, and I wasted so many hours on this I might as well share it.

fritz-smh, 2016-04-11:
Hi,

No issue :)
I may add a warning in the doc about this in the next days.

++

paulolopes87, 2016-04-11:
I tried to keep "" quotes in the wpa_supplicant file but it doesn't work.

I tried almost everything. Reset in every step. All the firmware and always
the same error in line on log. Txt

Erros to pars ssid.

I hope someone can help us.

Thnks

fritz-smh, 2016-04-17:
Can you show the wpa_supplicant file ? Just replace the password by some xxxx

fritz-smh, 2016-05-11:
No response. Closing. Please reopen if needed :)

closed 2016-05-11 by fritz-smh