fritz-smh/yi-hack #21: Pb avec les flux RTSP

djnike, 2016-04-17:
Bonjour
merci pour le boulot et pour le partage ! :)
j'ai utilisé ton package et je suis parvenu à downgrader et passer en firmware M.
Je parviens à pinguer ma cam , et j'ai accès à la page YI HTTP Server.
Par contre lorsque je clic sur les 3 liens pour avoir accès aux flux SD HD ou audio j'ai une erreur sur la page.
J'ai testé de visualiser le flux directement avec VLC mais ça ne passe pas .. 
avez vous des idées pour orienter mes recherches ? :)
merci pour vos retours

fritz-smh, 2016-04-17:
Salut,

Peux-tu préciser quelle erreur tu as ?

Merci,

djnike, 2016-04-17:
bin dit donc réponse rapide ;)
j'avais une erreur dans l'explorateur indiquant que la page n'existait pas .. je l'ai toujours . J'ai redemarrer la cam sans la carte SD , je l'ai laissé tourner. J'ai fait un reset, et ai remis ta config via la sd. J'ai laissé reflasher la cam et là ... le flux apparait avec VLC. Maintenant je suis entrain de monter la config avec syno. En tout les cas merci pour ton partage :)

closed 2016-04-17 by fritz-smh