fritz-smh/yi-hack #22: 1.8.6.1 B FW doesn't seems to be able to downgrade

elem-bee, 2016-04-18:
I tried to downgrade using querex files and procedure, then applying the hack. My Camera is on the good IP, I can access it via FTP, but not any other access. And Firmware is still 1.8.6.1 B. Any chance to have the hack working for 6.1 FWs? 
Thx
elembee

k4r4su, 2016-04-26:
Is your camera CN or International ? The firm 1.8.6.x is bugged. you have to take a [1.8.5.1 firmware](http://www.dealabs.com/url/?e=OwSjeUmmteXw4ZA%2Fzw7YueLC6gOAV47Ar77sCe4Moq%2B%2B6XyY6lUpQ4HnE8sXNyQGNXXkjcZRZj0%3D&w=all&l=forum_thread), then unzip it on the sd card in root directory.

Insert your SD card in the camera, start it.
Click on reset button.
Wait.
Enjoy.

SkyThes, 2016-05-01:
Unfortunately I have the same issue on my CN model. Can't downgrade after my girlfriend updated it to 1.8.6.1B

fritz-smh, 2016-05-11:
No idea yet on my side

Basseytje, 2016-05-19:
You can downgrade the CN model with the following steps:

* Turn off the camera 
* Pull out the memory card
* format the sd in FAT32
* extract and copy the firmware file to the memory card home (link for the firmware I use)
https://yadi.sk/d/KCkz0xjpjmK5o
* Turn on the camera and hold down the reset button for 6-8 seconds
* Turn off the camera 
* Insert the memory card
* Turn on the camera 
* Wait for about 5 minutes 
(will flash orange, then turn orange, then speak Chinese and then will flash orange)

jszetoo, 2016-05-23:
Dear Basseytie,    I'm not sure what I did wrong , I have already try downgrade with, 1.8.5.1M & K, 
but still not yet solving the issue.
thx

Basseytje, 2016-05-23:
That was the last option. i dont know what else

fritz-smh, 2016-06-08:
Did you try to do this : https://diy.2pmc.net/solved-xiaomi-xiao-yi-ant-home-camera-can-used-china/
?

dieskim, 2016-07-21:
I am on 1.8.6.1C_201605191103 and also Unable to Downgrade via adding the the "home" file on SD as in the past - any other way to manually downgrade from telnet?

ndbinh99, 2016-08-04:
I found out a solution from http://diy.2pmc.net/solved-xiaomi-xiao-yi-ant-home-camera-can-used-china/
Just note it here for someone needs
-----------
I have found a way to add RTSP to 1.8.6.1C firmware. For that i needed FTP and telnet access to the yi camera. In my case both FTP and telnet were enabled after updating my yi camera. what i did was the following:

1 download 1.8.5.1M_test-rtspfix-4.zip (RTSP enable script) from http://xiaoyi.querex.be/
2 unpack 1.8.5.1M_test-rtspfix-4.zip to your harddisk
3 open an FTP connection to your Yi camera (username root, no pass)
4 copy rtspsvrM (from 1.8.5.1M_test-rtspfix-4.zip) to /home/rtspsvr
5 open a telnet connection to your Yi (username: root, password: 1234qwer)
6 in the telnet session type (without the “#”):
# chmod 0775 /home/rtspsvr
# chown 1001:1001 /home/rtspsvr
# reboot

wait about a minute for the Yi camera to boot and now open rtsp://<your ants ip>:554/ch0_0.h264 with VLC on your PC

elem-bee, 2016-08-04:
Thx for the help....But sadly, it's impossible to change the rtspsvr rights in my case, still 644.

notdone, 2016-10-13:
me too. fw is 1.8.6.1, also cant downgrade to 1.8.5.1. That mean no way to use this hack?
Beside, with fw 1.8.6.1, my cam still can use HTTP, RTSP, FTP, TELNET. But the index page of HTTP server only show the record files in the date / hour folders (60s per clip in "time folders" in "date folders"), dont have the link for viewing live stream channels (like this hack). So please help me to edit the index page (or server) to add the link live stream channels.
Tks

zengxinhui, 2016-11-24:
to downgrade:
1. be able to block tcp traffic from your Yi camera towards internet.
1.1 give it fixed ip if you are using dhcp, or just block the whole local subnet while downgrading. e.g.
you could do this on your internet gateway:
iptables -I forward -p tcp -s <Yi ip/LAN subnet> -j DROP
2. find 1.8.5.1K firmware, to be used for downgrade with the typical SDcard method, rename to home blah blah. link: http://xiaoyi.querex.be/instructions/
3. restore the filter
e.g. iptables -D forward 1

orangpelupa, 2017-01-07:
me too, my 1.8.6.1C wont downgrade by using the "home" file on SD card root.

andyshultz, 2017-01-30:
Anybody make any progress with this? I got this not realizing I would have to use their app. This firmware would fix it for me but I might return it for something else if I can't make it work.

ygator, 2017-01-30:
You do not want to downgrade if your camera uses a broadcom device for the wireless.  From the firmware files I have prior to 1.8.6 there is no check and install for broadcom.  To find out what your camera has do a lsusb command.  If you have an device with an id of 0a5c:bd1e or 0a5c:0bdc then you have broadcom wireless.  If so you need to have firmware 1.8.6.  The equip_test.sh file in this git also does not have a check and install for broadcom so it will not work without modification.
You can use the code I came up with for my cameras.  This should work on any firmware out so far.  Place it before the ifconfig ra0 up command.
```
bcmver="bd1e"
bcmver1="0bdc"
bcmcmd=$(lsusb|grep "0a5c"|cut -d':' -f3)
trx="$(ls /home/bcm/*.trx)"

if [ "$bcmver" = "$bcmcmd" ];then
	echo 1 > /tmp/isbcm
	#/home/bcm/nvram set /home/bcm/nvram.bin
	/home/bcm/nvram get
	mount -t usbfs none /proc/bus/usb
	/home/bcm/bcmdl -n /tmp/nvram.bin "${trx}" -C 5 
	insmod /home/bcm/bcmdhd.ko iface_name=ra0
	echo "BCM 43143"
elif [ "$bcmver1" = "$bcmcmd" ];then
	echo 1 > /tmp/isbcm
	#/home/bcm/nvram set /home/bcm/nvram.bin
	/home/bcm/nvram get
	mount -t usbfs none /proc/bus/usb
	/home/bcm/bcmdl -n /tmp/nvram.bin "${trx}" -C 5 
	insmod /home/bcm/bcmdhd.ko iface_name=ra0
	echo "BCM 43143"
else
	echo 1 > /tmp/ismtk
	insmod /home/mtprealloc7601Usta.ko
	insmod /home/mt7601Usta.ko
	echo "MTK 7601"
fi
```

AdySan, 2017-05-25:
FWIW, 1.8.6.1 B from here worked. http://xiaoyi.querex.be/firmwares/

TDPFlorida, 2017-07-06:
Any luck in figuring out how to DOWNGRADE the firmware? I have 17CN camera and it does NOT seem to update the firmware when I put any "home" files (tried most of the firmware versions) in the root of the SDCard.
Thanks

psydevascender, 2017-07-17:
@TDPFlorida unfortunately for 17CN cameras if you did update, you cannot downgrade with flash card. You need TTY interface to do it ($5 cable)

TDPFlorida, 2017-07-17:
Thanks @psydevascender. I DO have the connector but there is no way that I can solder it to the connectors. There is a connector on the motherboard but it looks like the connections (and I confirmed it online) do not match. I was afraid you are correct:(

psydevascender, 2017-07-18:
@TDPFlorida You would need Soldering tool to make direct wire to wire connection. There is neat Russian forum http://4pda.ru/forum/index.php?showtopic=638230 check the header
or this post http://4pda.ru/forum/index.php?showtopic=638230&st=12180#entry57197268 
hope google translate make it reasonable for you)
I found also video instructions here https://wn.com/unlock_xiaomi_xiaoyi_ants_17cn_q_version

TDPFlorida, 2017-07-18:
I found that hack and installed it correctly on the camera BUT then I made the mistake of upgrading the firmware:( and finally had to downgrade via serial. I have a solder-gun but the connections were just to hard so I chucked the camera. Thanks  anyway!!!