fritz-smh/yi-hack #25: Impossible de se connecter à la caméra

beetlealx, 2016-04-24:
Bonjour,

Il m'est impossible d'accéder à la caméra (ni ftp, http ou rstp). Pourtant la caméra se connecte au wifi (lumière bleue fixe) et je peux la pinger. Elle enregistre d'ailleurs dans le répertoire sur la sd.
J'ai l'impression que le firmware 1.8.6.1B n'est pas downgradable..

Voici mon fichier log 👍 
Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.6.1B_201603181307
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'B'
The RTSP server binary version which will be used is the 'I'
The HTTP server binary version which will be used is the 'J'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Mar 28  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Mar 28  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Mar 28  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Mar 28  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Mar 28  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = yes
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...

Status for wifi configuration=0  (0 is ok)
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:1D:94:6C  
              inet addr:192.168.1.121  Bcast:192.168.1.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe1d:946c/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:815 errors:0 dropped:0 overruns:0 frame:0
              TX packets:2 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:186460 (182.0 KiB)  TX bytes:200 (200.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.uk.pool.ntp.org':
PING 0.uk.pool.ntp.org (87.81.181.2): 56 data bytes
64 bytes from 87.81.181.2: seq=0 ttl=51 time=83.049 ms

--- 0.uk.pool.ntp.org ping statistics ---
1 packets transmitted, 1 packets received, 0% packet loss
round-trip min/avg/max = 83.049/83.049/83.049 ms
Previous datetime is Sun Apr 24 15:30:40 GMT 2016
Done
New datetime is Sun Apr 24 15:30:40 GMT 2016
Start blue led on
Start ftp server...
Check for ftp process : 
 1191 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverJ...
Check for http server process : 
Start rtsp server : rtspsvrI...
Check for rtsp process : 
 1268 root      2244 S <  ./rtspsvrI
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  620 root         0 SWN  [jffs2_gcd_mtd4]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  682 root      1704 S    telnetd
  686 root       984 S    /home/log_server
  798 root         0 SW   [RtmpTimerTask]
  799 root         0 SW   [RtmpMlmeTask]
  800 root         0 SW   [RtmpCmdQTask]
  801 root         0 SW   [RtmpWscTask]
  931 root         0 SW   [hidog]
  969 root      1164 S    ./peripheral
  970 root      1144 S <  ./dispatch
  971 root      1068 S    ./exnet
 1106 root      3140 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1169 root      1708 S    ntpd -q -p 0.uk.pool.ntp.org
 1185 root     97580 S    ./rmm
 1191 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1252 root      1708 S    /bin/sh /home/hd1/test/check_motion.sh
 1268 root     26836 S N  ./rtspsvrI
 1275 root      1700 S    sleep 30
 1371 root       904 S    /home/led_ctl -boff -bon
 1374 root      1708 R    ps

Quelqu'un peut éclairer ma lanterne ?
Merci d'avance.

beberr56, 2016-04-24:
bonjour,
j'ai eu le meme probleme il faut juste prendre la version "M" du firm et ca marche nickel

beetlealx, 2016-04-24:
Bonsoir,
J'ai bien mis le firmware M (c'est le fichier home livré dans le répertoire sd du hack) mais j'ai l'impression que la caméra charge le firmware de base et ne prend pas en compte la version sur la carte.

beberr56, 2016-04-24:
`Software version informations : 
 [sys]
 version=1.8.6.1B_201603181307
 [kernel]
`
ca c'est ton log .. celui fourni est bien le "B" il faut que tu mettes le M ;)

beetlealx, 2016-04-24:
J'ai bien essayé de downgrader mon firmware mais sans succès. J'ai mis sur la carte sd le fichier home correspondant au firmware M mais mon log indique toujours le firmware B. Comment faire ?

beberr56, 2016-04-25:
as tu bien fait un reset .?? je sais que j'ai pas mal galéré aussi ,si tu veux je t'envoie mon fichier "home"
ta cam c'est une version chinoise .??

beetlealx, 2016-04-25:
Que je fasse un reset ou non c'est pareil. Ma caméra est bien un modèle chinois, ça ne marche pas avec cette version ? merci :)

fritz-smh, 2016-04-25:
C'est dur à dire d'où vient le souci sans avoir la cam sous la main...
Est-ce que l'accès telnet fonctionne ?
Ca pourrait aider

beetlealx, 2016-04-25:
Bonsoir,
Oui, j'arrive à me connecter en telnet et avoir accès au linux embarqué. Que dois je vérifier pour être sûr que le bon firmware est utilisé ? Je peux peut être remplacer le fichier home sur la caméra via telnet (par la version M)

k4r4su, 2016-04-26:
J'ai eu un soucis similaire quand j'ai eu ma XiaoYi (avant de mettre le Yi-hack). J'ai upgradé avec le soft de Xiaomi en version 1.8.6.x et paf, plus aucun accès au soft.

>> Downgrade en 1.8.5 (look ici :http://www.dealabs.com/forums/le-bar/le-laboratoire-high-tech--informatique/camra-ip-xiaomi-xiaoyi-720p/11335 , partie "Accès RTSP, TELNET & HTTP et fuseau FR, compatible caméra vision de nuit et compatible frimware K et M! ")

Et ensuite tu pourras de nouveau avoir accès a la camera via l'appli Xiaomi en tout cas.
Ensuite libre à toi d'installer le Yi-hack :)

beetlealx, 2016-04-27:
Merci beaucoup pour les infos, je viens de comprendre pourquoi le downgrade ne s'opérait pas. Il est essentiel de formater la carte SD en FAT32. L'idéal étant de formater à chaque nouvel essai de flashage de firmware pour être sûr. Ca marche maintenant, le firmware B n'autorisait que l'accès telnet. Merci à tous.

beetlealx, 2016-04-27:
Pour info, une fois la mise à jour en 1.8.6, la mac adresse de la caméra est bannie au niveau des serveurs chinois. Impossible de se reconnecter avec l'application officielle même après un downgrade.

closed 2016-04-27 by beetlealx