fritz-smh/yi-hack #31: Camera not in the local network

didierlb, 2016-05-14:
Hello guys,

I am not able to see the camera in my network.
I have a recent CN version of the camera.
I enable the log 

Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Successfully initialized wpa_supplicant rfkill: Cannot open RFKILL control device Could not read interface ra0 flags: No such device WEXT: Could not set interface 'ra0' UP ra0: Failed to initialize driver interface
Status for wifi configuration=0  (0 is ok)
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    

Do you have any idea ?

Thank you in advance for your help

nectuss, 2016-05-19:
Have exactly same problem as didierlb. Asking for some help as I'm not able to see camera on my network table.
I will appreciate your help with my situation.

pionpion, 2016-05-20:
Hi, I've got the same problem, led blinking blue, then fix blue but the cam is not on my local network

nechry, 2016-05-20:
I have also exactly the same issue. The original firmware version contain the check to avoid usage outside China. me be the hardware contain also something different.

I will appreciate any help, many thank for your work

jszetoo, 2016-05-22:
Same here, got the Bluelight fixed, but nothing happen !

DanoneKiD, 2016-05-23:
Same problem here, led blue and no ping response.

altoinou, 2016-05-26:
Exactly the same problem:

> Start wifi configuration...
Successfully initialized wpa_supplicant rfkill: Cannot open RFKILL control device Could not read interface ra0 flags: No such device WEXT: Could not set interface 'ra0' UP ra0: Failed to initialize driver interface
Status for wifi configuration=0  (0 is ok)
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
Do network configuration 2/2 (DNS)
Done

bendu45, 2016-05-28:
Hi everybody,
I have the same problem too.
I think the latest hardware of the camera isn't yet compatible with another firmware than 1.8.6.1.X 
I can only connect the app with 1.8.6.1.B and have the message "this camera can only be used within china".
I tryed the test folder with 1.8.6.1.B home firmware but probably some files need to be upgraded.
So it won't work.
I tryed to deactivate IPv6 but the problem is the same.
If someone could help us, it would be great!
Have a nice day.

sfl3005, 2016-05-28:
Hi
I have the same issue with Xiaoyi night vision cam china version. Do you have any official firmware working ?

bendu45, 2016-05-29:
Hi,
the only way to make it work is to flash a russian firmware 
http://www.dealabs.com/url/?e=d0LDKXKUsr3Z6TCkdXgtSSs%2FPZVk2W3DbZ4vhcL1zkDC1EwiaVnq3ZaP2EwzNuqV44si4q4UUFaUayUN760mnNnvoydTvKskJXCLiijp6qypNgntTszEm0Rq3qwfefkUpLdq1%2BHGGsDILteu%2Bv%2F1zzFe9VcZeQc7&w=all&l=forum_post&i=589195

Good luck

fritz-smh, 2016-06-08:
Hi,

Sorry, on my side without having the good hardware model, I can't help :(

Could you tell me the hardware model in order to update the doc with it to help other users.

Thanks,

Woerd88, 2016-07-03:
I also have this issue. When i follow the insttructions the device solid blink blue at the end for 3 seconds and shutdown.

I tryed current firmware 1.8.5.1M_201512011815 and latest 1.8.6.1B_201603181307

Successfully initialized wpa_supplicant rfkill: Cannot open RFKILL control device Could not read interface ra0 flags: No such device WEXT: Could not set interface 'ra0' UP ra0: Failed to initialize driver 

Is there anything i can do to assist? Create a logfile with debug=yes?

Bacto, 2016-07-04:
Hi guys,

I have the same problem.
I've tried a lot of things but it's quite difficult without any telnet access (I have to build a serial cable).

What I can say is that @fritz-smh's script load only mt7601U wifi module.
On my camera, I have a bcm43143.

I've tried to manually load the bcm43143 module but even after, ra0 interface is not defined.

On the original firmware, in init.sh, the wifi chipset's firmware is updating by bcmdl. I've tried to update it again but bcmdl says that it can't find the chipset (but it is detected in lsusb).

So, for now, I think my wifi chipset has been flashed and bricked. I've bought a new camera to make some test (I'm waiting to receive it).

But I think that for other people who have this problem, @fritz-smh has just to add support for bcm43143 card in equip_test.sh :)

Woerd88, 2016-07-04:
@Bacto 
Thanks for writing your findings.

This explains a lot, when it first arrived i used the EU app to configure the device using a QR Code.
It gave me the message: Can only be used in China, so it worked on my wifi.

Then i used this project and all kind of firmware versions and i never got it working on my wifi again.
So i guess i also bricked it?... :(

Bacto, 2016-07-04:
That's super interesting to see that you have exactly the same issue.

I don't have the cam next to me, but I'll send you soon a script to see if we have exactly the same hardware and the same problem.

Hope to find a solution soon :)

Woerd88, 2016-07-04:
My device's serial number starts with 12CN, i think older models start with 11CN?
I'll try any script you have :), especially if it fixes my wifi connection.

Woerd88, 2016-07-04:
Wow!, i got wifi working again using the russian firmware from @bendu45 post. Thanks!
I believe i tryed this before and didn't worked, but now it does!

pionpion, 2016-07-04:
Thks @Woerd88  for your post, it's working for me too, with the russian firmware form @bendu45 . Work fine with the chinese version of the app

pionpion, 2016-07-04:
For info my cam is 12CN serial number too

arsya, 2016-07-05:
@Bacto thanks for pointing out about bcm43143. How did you know that your camera is using that module? I would like to check mine, perhaps the hardware is the same.
I am gathering more info about bcm43143 and seems like the driver for that should be brcmfmac.

Bacto, 2016-07-05:
@arsya: with lsusb in equip_test.sh
If you have a device 0a5c, that's the bcm43143 wifi chipset :)

@Woerd88 & @pionpion: I have a 12CN too but the firmware isn't working for me :(
Which app do you use to generate the QR code?

pionpion, 2016-07-05:
@Bacto I use first the european app, the QR code work, but the connection failed, then the app ask to scan the qr code at the back of camera and link to the correct chinese app, and with the chinese app, the QR code work and the connection too.  (sorry for my poor english, I'm French)

pionpion, 2016-07-05:
I ues the app on Android

Bacto, 2016-07-05:
Thanks for the informations @pionpion 
Unfortunately I think that my Wifi chipset is bricked. I'm waiting for my new camera and I'll make some test to confirm this.

Woerd88, 2016-07-05:
@Bacto
I kinda have the same story as @pionpion.
First EU app, then switched over to chinese app:
yicamera_v2.10.0.6_20151105_rele.apk

pionpion, 2016-07-05:
@Bacto ok, @Woerd88 yes v2.10.2.5_20160627.apk   for me

sngreco, 2016-07-06:
I tried the russian firmware, but it made no difference for me.

arsya, 2016-07-08:
Thanks @Bacto for the hint to check my wifi chipset. I got 0a5c:bd1e, definitely bcm43143 chipset. I am still searching for brcmfmac.ko binary file.

gohai, 2016-07-23:
I am having the same issue with a vanilla camera I recently bought. My SN starts with 12CNY.

    Start wifi configuration...
    Successfully initialized wpa_supplicant rfkill: Cannot open RFKILL control device Could not read interface ra0 flags: No such device WEXT: Could not set interface 'ra0' UP ra0: Failed to initialize driver interface

lsmod shows the same `0a5c:bd1e` (BCM43143 ) device that previous commenters are also seeing.

Any help would be greatly apprechiated.

gohai, 2016-07-27:
@fritz-smh I'd chip in EUR 10 for you to get a new model camera - could also send mine over for testing... let me know!

vkthor, 2016-08-01:
I've got the same issue. Russian firmware didn't do the trick for me :(
Bricked wifi? 
I've got another one working flawlessly.
Mine starts with 10CN... pretty old I guess? 
I had some trouble with the camera, updated the firmware, and can't connect to wifi since then, no matter what firmware I try.
Any ideas?

hermitcrabslab, 2016-08-27:
Maybe this will work?
`@@ -118,8 +118,32 @@ himm 0x20050074 0x06802424
 led -yoff -bfast
 
 
-insmod /home/mtprealloc7601Usta.ko
-insmod /home/mt7601Usta.ko
+bcmver="bd1e"
+bcmver1="0bdc"
+bcmcmd=$(lsusb|grep "0a5c"|cut -d':' -f3)
+
+if [ $bcmver = $bcmcmd ];then
+    echo 1 > /tmp/isbcm
+    #/home/bcm/nvram set /home/bcm/nvram.bin
+    /home/bcm/nvram get
+    mount -t usbfs none /proc/bus/usb
+    /home/bcm/bcmdl -n /tmp/nvram.bin /home/bcm/fw_bcmdhd_xy_r2.bin.trx -C 5 
+    insmod /home/bcm/bcmdhd.ko iface_name=ra0
+    echo "BCM 43143:bd1e"
+elif [ $bcmver1 = $bcmcmd ];then
+    echo 1 > /tmp/isbcm
+    #/home/bcm/nvram set /home/bcm/nvram.bin
+    /home/bcm/nvram get
+    mount -t usbfs none /proc/bus/usb
+    /home/bcm/bcmdl -n /tmp/nvram.bin /home/bcm/fw_bcmdhd_xy_r2.bin.trx -C 5 
+    insmod /home/bcm/bcmdhd.ko iface_name=ra0
+    echo "BCM 43143:0bdc"
+else
+    echo 1 > /tmp/ismtk
+    insmod /home/mtprealloc7601Usta.ko
+    insmod /home/mt7601Usta.ko
+    echo "MTK 7601"
+fi
 
 ifconfig ra0 up
 
`

gohai, 2016-08-27:
@hermitcrabslab 
Is firmware `1.8.6.1B_201604071154` (EU) supposed to work?

`$bcmcmd` is `bd1e` for me, but I still have the following in the log:

    Start wifi configuration...
    Successfully initialized wpa_supplicant rfkill: Cannot open RFKILL control device Could not read interface ra0 flags: No such device WEXT: Could not set interface 'ra0' UP ra0: Failed to initialize driver interface
    Status for wifi configuration=0  (0 is ok)

Thanks a lot in any way!

hermitcrabslab, 2016-08-27:
@gohai Dunno if 1B is good, I will have to download it and open it to see if it has **bcm** folder, for sure 2B(from russian forum that is floating around) has it and works I tested 2B on my 12CN camera.
What I couldn't get to work with any firmware I tried is the rtsp.

xumixu, 2016-08-28:
I found a solution for this, my camera serial number is 12CNXXXXXX (can only be used in china)

Juts follow this thread instructions, worked for me: http://en.miui.com/forum.php?mod=redirect&goto=findpost&ptid=178474&pid=5758323

Seems like the wifi driver got deleted, cause the root folder of the camera got full.

The file or folder that filled it up can be different, in my case was a .zup file, in another user case a .img, in others a folder, etc.

AdmiralFreebee, 2016-09-09:
Has anyone tried replacing the firmware in this package with the working Russian and then renaming the network device ra0 to wlan0 in the script? It seems to me that a different wifi chip might simply have a different device name.

jrotaetxe, 2016-11-04:
Negative. The problem is the device, not its name.

@Fritz.First and most important. Thank You
Second I beg you to inform the hack version is not for the "new" broadcom cameras. I have need some time to discover version less than 1.6 does not run in my cam.

Would like to encourage Fritz to make a version for the Broadcom 43143 version camera. Is quite annoying to have to ask permission to a remote chinese server to get an IP in your own network.
And perhaps a suggestion. Could the hack write the wpa_supplicant.conf file in the flash memory of the cam and leave there in RO until another hacked procedure relieves it and changes?
Is to have the possibility of start without an SD 
Thank you again and best regards

umasse, 2017-01-14:
@jrotaetxe In case you don't know, this fork has solved it for me:
https://github.com/hermitcrabslab/yi-hack
Has support for broadcom, and uses Chinese firmware 1.8.6Q.

Es el mejor que he encontrado hasta ahora. Firmware hau onena da!

pradgapaty, 2017-02-07:
i have this issue too. I find solution on 4pda.
original: [url](http://4pda.ru/forum/index.php?showtopic=638230&st=5560#entry47896966)
translate from russian...
in root file system created file with large size (why? idn)
How check this problem? Format SD card and insert to YI->try connect YI with app->after wifi connect error on SD card will create log file  log_<serial-number>_pwd_wrong.tar.gz
 - open archive and find next files:
1. in file ls1.txt after line /etc/Wireless/RT2870STA no record about file RT2870STA.dat(hi is missing)
2. record about this file (RT2870STA.dat) is in file file.txt.
3. in file df.txt we see low free space in root dir "/" (90-100kB) 
for example /dev/root 3.5M 3.4M 96.0K 97% /
If u have this points, then method corrected, but in some cases (8 good from 12) can't help.

i cant attach zip, so write file name and code
Fix:
**Step 1**
format SD card and in folder test create file equip_test.sh with code
`#!/bin/sh
# put this file into the "test" folder on SD card
log="/home/hd1/log"
mkdir -p $log
mv /info.zip ${log}/
sync
df -k >${log}/df_2.log
sync
#
sh -c "sleep 1; mv /home/hd1/test/equip_test.sh /home/hd1/test/equip_test.sh.step2; sync; reboot" &`

after reboot you find folder with name log, in this folder show file ls_rootfs.log with list files in root dir. Find large file (in article this is info.zip, as for me this is Resource.scr)

**Step 2**
format SD card and in folder test create file equip_test.sh with code
`#!/bin/sh
# put this file into the "test" folder on SD card
log="/home/hd1/log"
mkdir -p $log
mv /LARGEFILENAME ${log}/
sync
df -k >${log}/df_2.log
sync
#
sh -c "sleep 1; mv /home/hd1/test/equip_test.sh /home/hd1/test/equip_test.sh.step2; sync; reboot" &`

LARGEFILENAME replace for example info.zip

After reboot write hack to SD card and turn on cam!

If my post will help, writ about this please!
Good luck.

myounges, 2017-07-17:
Hello,

I was having the same issues as above... or at least one of them.
When I booted I was getting not able to start my wifi checking the ifconfig message it was:
WEXT: Could not set interface 'ra0' UP

Seems as mentioned above problems with the drivers.

I collected the driver files and added a part of the script @hermitcrabslab attached which was the first lines to copy the drivers to the wifi drivers folder.

if [ -f "/home/hd1/test/RT2870STA.dat" ]; then
    cp /home/hd1/test/RT2870STA.dat /etc/Wireless/RT2870STA/
    sync
fi

Long story short I have a working version of fritz-smh and I added all files to a zip so you just have to copy them to a clean SD, please add the required info in the config file and the wifi info to the wpasuplicant.

Commented out the voice messages in the equip_test.sh because they were a little useless once it is working... they look like this: /home/rmm "/home/hd1/test/voice/welcome.g726" 1

hope it helps.
Thanks to all for your help as well!!!!
[12CNyi.zip](https://github.com/fritz-smh/yi-hack/files/1153524/12CNyi.zip)