fritz-smh/yi-hack #34: Not storing any video files

Basseytje, 2016-05-19:
Hi,

The stream works perfect but the camera won't store video files on the SD card or the system via ftp.

Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'

giks, 2016-06-03:
J'ai exactement le même problème (same problem)
Date : 1970/01/01 & aucun enregistrement sinon je vois bien les flux via RTSP

giks, 2016-06-03:
Dans la log détaillé j'ai cela ; 

_____Gamma test for table 1__________
[./rmm][1/1/2:1:20:873]: modify info
[./rmm][1/1/2:1:20:874]: set Rock_COMM_ISP_SetDRC 1
[./mp4record 60][1/1/2:1:24:832]: cmd=mount -t vfat /dev/hd1 /home/hd1;
[./record_event][1/1/2:1:25:720]: g_ptr_mmap sucess
[./record_event][1/1/2:1:26:53]: g_ptr_mmap->hdSize check sucess
[./record_event][1/1/2:1:26:54]: /ipc_dispatch mq_open ok, id is 5
[./mp4record 60][1/1/2:1:26:54]: cmd=mount -t vfat /dev/hd2 /home/hd2;
[./record_event][1/1/2:1:26:430]: no newest_file_find
[./record_event][1/1/2:1:26:430]: no oldest_dir_find
[./record_event][1/1/2:1:26:431]: no record files
[./record_event][1/1/2:1:26:431]: first_search_record_file fail
[./rmm][1/1/2:4:53:619]: got a new motion start 
[./rmm][1/1/2:5:27:351]: got a new motion stop

fritz-smh, 2016-06-08:
Que donne la commande "mount" sur la camera ?
que dit la commande "df" ?

giks, 2016-06-09:
Mount : 
/dev/root on / type jffs2 (rw,relatime)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
tmpfs on /dev type tmpfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,mode=600,ptmxmode=000)
/dev/mtdblock5 on /home type jffs2 (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime,size=32768k)
/dev/hd1 on /tmp/hd1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=cp437                                                                                                                               ,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
none on /dev/mqueue type mqueue (rw,relatime)
tmpfs on /home/mmap_tmpfs type tmpfs (rw,relatime,size=512k)
tmpfs on /home/tmpfs type tmpfs (rw,relatime,size=16384k)
tmpfs on /home/jrview type tmpfs (rw,relatime,size=16384k)
/dev/hd1 on /tmp/hd1/test/http/record type vfat (rw,relatime,fmask=0022,dmask=00                                                                                                                               22,codepage=cp437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)

df :
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                 3584      2748       836  77% /
tmpfs                    17604        12     17592   0% /dev
/dev/mtdblock5            9088      8188       900  90% /home
tmpfs                    32768       144     32624   0% /tmp
/dev/hd1              30216192      9568  30206624   0% /tmp/hd1
tmpfs                      512        16       496   3% /home/mmap_tmpfs
tmpfs                    16384         0     16384   0% /home/tmpfs
tmpfs                    16384      3048     13336  19% /home/jrview
/dev/hd1              30216192      9568  30206624   0% /tmp/hd1/test/http/recor

ephestione, 2016-06-13:
When the time on my camera was off like yours, it didn't record anything... then I unblocked the internet connection from the router, the camera synced time, and it started recording.
Now the time is ok, but the videos stopped being recorded anyway, sooooo who knows?

Basseytje, 2016-06-13:
Wat internetconnection was blocked? and how did you unblock it?

ephestione, 2016-06-13:
I just blocked the camera's MAC address from the router so it couldn't connect to xiaomi servers and check hardware version vs firmware version, in your case it probably cannot reach the internet anyway, and so won't sync time with ntp servers

Basseytje, 2016-06-16:
i tried to ping to the google servers but i got a 100% package loss so the camera is not connected to the internet. how do i do that?

ephestione, 2016-06-16:
you need to make sure yi_hack.cfg is set correctly with a useable static IP inside the network, and the correct gateway IP, you have to check both things against the router settings

Basseytje, 2016-06-16:
Found the problem. mistyped the gateway. Now it's syncing and saving videos on the SD card. In my opinion the webcam is fully functional.

closed 2016-06-16 by Basseytje