fritz-smh/yi-hack #35: Pas d'accès à la caméra

Jayknightfr, 2016-05-23:
Hello,

Merci pour le boulot effectué !
Je n'ai malheureusement pas d'accès à la caméra.

Pourtant la connexion wifi semble bien se passer, elle est détectée par ma box, par contre mon gateway est en 192.168.1.1 et elle semble être en dur à la valeur 192.168.1.255, si j'en crois les logs. Ou alors c'est parce que je me suis trompé la première fois et du coup c'est gravé quelque part ?

Le serveur FTP semble attendre des connexions entrantes sur le port 0.0.0.0 
Le serveur RSTP fait état d'une "unknown operation 0x1 error"
Quand à la log HTTP elle essaie de faire un bind sur une adresse qui est déjà utilisée, or je n'ai pas de 192.168.1.128 d'utilisé.

Pas de mise à jour de l'heure non plus.

Je joints mes logs :

[log_rtsp.txt](https://github.com/fritz-smh/yi-hack/files/278167/log_rtsp.txt)
[log.txt](https://github.com/fritz-smh/yi-hack/files/278168/log.txt)
[log_ftp.txt](https://github.com/fritz-smh/yi-hack/files/278169/log_ftp.txt)
[log_http.txt](https://github.com/fritz-smh/yi-hack/files/278170/log_http.txt)

Merci pour le coup de main :)

PS : je me sentirai bête si j'ai "juste" à faire un reset de l'engin, mais j'ai tellement galéré pour la faire fonctionner par défaut avec mon smarphone que je n'ose pas ;)

fritz-smh, 2016-06-08:
L'ip de ta caméra est 192.168.1.14 

Peux-tu montrer ton fichier de configuration ? (en cachant le mot de passe)

Jayknightfr, 2016-07-05:
Hello,
Oui c'est bien l'IP que j'ai paramétré.

Voici le fichier yi-hack.cfg : 

`### Set the root password
ROOT_PASSWORD=1234qwer

### Network configuration
# Don't forget to also fill the file wpa_supplicant.conf for the wifi configuration
IP=192.168.1.14
NETMASK=255.255.255.0
GATEWAY=192.168.1.1

### Led color after full boot
# The sequence will be : 
# - orange : startup
# - blue blinking : configuration
# - blue : configuration finished, camera ready
# - ??? : color set a few seconds after the camera is ready

# values :
#    -boff -bfast : blue blinking
#    -boff -bon : blue on    
#    -boff -boff : blue off   
#    -boff -yfast : yellow blinking  
#    -boff -yon : yellow on    
#    -boff -yoff : yellow off 
LED_WHEN_READY=-boff -bon

### Timezone
TIMEZONE=GMT+2

### NTP server
NTP_SERVER=0.uk.pool.ntp.org

### Debug
# Debug mode, keep it to 'no' unless you known what you do
# Values : yes|no
DEBUG=yes

`

Jayknightfr, 2016-07-10:
Hello,

J'ai modifié le timezone dans le fichier de param et ça fonctionne à présent ... j'ai mis celui qui était dans le fichier sur github.

closed 2016-07-10 by Jayknightfr