fritz-smh/yi-hack #37: Lock down access to camera

samo-nz, 2016-05-25:
Currently anyone on the network can open the camera download files etc, each feature should be able to be locked down with a password or turned off. 

The best way to achieve this would be to have a settings link on the web-server that requires a user to log in, and all settings are managed there.

fritz-smh, 2016-06-08:
Hi,

This is indeed a good idea, but I have no idea on how I can do that. I have not the web server sources (and more over I am not familiar with cross compilation tools...)

I close the ticket but if anybody got an idea, feel free :)

closed 2016-06-08 by fritz-smh

fz883, 2016-06-12:
Simply turn off the webserver by prepending a "#" to the lines 341, 343 and 347.