fritz-smh/yi-hack #39: Installation problems

snormonster, 2016-05-28:
Good day, when I start yi camera with the SD card inserted the LED just stay orange and nothing happens. I have already configured the wifi and network configs.

Any other advice how to get this project running or how to debug this problem? Thanks in advance

kot7k, 2016-06-02:
I think you have to reset it first for it to work.
I had the same problem, just do this:
-Plug your camera without sd card
-Click and hold the reset button for about 6 seconds
-Unplug your camera
-Put sd card in your camera and plug it again

It should work that way. The thing is... i have to repeat this process every time i unplug the camera and move it to another place and i dont think thats the correct behaviour.

snormonster, 2016-06-02:
I have tried everything, I am giving up. It is not working. I have tried all ways of resetting, and flashing different firmwares. Now going back to Yi app, I can set up the camera to connect to wifi and it finds it, but I cannot even bind/view the camera.

pmorange, 2016-06-08:
Same thing here. Configured wifi and fixed IP, but stuck on orange.
I'll try to activate debug mode tonight to get a log file...

fritz-smh, 2016-06-08:
What does the log file says ?

pmorange, 2016-06-09:
It's weird because I don't have any log file, although I activated the debug mode.
My card is formatted in FAT32, and I have these files from the root level :

test/
home

Inside test/ :
http/
scripts/
check_motion.sh
equip_test.sh
rtspsvrI
rtspsvrK
rtspsvrM
wpa_supplicant.conf
yi-hack.cfg

...

It seems that even with everything seeming correctly put, the webcam is not recognizing the card, or is not reading the firmware.
I followed the proper way for installing a firmware (reset without the card with a pin, then turn off, insert the card and turn on...), but there must be another thing amiss.

Maybe the model of the camera is not compatible with your firmware. I have a Chinese model (I don't know the exact serial number but I checked it some days ago).
Before that, I had the K firmware downloaded on http://xiaoyi.querex.be/ and it worked well.

snormonster, 2016-06-09:
I concur, I enabled debug mode but there is no log file.

The card is read but I think somewhere upon execution something is crashing. It is just stuck on orange LED.

I also have a Chinese model camera with night vision, (bought from Gearbest in January). I'm not at home now but can check my serial number tonight if needed.

snormonster, 2016-06-09:
I upgraded to a newer firmware 1.8.6.1B from here: http://www27.zippyshare.com/v/nXZP3NJY/file.html

I then used China Yi Home app on Android to scan QR code to connect to wifi. However, the camera is not working because we are outside China. But it is still connected to Wifi. So then I telnet into camera - telnet is enabled on this firmware.

Camera is working again normally after applying the instructions from this website in telnet: https://diy.2pmc.net/solved-xiaomi-xiao-yi-ant-home-camera-can-used-china/

Now I try again, I take this project, I edit the two configuration files for wifi, and I copy it to a formatted SD card. Camera is stuck on orange LED again. No idea why this is happening...

kot7k, 2016-06-09:
Did you use any other previous hacks? The previous modifications around introduced some files to the camera system.
If you manage to telnet to the camera, look inside your system if there are any leftovers.
You should delete scripts about hppt telnet and ftp inside /etc/init.d/ cause this hack loads them from equip-test-sh, it doesn't modify the system.
Like i told you my camera did the same as yours, after setting everything up, if i disconnected it and power it on again it was stuck in orange just that it worked if i reset it before i plug. 
Now it's working every time, as it should.

And pmorange you said: "Maybe the model of the camera is not compatible with your firmware. I have a Chinese model (I don't know the exact serial number but I checked it some days ago).
Before that, I had the K firmware downloaded on http://xiaoyi.querex.be/ and it worked well."
As far as i know this hack is exactly for those of us who have the chinese model. I have it and it works.

pmorange, 2016-06-09:
Hi,
What you say about leftovers is interesting, because I used the hack on the
website you mention, and maybe it left some things left that now prevent
this hack from functionning.
I'll try to telnet and see if I can clean things up.
Thanks

--------------------------------------------------------------------------------------------------------------
Scan this QR code with your phone or tablet to get my contact information :

snormonster, 2016-06-09:
Me too I also installed a firmware from that website when I used the camera before trying this Yi-hack.

Could there be things left behind or changed that is not being cleaned when re-flashing a firmware?

Definitely going to look into this

snormonster, 2016-06-09:
Can you confirm what should we keep in /etc/init.d/ ?

I had directories: S00devs/ , S01udev , S80network , S88telnet , S89ftp , rcS

I deleted telnet and ftp and now (no surprise) my telnet is not working :D I'm desperate so willing to try anything

kot7k, 2016-06-09:
Well, it's clear then that for some reason the camera is not reading your sd card.
In camera system there is a file called /home/init.sh with the following lines: (they arent commented there)
#if [ -f "/home/hd1/test/equip_test.sh" ]; then
#	/home/hd1/test/equip_test.sh
#	exit
#fi
Justs means that if there is a file called test/equip_test.sh in your sd card, it should run it and then exit the init (this means that it wont do the usual camera stuff and just the stuff you have in test/equip_test.sh).
I already told you in a previous message, but try to fully reset the camera, like "for sure", like there is no doubt about it that you reset it (you should hear a chinese voice if you press the reset button long enough, when you hear it, unpug the camera, put back the sd and pulg it back on).
Also, you could try using the old equip-test-sh script that you can find in the other webpage to see if that works, but the main problem here is that you have to make the camera read your test/equip-test.sh file in your sd.

snormonster, 2016-06-09:
Thanks for your explanation, I was a bit slow but now I understand what is going on.

I must admit, I discovered something now, when I remove SD and boot camera as normal I and going into settings via the Yi app, my cam firmware is M version. This is what I used back in February with RTSP script. So I never actually managed to reset or reflash the cam (!!!) I tried something now, I added that old equip_test script for telnet and RTSP back on SD card and put SD back into camera, and it executed successfully, because now I can Telnet into the camera again.

The biggest problem therefore it seems is that I cannot get my camera to do an actual reset. (!!!) I have tried all combinations and durations of pressing the reset button. :(

pmorange, 2016-06-09:
I tried an upgrade to the 1.8.6.1B modified firmware you gave, and oddly
enough I could connect and use the camera.
This time it did not tell me that the camera could not be used outside
China as it did when I updated with the official firmware.
What I don't like is that the firmware has been modified by some Russian
guys. And god knows where my videos go now, and if I now have a back door
opened in my network :-p

If this most recent firmware proves good, it could be a nice place to start
and add some features. RTSP works and also telnet. I only miss the web
server as it tells me "403 forbidden - File is protected."

--------------------------------------------------------------------------------------------------------------
Scan this QR code with your phone or tablet to get my contact information :

kot7k, 2016-06-10:
You just have to press reset button until you hear a chinese voice saying something that is not the thing that it normally says, then unplug the camera and plug it again with the yi-hack.
If you managed to do it with the old script it means that it's working (its reading from the sd at that point). You should try to delete the scripts again from /etc/init.d (the ftp and telnet, then reset, unplug the camera, insert sd card with yi-hack, plug the camera again and wait long enough to see if it is trying to connect to your wifi (blue led blinking at least then fixed blue). If shouldnt speak chinese at this point, that means its reading from the sd, if it blinks in blue but not fixed blue it means that maybe your wifi settings are wrong or you miss some of the configuration files (but you said you only had fixed orange).

snormonster, 2016-06-10:
Hi kot7k

Thanks for your help. At this point all I can say is:

1) If I hold reset button long enough, then yes the camera "forgets" the
settings, so reset kind of works.

2) When I was on M firmware with disabled telnet, and then rebooted with
telnet script in test folder it worked, which means yes it reads the SD
card.

3) I tried deleting all the other init.d stuff like you said, then rebooted
with Yi-hack - still it was stuck on orange LED.

4) Back to the reset issue, when I reset camera, unplug power, insert SD
with stock firmware it reboots but it does NOT reflash and update the
firmware. I think my camera is semi broken because it never installs new
firmware from SD.

5) As a test to see if camera can still update firmware, I tried firmware
upgrade through the Yi app. Yes it worked! BUT now I am on latest C
firmware. Unfortunately I am outside China so the camera is useless now,
and because I still can't flash a different firmware, the camera is now
rendered useless.

kot7k, 2016-06-10:
Don´t you have access to telnet?
Cant you do this https://diy.2pmc.net/solved-xiaomi-xiao-yi-ant-home-camera-can-used-china/ (you posted this link) and get it working?
Never say its useless surely the russian folks will hack it or something.

FSKola, 2016-12-10:
Hey guys, I have a similar issue, but unfortunately what was suggested so far did not help me.
I have a european version camera and got it to "work" with this russian all-in-one hack, http://www27.zippyshare.com/v/nXZP3NJY/file.html, but there are some problems with it:

    I can not seem to pair the camera with any app
    I need to scan the QR-Code to get the camera to join my wifi on every boot
    No idea how to disable LED and annoying sounds
    After a few minutes, camera complains that "pairing is lost" (when it never was achieved) and that I should reset

fritz-smh hack sounds SO much better, but I can not even get it to boot properly. I get orange flashing and then a constant orange led, with no change. I already tried lots of flashing and rebooting, so I think there is no simple, "stupid" mistake. I suspect that some of the previous hacks I tried changed something about the camera's permanent software that is now preventing the equip_test.sh from being called, as hinted by you guys before. With that russian firmware I managed to telnet the camera (but not use ftp...) and check that there is no unnecessary file in /etc/init.d.
It was also mentioned:

"Well, it's clear then that for some reason the camera is not reading your sd card.
In camera system there is a file called /home/init.sh with the following lines:
if [ -f "/home/hd1/test/equip_test.sh" ]; then
/home/hd1/test/equip_test.sh
exit
fi
Justs means that if there is a file called test/equip_test.sh in your sd card, it should run it and then exit the init (this means that it wont do the usual camera stuff and just the stuff you have in test/equip_test.sh)."

And I think something like this could be my issue. when i tried to inspect that file with "vi init.sh" I got something completely different:

//----------------- start of paste
killall telnetd

ifconfig eth0 down

if [ -f "/home/do_format" ]; then
/home/format
fi

mkdir -p /tmp
mount tmpfs /tmp -t tmpfs -o size=32m

if [ -f "/home/RT2870STA.dat" ]; then
cp /home/RT2870STA.dat /etc/Wireless/RT2870STA/
sync
fi

rm /home/hd1 -r
rm /home/hd2 -r
mkdir -p /tmp/hd1/record
mkdir -p /tmp/hd1/record_sub
mkdir -p /tmp/hd2/record
mkdir -p /tmp/hd2/record_sub
ln -s /tmp/hd1 /home/hd1
ln -s /tmp/hd2 /home/hd2

//--------------------- end of paste

So I am a complete linux/telnet/vi/filesystem beginner, but is that the same file as mentioned above? If so, is mine wrong and can I change it to the version needed for fritz-smh hack? I have no idea how to operate this telnet vi editor and couldn't open the "home" firmware file myself.

Can anyone shed some light on this?

kasperb2, 2017-01-23:
@FSKola - were you ever able to resolve your issue? I'm having similar challenges, and wonder if it could be related to yours.

gedim21, 2017-03-27:
I faced a similar situation where the camera was stuck on constant yellow led with no logs written. The problem was that the files on the test folder had windows line endings because my git was converting them on checkout. Thus the camera's linux could not execute the equip_test.sh. 

The solution is to either convert back to unix line endings or download the yi-hack as zip instead of checking out.

snormonster, 2017-03-27:
@gedim21 I checked my original files and can confirm the line endings ended with CR LF. Thanks for the tip, will check if this solves the problem!