fritz-smh/yi-hack #40: comportement bizarre

digole51, 2016-05-29:
Hello,

Merci bcp pour ce super boulot!
J'aimerai bien pouvoir en profiter mais je rencontre quelques problèmes :-)

lorsque je démarre la CAM avec la carte SD la led passe par les états suivants
jaune fixe
bleu clignotant
bleu fixe
et puis la led s'eteint

Ci-joint mon fichier de log mais la j'avoue être un peu déconcerté...
[log.txt](https://github.com/fritz-smh/yi-hack/files/288477/log.txt)

J'essaie de la pinger mais rien n'y fait.

Si quelqu'un a une idée, je suis preneur :-)
Merci

niclasvahlin, 2016-05-29:
Hi,

I have the exact same problem. I extracted the problem from the log below:

_Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Successfully initialized wpa_supplicant rfkill: Cannot open RFKILL control device Could not read interface ra0 flags: No such device WEXT: Could not set interface 'ra0' UP ra0: Failed to initialize driver interface_

The metadata: 
_Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'_

From what I understand, it has problem either starting up ra0 (renamed?), or the driver named WEXT doesn't exist. I hope someone knows how to solve this!

Thanks in advance!

fritz-smh, 2016-06-08:
The issue is similar to this one : https://github.com/fritz-smh/yi-hack/issues/31

No idea for now on my side

xumixu, 2016-08-28:
I left the solution in #31

umasse, 2017-01-14:
This fork:
https://github.com/hermitcrabslab/yi-hack
has support for the Broadcom chipset, and upgrades to firmware 1.8.6Q.
It has made my camera work great.