fritz-smh/yi-hack #44: Mise à l'heure GMT

pmorange, 2016-06-07:
Bonjour,
Une mini question à propos de la mise à l'heure : GMT-2 veut-il dire Paris/Madrid par exemple : GMT + 2, ou bien GMT - 2... Ce qui ne correspondrait à aucun pays...
Dans les autres firmwares il y avait un calcul bizarre à faire et je n'avais pas trop compris comment m'y retrouver. Si on mettait 0 c'était l'heure principale de la Chine, et il fallait faire la différence entre les deux... 

J'habite sur le fuseau horaire Paris/Madrid justement. Si GMT-2 est bien comme par défaut, je laisse comme ça :-)

Merci d'avance.

fritz-smh, 2016-06-08:
J'ai poussé des modifications (proposées via une pull request) qui incluent le bon paramétrage (à priori) à mettre en place pour Paris avec hiver et été. 
Je n'ai pas encore testé mais ça semble pas mal.

https://github.com/fritz-smh/yi-hack/blob/master/sd/test/yi-hack.cfg#L29

pmorange, 2016-06-09:
Super ! 
Si j'arrive à installer tout ça je te dirais si c'est bon. 
En tous cas la syntaxe n'est effectivement pas si "user-friendly"