fritz-smh/yi-hack #45: accès caméra en http et de l'extérieur

Barny80, 2016-06-07:
Bonsoir,
j'ai suivi avec intérêt ce magnifique tuto , et quel travail ? Bravo et merci.
tout a fonctionné pour moi, l'installation, la config et l'utilisation sur mon réseau local en accédant à la caméra avec une adresse interne en 192 en protocole rtsp.
J'ai mis l'url rtsp://198...:554/ch0_0.h264 dans VLC et ça marche très bien.
Par contre je n'y accède pas en http avec l'adresse indiquée : http://198...:554/ch0_0.h264
Et donc, je pense que c'est à cause de cela que je n'y accède pas de l'extérieur vers mon ip fixe.
Mon routeur (srf, virgin mobile) me permet des translations de port uniquement en tcp et udp . Est-ce lié ?
En fait je ne comprends pas grand chose au réseau.
Merci de votre aide
Barny80

vosmont, 2016-06-08:
C'est une très mauvaise (mais alors très mauvaise) idée, surtout si tu n'y comprends pas grand chose.

Tout d'abord, la réponse de la caméra ne peut être lue directement dans un navigateur. Seul VLC sait lire un flux RSTP. C'est pourquoi tu as l'impression que ça ne fonctionne pas sur http://198...:554/ch0_0.h264

Ensuite, il n'y a AUCUNE protection sur la caméra. Ouvrir un port sur le routeur revient à laisser la porte de ta maison ouverte 24/24 sur une rue passante (ce que tu ne ferais pas).

Il faut absolument mettre la caméra derrière un reverse proxy + tunnel VPN.

Barny80, 2016-06-08:
Bonsoir vosmont et merci de ta réponse
cependant, je me doute bien qu'il n'y a pas de sécurité, mais ça ce sera l'étape suivante.
Je n'en suis qu'au début. J'aimerai déjà accéder à ma caméra de l'extérieur ponctuellement avant de sécuriser et de l'utiliser en mode surveillance.
Donc en l'occurence j'ai vlc sur mon téléphone.
J'accède bien à l'image avec une connexion wifi et l'ip interne.
Par contre je n'accède pas à l'ip externe avec une connexion data.
Et là j'accuse mon routeur (nb6) de me bloquer.

fritz-smh, 2016-06-08:
comme l'a dit @vosmont il te faut mettre en place un reverse proxy.

Un exemple de lien qui explique (il y en a moultes autres) : https://blog.victor-hery.com/article9/configurer-un-reverse-proxy-apache-http-https

Barny80, 2016-06-09:
merci à tous les deux.
j'ai bien noté qu'il faut mettre en place une couche de sécurité.
Mais avant je pense qu'il faut être sûr que l'on atteind bien la camera sans.
Il serait encore plus difficile pour moi d'identifier d'où vient le problème en multipliant les couches à traverser,
Si, sur mon routeur, je fais une translation du port 554 vers l'IP interne:554, ne devrais je pas accéder à ma camera à partir de vlc en mettant l'adresse rstp://ipExterne:port554/ch0_0.h264 ? à partir de mon smartphone en connexion data ?
merci d'avance

Barny80, 2016-06-14:
Bonsoir,
j'ai vu sur des forum parler de port de controle . Est-il necessaire d'ouvrir un autre port que le 554. Si oui lequel ?
Quelqu'un a-t-il réussi à faire fonctionner cette cam avec une box SFR NB6 ?
merci d'avance.
Barny80

Shuunen, 2016-10-05:
Hello @Barny80 

J'ai mis en place un accès externe hier soir et oui en effet il faut faire une redirection de port NAT :

Imaginons que ta caméra en IP fixe aie l'addresse 192.168.1.12

Alors sur ta box/routeur du dois ajouter les règles suivantes :

| Nom de la règle        | Port ext | Port int |      Cible        |
| -----------------------|----------|----------|---------------|
| Yi Cam Serveur Web |      80     |     80     | 192.168.1.12 |
| Yi Cam Flux Vidéo     |    554     |   554     | 192.168.1.12 |

Imaginons que ta box est une IP publique : 150.10.20.30

Tu pourras alors voir ton serveur web sur http://150.10.20.30

Et voir ton flux vidéo via un lecteur de flux RTSP sur rtsp://150.10.20.30:554/ch0_0.h264

Le problème ici c'est que c'est le genre de ports connus que quelqu'un de mal intentionné peut scanner, alors tu peux faire une translation de port comme par exemple :

| Nom de la règle        | Port ext | Port int |      Cible        |
| -----------------------|----------|----------|---------------|
| Yi Cam Serveur Web |   9876    |     80     | 192.168.1.12 |
| Yi Cam Flux Vidéo     |   4314    |   554     | 192.168.1.12 |

Ton serveur web sera désormais accessible sur http://150.10.20.30:9876    

Et ton flux vidéo sur : rtsp://150.10.20.30:4314/ch0_0.h264

Comme dis précédemment, ces accès restent ouvert à quiconque a l’adresse, mais au moins elle est plus difficile à trouver.

En espérant t'avoir éclairé :)

vosmont, 2016-10-05:
@Shuunen, les personnes qui cherchent des failles sur une ip ne le font pas à la main. Que le port ne soit pas un port standard ne change absolument rien.

**Ce type de tutorial ne devrait pas être diffusé.**
Encore une fois, ça revient à laisser la porte d'entrée de sa maison grande ouverte 24/24 sur une rue passante.
Personne ne fait ça.

Un test online pour voir les ports ouverts : 
https://www.grc.com/x/ne.dll?bh0bkyd2

Shuunen, 2016-10-05:
@vosmont : je trouve ta réponse quelque peu extrême et je ne suis pas d'accord avec tout :)

> les personnes qui cherchent des failles sur une ip ne le font pas à la main

C'est vrai (merci captain obvious) :) 

> Que le port ne soit pas un port standard ne change absolument rien.

Mais la plupart des logiciels de détection de faille ne scannent que les port connus et/ou sensibles qui ont un intérêt d'exploitation derrière. Dans mon exemple un port random comme le 9876 ne subira pas des assauts de milliers méchants hackers tous les jours.
Si quelqu'un a vraiment une dent contre vous alors oui il essaiera surement de scanner tous les ports, en testant pleins de protocoles (ssh, telnet, ftp, rtsp), mais encore une fois, ce ne sera pas la caméra de Mme Michu qui filme son jardin...

> Ce type de tutorial ne devrait pas être diffusé.

Pour un plan de bombe artisanale je dirais pourquoi pas, mais là je ne vois pas, le fait de savoir comment mettre en place un routage NAT c'est une chose, en connaissant les risque les gens sont responsables, ce n'est pas le tutoriel qui mettra en danger les lecteurs.

> Encore une fois, ça revient à laisser la porte d'entrée de sa maison grande ouverte 24/24 sur une rue passante. Personne ne fait ça.

Juste non. Faut arrêter de caricaturer, bien sur personne ne laisse la porte d'entrée de sa maison ouverte, mais ce n'est pas comparable. Dans notre cas de figure on a soit :
- quelqu'un qui cambriole votre maison avec les biens qu'il y a à l'intérieur, plusieurs milliers d'euros de préjudice, qui peut aussi saccager l'intérieur et mettre votre vie en danger si vous êtes à l'intérieur
- quelqu'un qui regarde votre flux vidéo

La 2e option ne ferait pas un bon film :)

Donc voilà le fond de ma pensée, je ne dit pas qu'une redirection NAT équivaut à une protection totale, c'est juste une manière de ne pas mettre un panneau "porte ouverte ici vers un flux vidéo". 

Je pense qu'il faut aussi éviter de diaboliser internet, comme si y'avait des hackers partout à l’affût, prêt à vous mater faire la cuisine sur votre propre caméra parce qu'en passant dans votre rue il ont vu votre Wifi parmi les 20 réseaux disponibles et ils se sont dit, ce wifi là on le scan, si on trouve une caméra de surveillance on aura bien gagné notre journée parce que c'est trop cool et trop rentable d'espionner ça rapporte à peu près 1000 euros par minute.

Désolé pour mes sarcasmes mais c'est tout ce que ça m'inspire ce genre de paranoïa :)

vosmont, 2016-10-05:
@Shuunen, nous ne sommes donc pas d'accord :)

> Je pense qu'il faut aussi éviter de diaboliser internet, comme si y'avait des hackers partout à l’affût

Je reste sur ma position : il y a effectivement des hackers sur le Net, qui n'ont pas besoin de faire le tour du quartier pour se brancher sur le Wifi... leurs outils sont capables d'attaquer massivement sans se déplacer.

Le risque principal n'est pas le visionnage de la vidéo prise par la caméra, mais l'utilisation de celle-ci pour pénétrer votre LAN ou la transformer en zombie (cf les dernières attaques DDOS depuis des caméras de sécurité).

Je suis surement paranoïaque. Mon réseau domestique est derrière un routeur Ubiquiti et j'ai plusieurs VLAN plus ou moins étanches en fonction des types d'utilisation pour isoler les objets connectés.
Et je t'assure que les logs sont plein de tentative de connexion venant de l'extérieur... et sur pleins de ports différents.

Chacun est libre de suivre ton tutorial, mais il me semblait quand même nécessaire de faire un rapide rappel sur le danger d'ouvrir une porte non maîtrisée dans son réseau local.

vosmont, 2016-10-05:
@Barny80, une solution pour accéder à la caméra de façon un peu plus sécurisée est de passer par un serveur Synology (ou équivalent).
Son application Surveillance Station permet d'agréger les flux video venant du réseau domestique et de les présenter sur un portail.

Le tutorial de @Shuunen pourrait alors être suivi pour exposer le Synology sur Internet. Il y a eu un sérieux travail de sécurisation sur ce serveur.

L'idéal étant quand même de monter un VPN.

Barny80, 2016-10-06:
Bonjour et merci,
j'avais laissé un peu ma cam de coté, ne sachant pas y accéder de l'extérieur.
Je vais maintenant refaire des essais, en appliquant le tuto de Shuunen dans un premier temps pour m'assurer que mon modèle de cam avec mon modèle de box est bien accessible de l'extérieur.
Si tout va bien, j'y ajouterai la couche de sécurité préconisée par vosmont.
merci encore
Barny80

Shuunen, 2016-10-06:
@vosmont : On est d'accord sur certains points :) Il y a des hackers et oui c'est vrai avec l'affaire des caméras de sécurité on voit comment nos Yi Cam pourraient être exploitées. Après je pense que c'est à chacun de se documenter pour savoir comment fonctionne les choses, connaitre leur risques et quelles solutions sont disponibles pour limiter ces risques. Je trouve aussi que la décision finale revient à chacun de sécuriser ou non son réseau.

Perso pour une sécurité moyenne, j'aurais bien aimé avoir un login/pass sur le flux RTSP mais je sais pas comment ça pourrait être mis en place sur yi-hack.

Pour une sécurité forte, c'est clair que le VPN est l'idéal ici.

Barny80, 2016-10-06:
Bonsoir,
Jusque là je n'avais pas ouvert le port 80, pour un accès http.
j'ai donc paramétré mon NAT comme Shuunen le proposait 👍 
# 	Nom 	        Protocole 	Type 	Ports externes 	Adresse IP de destination 	Ports de destination
1 	antsServWeb 	TCP 	                        Port 	80 	       192.168.1.38 	                80 	
2 	antsFluxVid 	les deux 	                Port 	554 	       192.168.1.38 	                554
Ensuite avec mon smartphone,
  une connexion en wifi sur l'adresse interne fonctionne très bien (http et rtsp)
  par contre une connexion en data sur l'adresse externe ne donne rien (ni http, ni rtsp)
Le problème subsiste donc. 
Peut être un pb de box alors ?

hermitcrabslab, 2016-10-08:
Hi and sorry for interfere here..
@vosmont I think is best if you can explain, **in detail**, how you did your home network behind a second router and also/especially how you connected your camera(s) in a secure way.
That way users can learn how to do it and everyone can benefit of your knowledge, me included.
Thanks.