fritz-smh/yi-hack #47: IR illuminator doesn't seem to turn on

tsangc, 2016-06-09:
First, thank you for all of your work on this project!  This distribution of hacks seems to be the best combination of all the changes out there for the camera.

My problem is that the IR illuminator has stopped turning on.  I can enable and disable the IR filter via sendMQ mask commands and can verify as I can hear the filter clicking in and out:

http://en.miui.com/thread-135585-1-1.html

However, I can't seem to turn the IR illuminator LEDs manually using those commands and they don't come on anymore when it is dark.  I know they used to work when I ran the original firmware from the manufacturer, as I verified these using another camera pointed at the front of the Yi camera.

Is there a sequence I need to set them up with?  (eg leave them off before I reboot the camera) Or is there something I can do to help debug this issue?

fritz-smh, 2016-06-09:
Hi,

On my side, the Ir works fine with the hack.
Could you try be removing the sdcard and reuser the yi smartphone
application to test ?
Maybe your Ir stuff is broken on your camera ?

++

tsangc, 2016-06-10:
I think I've managed to get blocked as I'm using a CN version outside of China and I had loaded a modified firmware, so I can't seem to get to the original firmware.  I'll keep trying, I just need to find an unmodified firmware to test again with.   Will report back.

tsangc, 2016-06-12:
After disassembling my camera, I think my illuminator has a burnt out LED.  Will report back once I can verify that.  It's likely a hardware problem as you mentioned.

tsangc, 2016-06-14:
Turns out it was a burnt LED.  One was slightly discoloured and when measured with a multimeter in diode mode, read open.  The others seem to read around 0.780v or around there.  I found an old remote control, broken it open, and checked that it's voltage drop was also around 0.780.  I desoldered the burnt one from the YI camera and replaced it with the salvaged one from the remote.

I confirmed their operation with the sendMQ mask command and using a smartphone camera pointed at the front of the YI camera.

A couple of notes:  
a) I'm not sure what burnt out the LED, but it was the closest to some circuitry which I assume steps up 5V USB voltage to drive the whole ring of lights.

b) There are eight LEDs and they are in series.  So if one burns out, the whole ring would go out.

Anyways, thanks again for all of your work on this hack package, it's great!

closed 2016-06-14 by tsangc