fritz-smh/yi-hack #49: Timestamp

jeremielorente, 2016-06-12:
Hi,

is there any way to add a parameter to disable timestamp?

thx :)

smokingcraterr, 2016-07-20:
I'd be happy with ANY way of disabling it!  Searched all the files, /etc/ui.conf has a bunch of osd=1 and osd_time=1.  Changed them all to 0's and no affect.  I'm using an upside down camera and rotating in the DVR software, so the timestamp is really annoying.

tuaminx, 2016-08-24:
Camera and its body are different parts. Camera can be rotaled on its body.
In my case, I jsut upside down camera's body 😄

dmolner, 2016-10-16:
This is great.

And I'm very interested in disable timestamp.

Best regards.

sirken, 2017-01-17:
See #87 for partial solution to disable timestamp.

@smokingcraterr, instead of using DVR software to rotate image, edit /etc/ui.conf and change two lines:
ptz_hflip=1      
ptz_vflip=1

cheahkhing, 2017-06-27:
Hi, anyone know about how to rotate the camera captured then? just like how you can toggle the rotation in YiHome app?