fritz-smh/yi-hack #52: Timezone off 2hrs

ephestione, 2016-06-17:
I am on Rome time, which is pretty much the same of Paris; using the given string for Paris, which is the exact same for Rome when checking the referral website from yi_hack.cfg, I always get my real time -2hrs. Tried different ntp servers, from two different places.
I appear to not be able to get the time right, I might just put there the timezone of a city normally two hours ahead of Rome, but that would be a quite ugly workaround :)

ephestione, 2016-06-18:
I see equip_test echoes the timezone string into /etc/TZ, well in my case that file already contains the timezone string of GMT0, an after that is added the timezone string I configured in yi_hack.cfg

kot7k, 2016-06-18:
In the webpage provided it says for rome is this Europe/Rome CET-1CEST,M3.5.0,M10.5.0/3
The example might be wrong then

ephestione, 2016-06-20:
Alas, that is the same I currently have, Rome and Paris have the same timezone :)

wolfse, 2016-06-21:
I have the same problem here in Stockholm, however that's also the same time zone as Paris and Rome. Europe/Stockholm CET-1CEST,M3.5.0,M10.5.0/3

ephestione, 2016-06-21:
So either there's something wrong with that string, or the firmware's implementation is buggy... I don't know what those parameters do, otherwise I would try changing them.
For example I tried "CET+1blahblah" instead of CET-1, to no avail.

xyannis, 2016-08-16:
I solved this issue. Instead of coping the line from http://svn.fonosfera.org/blabla, go to http://www.remotemonitoringsystems.ca/time-zone-abbreviations.php

With the fonosfera info time was 2 hours wrong.

Example: 
Wrong Time:
TIMEZONE=EET-2EEST,M3.5.0/3,M10.5.0/4

Correct time:
TIMEZONE=EET-2EEST-3,M3.5.0/03:00:00,M10.5.0/04:00:00

ephestione, 2018-08-01:
For the record, that doesn't solve my problem anyway :(