fritz-smh/yi-hack #53: Motion Detection

Edylc, 2016-06-20:
Hi,

After installing the hack, the motion detection and subsequent file creation seemed to work at first, but later the camera stopped creating video logs for motion detection despite there being movement in front of it.  It acted as if no motion was detected.  One possibility is that we were working on the FTP connection, but we'd prefer if both worked.  Do you have any advice on what could be causing the motion detection issue?

Thanks.

Edylc, 2016-06-20:
I'm not sure if it's relevant or a separate issue, but the time stamp is displaying as 1970.

YWEPCM1, 2016-06-22:
Hi,

I found yi-hack last week and have installed it on my Chinese (CN) version camera. Good results so far for streaming video to VLC. Thanks fritz-smh. Hope to add something back to the project going forward.

I too had problems with motion detection until I got the date and time update from NTP working - may I suggest you ensure the NTP update works. My network problem was self inflicted due to a basic configuration error, once I had actually connected to my wifi network the ntpd update worked, time and date was set and motion was logged.

Edylc, 2016-07-01:
Thanks for the response,

Could you elaborate on how you got the NTP date/time update working?

YWEPCM1, 2016-07-01:
Hi,

Firstly I updated the configuration in yi-hack.cfg - changed the values under "### Network configuration" to match my network, didn't change values under "### Timezone" as they were OK for my location.

Secondly I updated the configuration in wpa_supplicant.conf - only changed the values for "ssid" and "psk".

Then, after trying out the camera and not seeing any motion recorded I examined the contents of log.txt against expected output as per equip_test.sh file and could see there was no response when pinging the NTP server.

I rechecked the changes I'd made to both config files, corrected the typos and successfully retested.

Suggest you share both the configuration files and the log.txt file.

Edylc, 2016-07-01:
yi-hack.cfg
```
### Set the root password
ROOT_PASSWORD=1234qwer

### Network configuration
# Don't forget to also fill the file wpa_supplicant.conf for the wifi configuration
IP=192.168.0.48
NETMASK=255.255.255.0
GATEWAY=192.168.0.1

### Led color after full boot
# The sequence will be :
# - orange : startup
# - blue blinking : configuration
# - blue : configuration finished, camera ready
# - ??? : color set a few seconds after the camera is ready

# values :
#    -boff -bfast : blue blinking
#    -boff -bon : blue on
#    -boff -boff : blue off
#    -boff -yfast : yellow blinking
#    -boff -yon : yellow on
#    -boff -yoff : yellow off
LED_WHEN_READY=-boff -bon

### Timezone
# Example timezones, see more at http://svn.fonosfera.org/fon-ng/trunk/luci/modules/admin-fon/root/etc/timezones.db
#   -Europe/London GMT0BST,M3.5.0/1,M10.5.0
#   -Europe/Paris CET-1CEST,M3.5.0,M10.5.0/3
#   -US/Central CST6CDT,M3.2.0,M11.1.0
TIMEZONE=PST8PDT,M3.2.0,M11.1.0

### NTP server
#If you want to change your ntp server you can see a list here http://www.pool.ntp.org/zone/@
NTP_SERVER=0.us.pool.ntp.org

### Debug
# Debug mode, keep it to 'no' unless you known what you do
# Values : yes|no
DEBUG=no

```

wpa_supplicant.conf
```
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="" //this is correct
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="" //this is correct
}
```

log.txt
```
Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
[... 87 lines cut ...]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1724 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  682 root      1704 S    telnetd
  686 root       984 S    /home/log_server
  929 root         0 SW   [hidog]
  968 root      1144 S <  ./dispatch
  969 root      1064 S    ./exnet
 1226 root     97468 S    ./rmm
 1232 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1268 root      1408 S    ./serverM 80
 1294 root      3360 S    ./record_event
 1295 root     14728 S    ./mp4record 60
 1296 root      1708 S    /bin/sh /home/hd1/test/check_motion.sh
 1308 root     27252 S N  ./rtspsvrM
 1319 root      1700 S    sleep 30
 1441 root       904 S    /home/led_ctl -boff -bon
 1443 root      1708 R    ps

```

YWEPCM1, 2016-07-02:
Hi,

You're wifi network interface (ra0) is not configured, look at the error message regarding RFKILL. This looks like existing issue numbers #31 and #40.

From your log file I think we are running same setup - h/w and f/w, this from my log file:
Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e

The QR code on back of my camera is alpha numeric and begins 11CN... which I understand to be correct hardware version for this hack.

You mentioned camera used to work, have you tried factory reset and redoing setup of SD card, including fresh copy from git to ensure no corruption?

Edylc, 2016-07-05:
Yeah, I've tried resetting with a fresh SD card setup, but that didn't work.  The reset button actually doesn't seem to be working (pressing and holding it has no effect).

Camera QR code begins with 22US.  I also noticed that my log file says 

`Cannot open RFKILL control device Could not set interface ra0 flags (UP): Operation not permitted`

whereas others in those existing issues seemed to be posting

`Cannot open RFKILL control device Could not read interface ra0 flags: No such device`

xumixu, 2016-08-28:
check #31 I found the solution for the rfkill message in another forum