fritz-smh/yi-hack #54: the camera does not record video

Jacques-Lamblot, 2016-06-22:
help
Everything works fine but the camera does not record video

Yet she has 000.qui FOUND create a folder contains a file
FILE 0000.CHK

thank you in advance for your help

Jacques L.

ephestione, 2016-07-13:
To me it only happened when I blocked the outside connection, so the camera couldn't connect to the NTP servers to update the time to the current one.

mustekito, 2016-10-25:
I have a similar problem. The camera connects to wifi, and telnet, rtsp, etc, works fine. The "LED" turning blue for short period and then it blinks orange. **BUT the videos are not saving**, the motion detection does not work...

The log does not show any problem.
```
Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Oct 25  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Oct 25  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Oct 25  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Oct 25  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Oct 25  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = yes
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=0  (0 is ok)
Wifi configuration answer: 
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:26:17:FA  
              inet addr:192.168.0.99  Bcast:192.168.0.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe26:17fa/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:496 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:130101 (127.0 KiB)  TX bytes:300 (300.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.europe.pool.ntp.org':
Previous datetime is Thu Jan  1 01:00:50 CET 1970
Done
New datetime is Thu Jan  1 01:01:20 CET 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1784 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1811 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1918 root      2660 S <  ./rtspsvrM
Processes after startup :
  PID USER       VSZ STAT COMMAND
[... 55 lines cut ...]
  989 root      1068 S    ./exnet
 1122 root      3136 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1784 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1811 root      1408 S    ./serverM 80
 1829 root      3360 S    ./record_event
 1830 root     14728 S    ./mp4record 60
 1859 root      1716 S    /usr/sbin/crond -b
 1908 root       904 S    /home/led_ctl -boff -yfast
 1909 root     94448 S    ./rmm
 1918 root     27252 S N  ./rtspsvrM
 2014 root      1708 R    ps
Storage status :
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.5M      2.7M    868.0K  76% /
tmpfs                    17.2M     16.0K     17.2M   0% /dev
/dev/mtdblock5            8.9M      8.0M    916.0K  90% /home
tmpfs                    32.0M     44.0K     32.0M   0% /tmp
/dev/hd1                 29.3G      8.7M     29.3G   0% /tmp/hd1
tmpfs                   512.0K     16.0K    496.0K   3% /home/mmap_tmpfs
tmpfs                    16.0M         0     16.0M   0% /home/tmpfs
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                 29.3G      8.7M     29.3G   0% /tmp/hd1/test/http/record
```

Can it be problem of setting the timezone? The camera could not connect to the NTP server due to blockage of the outside connection

ephestione, 2016-10-25:
This has been answered already, unless you let the camera get the correct time (which in your case is in 1970 ;)) videos will NOT be recorded.
Allow it to connect to the outside to get the time, or maybe setup a small internal NTP server just for that purpose

mustekito, 2016-10-25:
Setting the time manually, it works. Then, I don't understand. Nothing goes out of the local network, but before forbid it, the camera should be able to connect to the correct time.

How it is prohibiting external access?

niighthawk, 2017-07-16:
how to allow the camera to connect outside to get the time from the ntp server ?

explain step by step, im sure this will help others who dont know how.

ephestione, 2017-07-17:
You just set the correct parameters in the config files so it can connect to the internet! As per previous poster, that would be putting the correct IP of the gateway server, usually the wifi router

On July 16, 2017 5:39:34 PM GMT+02:00, niighthawk <notifications@github.com> wrote:
>how to allow the camera to connect outside to get the time from the ntp
>server ?
>
>explain step by step, im sure this will help others who dont know how.

niighthawk, 2017-07-17:
yeah this is obvious ... but how .. because my router ip is 192.168.0.1 , I also free a ip for the camera 192.168.0.103 

this is my information in the yi-hack.cfg

> ### Network configuration
> # Don't forget to also fill the file wpa_supplicant.conf for the wifi configuration
> IP=192.168.0.103
> NETMASK=255.255.255.0
> GATEWAY=192.168.0.1
> NAMESERVER=192.168.0.1

so I dont understand what else I should do to let the camera acces the internet ?
Do i have to allow a specific port on the router ?

ephestione, 2017-07-18:
Only thing that comes to mind, make sure that the IP you chose is outside of the DHCP range, other than that it seems to me there isn't anything more amiss

On July 17, 2017 2:43:43 PM GMT+02:00, niighthawk <notifications@github.com> wrote:
> yeah this is obvious ... but how .. because my router ip is 192.168.0.1
> , I also free a ip for the camera 192.168.0.103
>
> this is my information in the yi-hack.cfg
>
> configuration
>
> so I dont [...]