fritz-smh/yi-hack #56: Camera file system changed to read-only!

mastix, 2016-06-23:
Bonjour,

for the last 24 hours I was wondering why no new motion files have been recorded. I thought maybe it's a disk space issue... so I telneted into the cam and it turns out that the file system was read-only for some reason.

I rebooted the device... and voilá... I was able to write on the file system again.

How could that have happened?

vdomos, 2016-07-17:
Hello

Same problem for me, the last boot.
May be the quality of sd card for this type of use is significant.

mastix, 2016-07-17:
Firmware is running on a relatively new  high class SanDisk card (32 GB, class 10).

reginastracii, 2016-09-23:
same problem

rbpaul, 2017-04-23:
Any solutions to this problem.  My Filesystem is also repeatedly being set to read-only.

querex007, 2017-04-23:
Same problem over here, I will try again after reboot...

airdrummingfool, 2017-04-23:
I've written a small watchdog script that attempts to remount `hd1` as RW when this happens. It appears to be working for me, but I haven't done a lot of testing yet. Eventually it will make its way into my fork of this repo as a service, but it's not there yet. Here's the code:

sd/test/services/rw_watch.sh:
```

airdrummingfool, 2017-04-23:
I've written a small watchdog script that attempts to remount `hd1` as RW when this happens. It appears to be working, but I haven't done a lot of testing. Eventually it will make its way into my fork of this repo as a service, but it's not there yet. Here's the code:

file `sd/test/services/rw_watch.sh`:
``` bash
# This service acts as a watchdog to make sure /tmp/hd1 remains writable

check_mounts() {
	cat /proc/mounts | grep "$1" | grep rw > /dev/null
}

fix_mounts() {
	echo "Loss of RW on $1 at `date`" >> /home/rwlog.txt
	mount /dev/hd1 /home/hd1 -o remount
	mount /home/hd1/record/ /home/hd1/test/http/record/ -o bind -o remount
	echo "Attempted recovery of RW at `date`"
	sync
	sleep 2
	# reboot
}

start() {
	echo "Starting RW check..."
	while true; do
		check_mounts "/tmp/hd1" || fix_mounts "/tmp/hd1"
		sleep 30
	done
}

stop() {
	pkill -TERM -f "`basename $0` start"
}

# Call chosen script in args
"$@"

```

to start the service, use the following (make sure you fork (`&`)!). I added it near the end of `equip_test.sh`:
``` bash
/home/hd1/test/services/rw_watch.sh start >> /home/hd1/test/log.txt &
```

to stop the service, use (notice no fork necessary)
``` bash
/home/hd1/test/services/rw_watch.sh stop
```

When the script thinks that RW is lost, it logs to `/home/rwlog.txt`; `/home` never loses RW, so I log things there that I want to stick around even if RW has been lost. Once it has attempted remount, it logs to stdout, which is redirected to `/home/hd1/test/log.txt` (the hack's logfile). Since that file is on `hd1`, if the log message shows up, then remounting worked.

querex007, 2017-11-26:
Thanks, this helps me out!