fritz-smh/yi-hack #63: Force led off after boot

vdomos, 2016-07-22:
Hello,

I don't understand how to turn off the blue led after the boot.

I test this in /tmp/hd1/test/yi-hack.cfg:

     ...
     # values :
     #    -boff -bfast : blue blinking
     #    -boff -bon : blue on    
     #    -boff -boff : blue off   
     #    -boff -yfast : yellow blinking  
     #    -boff -yon : yellow on    
     #    -boff -yoff : yellow off 
     #LED_WHEN_READY=-boff -bon
     LED_WHEN_READY=-boff -boff 
    or 
     LED_WHEN_READY=-boff

The led stay always On.
With this configuration : LED_WHEN_READY=-boff -boff
The process is like this:

     1417 root       904 S    /home/led_ctl -boff -bon -boff -boff

A idea ?

fritz-smh, 2016-07-22:
Try -boff -yoff

<div>-------- Message d'origine --------</div><div>De : vdomos <notifications@github.com> </div><div>Date :2016/07/22  13:52  (GMT+01:00) </div><div>À : fritz-smh/yi-hack <yi-hack@noreply.github.com> </div><div>Objet : [fritz-smh/yi-hack] Force led off after boot (#63) </div><div>
</div>Hello,

I don't understand how to turn off the blue led after the boot.

I test this in /tmp/hd1/test/yi-hack.cfg:

 ...
 # values :
 #    -boff -bfast : blue blinking
 #    -boff -bon : blue on    
 #    -boff -boff : blue off   
 #    -boff -yfast : yellow blinking  
 #    -boff -yon : yellow on    
 #    -boff -yoff : yellow off 
 #LED_WHEN_READY=-boff -bon
 LED_WHEN_READY=-boff -boff 
or 
 LED_WHEN_READY=-boff
The led stay always On.
With this configuration : LED_WHEN_READY=-boff -boff
The process is like this:

 1417 root       904 S    /home/led_ctl -boff -bon -boff -boff
A idea ?

tuaminx, 2016-08-02:
Did you reboot your camera. It works for me.
I set 
```bash
LED_WHEN_READY=-boff -boff
```
then reboot camera by the simplest way is plugging out and in the micro USB cable.
It's possible to reboot by telnet with command
```bash
reboot
```

vdomos, 2016-08-31:
I found the issue.

I use comment before modify the line with LED_WHEN_READY variable

It's look like that in the file:

    # grep READY /tmp/hd1/test/yi-hack.cfg
    #LED_WHEN_READY=-boff -bon
    #LED_WHEN_READY=-boff -boff
    LED_WHEN_READY=-boff -yoff

When I reboot the camera, led process is with too much parameters:

    # ps |grep led
    1439 root       904 S    /home/led_ctl -boff -bon -boff -boff -boff -yoff

With only one ligne in the file:

    # grep READY /tmp/hd1/test/yi-hack.cfg
    LED_WHEN_READY=-boff -boff

It"s ok with led process and the led is off:

    # ps |grep led
    1428 root       904 S    /home/led_ctl -boff -boff

Stange behaviour with comment ?

Shuunen, 2016-10-05:
Weird, I just did the "hack" yesterday without issue with the blue led , I used :
`LED_WHEN_READY=-boff -boff`
And it worked great.

lifehome, 2018-10-25:
@vdomos Can you try the value out? If possible please close this issue after your success.