fritz-smh/yi-hack #64: Cannot connect (log attached).

magadan, 2016-07-27:
Hello, no connection successfull. Here is my log:

Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Jun  8  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Jun  8  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Jun  8  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Jun  8  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Jun  8  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Successfully initialized wpa_supplicant rfkill: Cannot open RFKILL control device Could not read interface ra0 flags: No such device WEXT: Could not set interface 'ra0' UP ra0: Failed to initialize driver interface
Status for wifi configuration=0  (0 is ok)
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.uk.pool.ntp.org':
Previous datetime is Thu Jan  1 00:00:15 GMT 1970
Done
New datetime is Thu Jan  1 00:00:15 GMT 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1160 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1196 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1236 root      2660 S <  ./rtspsvrM
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  620 root         0 RWN  [jffs2_gcd_mtd4]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  682 root      1704 S    telnetd
  686 root       984 S    /home/log_server
  927 root         0 SW   [hidog]
  965 root      1164 S    ./peripheral
  966 root      1144 S <  ./dispatch
  967 root      1068 S    ./exnet
 1154 root     97480 S    ./rmm
 1160 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1196 root      1408 S    ./serverM 80
 1222 root      3360 S    ./record_event
 1223 root     14728 S    ./mp4record 60
 1224 root      1708 S    /bin/sh /home/hd1/test/check_motion.sh
 1235 root      1700 S    sleep 30
 1236 root     27252 S N  ./rtspsvrM
 1369 root       904 S    /home/led_ctl -boff -bon
 1375 root      1708 R    ps
Start telnet server...

magadan, 2016-07-27:
Pls, can you update to the new hardware version of CN versions ?

In my log the ra0 device not exists !

Could not set interface 'ra0' UP ra0: Failed to initialize driver interface

Yboss2, 2016-07-31:
I have exaclty the same problem than you

Mika17, 2016-08-05:
I have the same problem too.
A lot of people have this problem. The WiFi cheap isn't the same.
Could we expected an update of yi hack or should we try an other solution?

Thank you !