fritz-smh/yi-hack #65: DHCP instead of static IP

jcconnell, 2016-07-29:
Possible to change to a DHCP reservation instead of a static IP? I attempted to change this from telnet using the below command without success. I would like to have the camera IPs managed from a central server.

`ifconfig r0 0.0.0.0 0.0.0.0 && dhclient`

buckshee, 2016-09-30:
These things use `udhcpc` so from line 266 I changed to this:

```
log "Do network configuration (DHCP)"
udhcpc -b -i ra0
log "Done"
```
I also removed the DNS bit below since it's not needed

TODO: Check what the script copying to /usr/share/udhcpc is for, and a way to force usage of an internal NTP server, possibly with udhcpc too

jcconnell, 2016-12-25:
Sorry I am so late to this, I never noticed your response. What specifically did you change? I see the spot you are referencing at line 266. But from 266 to what was replaced with that snippet above?