fritz-smh/yi-hack #69: New shell script to upload files to FTP server

tuaminx, 2016-08-05:
I created a new script to support this with features:

1. No looping, very suitable for crontab, increase streaming performance (I feel that)
2. Last uploaded file is saved for next uploading
3. Last failed uploading file is save for next try

How to use: 

1. Open file upload_to_ftp.sh and change the server's information.
2. Create a log file: /tmp/hd1/test/ftp_upload_log.txt
3. Move file ftp_upload_mem to /tmp/hd1/test/ftp_upload_mem
4. Execute the script manually without any parameter
5. Use crontab to execute the script automatically

Script: https://github.com/tuaminx/betterhome/blob/master/yi-hack-support/

TODO:
For (3): need to find a way to make give timeout to ftpput. Current, if ftpput is working and the FTP server is unreachable, I saw that it's hang.

trust199, 2016-08-23:
Did you delete that script?

tuaminx, 2016-08-24:
Oh, I pushed an improvement yesterday night and don't know what wrong with this folder. 
I will recover it soon.

tuaminx, 2016-08-24:
I bring them back :)

tuaminx, 2016-09-03:
@trust199 : if you are using the script, please update new version. I had a mistake which logging to /var/log. This leads to rootfs is out of space soon then camera cannot startup. 
The easy phenomenon is that camera cannot connect to wifi.

karepiu, 2017-03-15:
When you use FTP you are sending your password in plain text in network. I suggest to use SFTP. So far looks like it will be possible with cURL.