fritz-smh/yi-hack #71: Blue light stop blinking but don't work

expouic, 2016-08-13:
After finish all steps to setup with this hack tryed to list IP 192.168.1.121 on list of devices of my router and don't be appear!

Log:

> 
> Starting to log...
> Hardware version informations : 
>     hwplatform=2
> Software version informations : 
>     [sys]
>     version=1.8.5.1M_201512011815
>     [kernel]
>     version=3-0.8
>     provider=18e
>     devtype=18e
> Firmware letter is : 'M'
> The RTSP server binary version which will be used is the 'M'
> The HTTP server binary version which will be used is the 'M'
> Check for some files size...
>     -rwxr-xr-x    1 root     root        144885 Jun  8  2016 /home/hd1/test/http/serverJ
>     -rwxr-xr-x    1 root     root        144912 Jun  8  2016 /home/hd1/test/http/serverM
>     -rwxr-xr-x    1 root     root        121932 Jun  8  2016 /home/hd1/test/rtspsvrI
>     -rwxr-xr-x    1 root     root         39572 Jun  8  2016 /home/hd1/test/rtspsvrK
>     -rwxr-xr-x    1 root     root         39580 Jun  8  2016 /home/hd1/test/rtspsvrM
> The blue led is currently blinking
> Debug mode = yes
> Check for wifi configuration file...*
> /home/wpa_supplicant.conf
> Start wifi configuration...
> Successfully initialized wpa_supplicant rfkill: Cannot open RFKILL control device Could not set interface ra0 flags (UP): Operation not permitted WEXT: Could not set interface 'ra0' UP ra0: Failed to initialize driver interface
> Status for wifi configuration=0  (0 is ok)
> Do network configuration 1/2 (ip and gateway)
> Done
> Configuration is :
>     lo        Link encap:Local Loopback  
>               inet addr:127.0.0.1  Mask:255.0.0.0
>               inet6 addr: ::1/128 Scope:Host
>               UP LOOPBACK RUNNING  MTU:16436  Metric:1
>               RX packets:0 errors:0 dropped:0 overruns:0 frame:0
>               TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
>               collisions:0 txqueuelen:0 
>               RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
>     
> Do network configuration 2/2 (DNS)
> Done
> Get time from a NTP server...
> But first, test the NTP server '0.uk.pool.ntp.org':
> Previous datetime is Wed Dec 31 17:00:20 GMT 1969
> Done
> New datetime is Wed Dec 31 17:00:20 GMT 1969
> Start blue led on
> Start ftp server...
> Check for ftp process : 
>  1177 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
> Start http server : serverM...
> Check for http server process : 
>  1213 root      1408 S    ./serverM 80
> Start rtsp server : rtspsvrM...
> Check for rtsp process : 
>  1249 root      2660 S <  ./rtspsvrM
> Processes after startup :
>   PID USER       VSZ STAT COMMAND
>     1 root      1708 S    init
>     2 root         0 SW   [kthreadd]
>     3 root         0 SW   [ksoftirqd/0]
>     4 root         0 SW   [kworker/0:0]
>     5 root         0 SW   [kworker/u:0]
>     6 root         0 SW   [rcu_kthread]
>     7 root         0 SW<  [khelper]
>     8 root         0 SW   [kworker/u:1]
>   143 root         0 SW   [sync_supers]
>   145 root         0 SW   [bdi-default]
>   146 root         0 SW<  [kintegrityd]
>   148 root         0 SW<  [kblockd]
>   161 root         0 SW   [khubd]
>   174 root         0 SW<  [cfg80211]
>   175 root         0 SW   [kworker/0:1]
>   257 root         0 SW<  [rpciod]
>   260 root         0 SW   [kswapd0]
>   314 root         0 SW   [fsnotify_mark]
>   322 root         0 SW<  [nfsiod]
>   331 root         0 SW<  [crypto]
>   405 root         0 SW   [mtdblock0]
>   410 root         0 SW   [mtdblock1]
>   415 root         0 SW   [mtdblock2]
>   420 root         0 SW   [mtdblock3]
>   425 root         0 SW   [mtdblock4]
>   430 root         0 SW   [mtdblock5]
>   435 root         0 SW   [mtdblock6]
>   438 root         0 SW   [romblock0]
>   441 root         0 SW   [romblock1]
>   444 root         0 SW   [romblock2]
>   447 root         0 SW   [romblock3]
>   450 root         0 SW   [romblock4]
>   453 root         0 SW   [romblock5]
>   456 root         0 SW   [romblock6]
>   591 root         0 SW<  [kpsmoused]
>   600 root         0 SW   [kworker/u:2]
>   616 root         0 SW   [mmcqd/0]
>   620 root         0 SWN  [jffs2_gcd_mtd4]
>   624 root      1712 S    /bin/sh /etc/init.d/rcS
>   636 root       872 S <  udevd --daemon
>   642 root      1704 S    /bin/sh /etc/init.d/S80network
>   656 root         0 SWN  [jffs2_gcd_mtd5]
>   657 root      1708 S    /bin/sh ./init.sh
>   664 root         0 SW   [flush-mtd-unmap]
>   675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
>   677 root         0 SW   [flush-179:0]
>   681 root      1704 S    telnetd
>   686 root       984 S    /home/log_server
>   929 root         0 SW   [hidog]
>   967 root      1164 S    ./peripheral
>   968 root      1144 S <  ./dispatch
>   969 root      1060 S    ./exnet
>  1161 root     97488 S    ./rmm
>  1177 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
>  1213 root      1408 S    ./serverM 80
>  1239 root      3360 S    ./record_event
>  1240 root     14728 S    ./mp4record 60
>  1241 root      1708 S    /bin/sh /home/hd1/test/check_motion.sh
>  1249 root     27252 S N  ./rtspsvrM
>  1262 root      1700 S    sleep 30
>  1386 root       904 S    /home/led_ctl -boff -bon
>  1390 root      1708 R    ps

expouic, 2016-08-13:
CAM serial number 22CN307VHD

xyannis, 2016-08-16:
Check your wpa_supplicant.conf

You must change the ssid ans psk

Example:

ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="MyWifi"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="Mywifipassword"
}

expouic, 2016-08-16:
I have changed the configuration file with my network ssid and correct psk but still all the same!

arsya, 2016-08-27:
From your log:
--------
Successfully initialized wpa_supplicant rfkill: Cannot open RFKILL control device Could not set interface ra0 flags (UP): Operation not permitted WEXT: Could not set interface 'ra0' UP ra0: 
--------
I think this is the same issue as #31 
This is because most probably your wifi chipset is bcm43143, which is not this firmware expected.

expouic, 2016-09-04:
how can I confirm that my wifi chipset is this bmc43134?

If is, do you have any other solution?

arsya, 2016-09-05:
@expouic with lsusb in equip_test.sh
If you have a device 0a5c, that's the bcm43143 wifi chipset

umasse, 2017-01-14:
This fork of yi-hack supports the bcm43143 chipset, and comes with firmware 1.8.6Q.
It's working great with my Chinese camera I just bought.

expouic, 2017-02-06:
To solve this problem I had connected the camera from the serial port.

Create this video with instructions (step by step)

[https://www.youtube.com/watch?v=fBjKTfWN4TI](url)

closed 2017-02-06 by expouic