fritz-smh/yi-hack #74: Motion Detection Notification

Clem-, 2016-08-15:
Do you know if there's a easy solution to receive notification (on smartphone) when a motion is detected?

indeu, 2016-08-17:
Here is some scripts for sending to GCM. I'm not sure it can run this scripts on yi-cam server. 

REGISTRATION_ID=YOUR_GCM_REGISTRATION_ID
SERVER_KEY=YOUR_SERVER_KEY_FROM_GOOGLE_API_CONSOLE
curl --header "Authorization: key=$SERVER_KEY" --header  Content-Type:"application/json"  https://android.googleapis.com/gcm/send  -d  "{ \"data\" : {\"foo\": \"bar\"}, \"registration_ids\":[\"$REGISTRATION_ID\"]  }"

It need to fill up the ID and Key of GCM.

JeanPrendnote, 2016-08-20:
i've read something about a hack for this cam and Pushbullet app notification on 4pda forum thread about this camera

JeanPrendnote, 2016-08-20:
this may help : https://www.dealabs.com/forums/le-bar/le-laboratoire-high-tech--informatique/camra-ip-xiaomi-xiaoyi-720p/11335?page=8#comment_373815

karepiu, 2017-03-15:
The easiest way will be to use IFTTT. Just setup maker and then create a script to execute it when video is noticed ( wget should work )

utya1988, 2017-03-19:
```
#!/bin/sh
# each minute, check for a new video file (which is created in case of motion detection)
# and if found, create the appropriate file for the http server

API_TOKEN='API Token'

CHAT_ID_1='Chat ID'
CHAT_ID_2='Chat ID'
DIR=/home/hd1/record/

cd /home/hd1/record/
old_motion_file=1
new_motion_file=1

while [ 1 -eq 1 ] 
  do
    motion_file=$(find . -type f -name "*.mp4" -mmin -1 | tail -1)
    echo $motion_file | sed "s/.\//record\//" > /home/hd1/test/http/motion

send_motion_file=$(echo $motion_file  | sed "s/.\//\/home\/hd1\/record\//")

if [[ "$motion_file" = "$old_motion_file" ]];  then
	echo "math"
##	
	else 
	echo "Not match"
	
	## Send text to users
	#/home/curl --insecure --data "chat_id=$CHAT_ID_1&text=Moving detection" https://api.telegram.org/bot$API_TOKEN/sendMessage > /dev/null
	#/home/curl --insecure --data "chat_id=$CHAT_ID_2&text=Moving detection" https://api.telegram.org/bot$API_TOKEN/sendMessage > /dev/null
	
	## Send Video to users
	/home/curl --insecure  -F document=@$send_motion_file  https://api.telegram.org/bot$API_TOKEN/sendDocument?chat_id=$CHAT_ID_1 > /dev/null
	/home/curl --insecure  -F document=@$send_motion_file  https://api.telegram.org/bot$API_TOKEN/sendDocument?chat_id=$CHAT_ID_2 > /dev/null
		
	old_motion_file=$motion_file;
fi
    sleep 25
done
```

my result. I use  push from telegram. But sometime camera freeze.

gl-lamhnguyen, 2017-04-25:
@utya1988 every minute seems like a very long latency, is there anyway we can tap into the motion detect function of the camera and run script from there?

utya1988, 2017-04-25:
mp4record is a binary. So if you can find source of motion detect  and mp4record script, it will be great

karepiu, 2017-04-26:
You can start motion detection with just 1 second instead of 60. It accepts number of seconds as parameter. 

Also you can always try decompiling the file and modifying, but this will not be very easy task.