fritz-smh/yi-hack #79: How can turn off blue light when connected?

cuibap1212, 2016-08-31:


xyannis, 2016-09-01:
LED_WHEN_READY=-boff -boff

cuibap1212, 2016-09-05:
great!
thanks xyannis,

I change: TIMEZONE with:
TIMEZONE=GTM+1 - datetime show from 1969/01/01-time
TIMEZONE=ITC-7 - datetime show from 1970/01/01-time
What's happen?
How to change TIMEZONE to GTM + 7?
thanks,

xyannis, 2016-09-05:
Tell me your location city and county

cuibap1212, 2016-09-06:
my country: VietNam (GTM +7)

xyannis, 2016-09-06:
ITC-7 is correct. Make sure that you have internet connection on your router before you activate the camera. If you dont have internet access, timezone will show 1970

cuibap1212, 2016-09-06:
I think some mistake when synchronize timezone from NTP_SERVER=0.uk.pool.ntp.org, because Internet is connected!
I saw you post link: http://www.remotemonitoringsystems.ca/time-zone-abbreviations.php
i'll try change timezone with WIB-7 (it's same GTM +7)

hermitcrabslab, 2016-09-06:
Looks like there is a bug with new BCM43134 enabled devices, NTP_SERVER is called too early IMHO.