fritz-smh/yi-hack #82: IP camera is not connecting to wifi and LED off after steady blue

pkgithub83, 2016-09-08:
I have used the same configuration for earlier xiaomi yi ant ip camera and it worked. However, recently I bought another new xiaomi yi ant ip camera and did the same thing which I did for the earlier one but this time it does not work. I am not sure if the firmware for both ip camera differs but upon startup, steady yeloow, blinking blue with clicking sounds, steady blue for seconds before the blue LED light died off. Below is what I have gotten from the log.txt and seemed like it was not able to connect to the wifi. May I know current model has beefed up their security against such hack or anyone else purchased recently and faced the same issue? I hope that I screwed up somewhere and get corrected as soon as possible. Thank you in advance.

_Cannot open RFKILL control device Could not read interface ra0 flags: No such device WEXT: Could not set interface 'ra0' UP ra0: Failed to initialize driver interface_

Regards,
Noob

cuibap1212, 2016-09-08:
Make you sure new iP address is not conflict on lan?

tuaminx, 2016-09-08:
It must be empty space of rootfs which leading to an issue that Wireless driver data cannot be copied to /etc/Wireless folder during camera startup.
Please try `df -h` on telnet then you can see.
Empty some stuff on / folder. Ex: /IMG*** (if it is existed) or /var/*. Then, reboot the camera

pkgithub83, 2016-09-08:
@cuibap1212 , I have tried to ping the IP address that I edited yi-hack.cfg but no response at the same time checked the connected clients to the router, so I do not think there was conflict of IP address.

@tuaminx , I think my issue is just like #64 and I read somewhere about the wifi chipset was difference. Perhaps the hardware itself is not the same. There is another issue that highlighted that wifi chipset is flashed and bricked during the process. I will still try out for suggested solution because I saw IMG001.exe somewhere and using RT2870STA.dat in certain path as well, but that is used when using an apps to configure the IP camera. I am intending to configure the IP camera just by editing wpa_supplicant.conf and yi-hack.cfg then connect it to the wifi hosted by rpi3, the previous IP camera worked but I do not have the serial number with me to determine if they are from the same batch.

tuaminx, 2016-09-09:
@pkgithub83 In both cases of using Xiaomi apps or this yi-hack to configure camera, the /home/init.sh script is executed and below step is done:
```
if [ -f "/home/RT2870STA.dat" ]; then
    cp /home/RT2870STA.dat /etc/Wireless/RT2870STA/
    sync
fi
```
So, in your equip_test.sh, just add a command to check if the RT2870STA.dat exists.
```
ls /etc/Wireless/RT2870STA/RT2870STA.dat > /tmp/hd1/test/check_wifi.log
```
This dat file is not always in the path, it's copied to /etc/Wireless folder each time the camera reboot/startup. It's existed when you flash your camera by `home` file.
This file is not the wireless driver, it is the data for WEXT driver which is used by below command is in your equip_test.sh
```
/home/wpa_supplicant -B -i ra0 -c /home/wpa_supplicant.conf
```
In my case, the rootfs is empty so there is no space for that file. It make the above `cp` command wrong. But there is not failure check in the init.sh so we don't have any information.
I had this issue and spent 2 full weekend days to find the cause 😢

hermitcrabslab, 2016-09-10:
@pkgithub83 try [my fork](https://github.com/hermitcrabslab/yi-hack) and see if works ;)
Make sure you flash first latest firmware from my fork.
Good luck!

arsya, 2016-09-11:
@hermitcrabslab I tried [your fork](https://github.com/hermitcrabslab/yi-hack) and it's working as expected. Thank you so much for updating the binaries. I hope your updates could be integrated with this repository.

pkgithub83, 2016-09-13:
@hermitcrabslab , I have tried your fork and it eliminated the ra0 issue. Thanks for the nice tune during the startup too, but it could not connect to wifi and ended up with steady blue turned blinking yellow light. I will try to connect with other wifi just in case it was due to my own carelessness. Thank you anyway for your fork!

hermitcrabslab, 2016-09-14:
@pkgithub83 upload somewhere your log.txt and all .cfg files to check(remove passwords)

pkgithub83, 2016-09-15:
@hermitcrabslab , I have included the bind timeout log report as well, though I do not know what it meant by bind timeout, due to DNS setting request? 
[https://drive.google.com/drive/folders/0B7Wg-7l8FmGab2xCZlk4b2FQMFk?usp=sharing](url)

By any chance, anyone tried to stream video online from this IP camera, so that you can view though the internet without using its apps? Other than Wowza, there is no other way without relying on third party services?

cuibap1212, 2016-09-15:
Please check: **wpa_supplicant.conf**
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="ssidname"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="password"
}
**yi-hack.cfg**
IP=192.168.0.88
NETMASK=255.255.255.0
GATEWAY=192.168.0.1

Maybe you clear NameServer.

> NAMESERVER=192.168.0.1

hermitcrabslab, 2016-09-15:
@pkgithub83 I get 404 on that link,  can you upload somewhere else?

pkgithub83, 2016-09-19:
@cuibap1212  I have tried removing that parameter but problem remained. I will look at the other parameters just in case.
@hermitcrabslab I uploaded files in dropbox and below is the link. Thank you in advance
https://www.dropbox.com/sh/1mnv7pdx4dd6ipx/AADwb-hA36lewHM-mgFA37GPa?dl=0

cuibap1212, 2016-09-20:
@pkgithub83: I have suggest:
First you test with firmware from xiaomi yi (china or international).
If connect successful, you process hack Yi.
1. Format SD
2. Download and Copy all file to SD card
3. Update wpa_supplicant.conf with ssid and psk
4. Check security your wifi: WPA-PSK
5. If you can, you change router with GATEWAY=192.168.1.1 (you should backup config)
6. Don't change anything yi-hack.cfg
7. Reject all device connect your network.
8. Check connect Yi

ephestione, 2016-11-07:
was pulling my hair out, re-read the issues, tried this fork, WORKS! I would just like for it to have no voices since I want to use this camera outside near the street, other than that, props!

ephestione, 2016-11-07:
I obviously have had positive experiences with fritz's build with other two (maybe older) cameras, so was puzzled since I did the same thing with the latest and wasn't working

robertbeal, 2016-11-22:
I've currently got this issue. Ironically my camera has been working fine for months. I power it off one day, and now when powering back on it, the wpa steps passes (status 0) but I can't see it on my network via the router or nmap. 

I'll maybe try the fork. Still puzzled why it was working and has recently decided to stop working,

mp3il, 2016-12-16:
@tuaminx  - how did you fixed the problem that the wifi driver dat wasn't copied? simply edited equip_test with a rm commands? what files did u delete? i'm spending a whole weekend on this here too :(

mp3il, 2016-12-16:
Update: Simply 
`ls -laRh /` into some file using equip_test.sh to find what file is taking the space in rootfs and preventing from the **RT2870STA.dat** from being copied.

now everything works, amazing.

Structure3creative, 2017-02-24:
Trying this code out and and I'm not sure if I'm adding the network info correctly. The startup process starts to cycle through the blinking blue, and to a solid blue, and then goes to a blinking orange - and stays there. Any ideas as to what I should try to get it working?

ephestione, 2017-02-24:
if you are using a newer version of this camera, you need to dig into 
the forks for the newer version of the hack. Sometimes, there isn't 
enough free space in the camera firmware chip for the modifications, so 
you need this new version which has a different allocation and does work.
I had a similar problem with the third camera I bought

skeeto1983, 2017-04-13:
@ephestione 
I have a newer model I just picked up on Amazon. The original fritz hack didn't work for me. I rolled back to the international firmware 1.85 and it seemed like it was going to stick but I had the same issue regarding the LEDs trying to sync up to wifi but eventually failing and going back to blinking orange. 

My log file has not been exactly helpful as it seems to just breeze past the wifi without any true ERROR. 

Before trying the @hermitcrabslab fork, is there a particular FW to flash to considering my newer camera the is likely experiencing this wifi chipset issue?

ephestione, 2017-04-13:
No particular idea about updating,I would just go with the fork as it is a firmware update in itself

On April 13, 2017 6:58:38 AM GMT+02:00, skeeto1983 <notifications@github.com> wrote:
> @ephestione
> I have a newer model I just picked up on Amazon. The original fritz
> hack didn't work for me. I rolled back to the international firmware
> 1.85 and it seemed like it was going to stick [...]

skeeto1983, 2017-04-14:
Tried the @hermitcrabslab fork for this hack on my camera. I got the infamous Solid Yellow -> Blinking Yellow -> Blinking Blue -> Brief Solid Blue -> Blinking Yellow sequence of doom. Here's my log file:

```
Starting to log...
Start telnet server...
MTK 7601
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.6.1Q_201607271501
    [kernel]
    version=3-0.8
    provider=1
    devtype=1
Firmware letter is : 'Q'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Sep  9  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Sep  9  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Sep  9  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Sep  9  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Sep  9  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=0  (0 is ok)
Wifi configuration answer: 
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:99:7C:C4  
              inet addr:192.168.1.121  Bcast:192.168.1.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe99:7cc4/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:1152 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:270701 (264.3 KiB)  TX bytes:300 (300.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.uk.pool.ntp.org
':
Previous datetime is Thu Jan  1 00:01:02 UTC 1970
Done
New datetime is Thu Jan  1 00:01:32 UTC 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1803 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1830 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1933 root      2660 S <  ./rtspsvrM
[... 59 lines cut ...]
 1803 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1830 root      1408 S    ./serverM 80
 1848 root      3360 S    ./record_event
 1849 root     14732 S    ./mp4record 60
 1878 root      1716 S    /usr/sbin/crond -b
 1927 root       904 S    /home/led_ctl -boff -yfast
 1928 root     94464 S    ./rmm
 1933 root     27252 S N  ./rtspsvrM
 2033 root      1708 R    ps
Storage status :
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.5M      2.7M    864.0K  76% /
tmpfs                    17.2M     16.0K     17.2M   0% /dev
/dev/mtdblock5            8.9M      8.1M    800.0K  91% /home
tmpfs                    32.0M     48.0K     32.0M   0% /tmp
/dev/hd1                  7.2G      9.4M      7.2G   0% /tmp/hd1
tmpfs                   512.0K     16.0K    496.0K   3% /home/mmap_tmpfs
tmpfs                    16.0M         0     16.0M   0% /home/tmpfs
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                  7.2G      9.4M      7.2G   0% `/tmp/hd1/test/http/record
```

Not sure exactly how to decipher or next steps to take without having any access to the camera over wifi. I have made the requisite changes to supplicant.conf and yi-hack.cfg regarding wifi details.

Any suggestions welcome. I have heard tell of needing to delete a file called RT2870STA.dat but no clue if this even applies to me or how to gain access to it. Perhaps a piece of script can handle this fix.