fritz-smh/yi-hack #84: turn off the IR leds

reginastracii, 2016-09-15:
can switching off the IR leds be added to the script?
I have the cam behind a glass window so low light mode is unusable. 
the option is available in the app so it is technically possible to switch them off.

irovder, 2016-11-29:
This is very important if the camera is used in the night behind a window or similar.

irovder, 2016-12-01:
Disable IR LED with the command:
/home/sendMq 0x10 1 0x1141 0
This works, but only once and after the light changes you need to run it again.
Workaround:
crontab job with that command or put it inside the loop in check_motion.sh.

reginastracii, 2016-12-09:
thanks,
here's the dumb guy work around  I came up with.
revert to the original software; install the annoying chinese app back on the phone and do the settings you want there - including IR off and image rotation. 
pop the card back in.
In my case all the settings were kept when running on the hack.

irovder, 2016-12-09:
Not bad;) Which chinese version do you use? Mine has no "IR off" option...

reginastracii, 2016-12-11:
I just let the app update to the current ver it has the IR-off setting.

airdrummingfool, 2016-12-11:
The final IR state can be configured in `yi-hack.cfg`:
```ini
LED_WHEN_READY=-boff -yoff
```

You can also manually update the LED state via telnet (note that you must kill the existing `led_ctl` process before starting another):
```sh
killall led_ctl
/home/led_ctl -yoff -boff &
```

irovder, 2016-12-11:
Interesting as I have the "YI Home Camera Version: 2.10.2.12_20161207" which I think is actual (and FW 1.8.5.1M_201512011815) and still no IR-off setting...

joozy, 2016-12-12:
I've the same Yi Home version, 2.10.2.12_20161207 (fw Q).

Select camera, select setting (top left gear icon) select camera setting - Night Vision IR Light toggle.

Do you have the toggle?

reginastracii, 2016-12-12:
mine looked same as joozy's I uninstall it when I'm done with settings so couldn't take a scrshot

irovder, 2016-12-12:
Nope but my FW is M instead of Q - maybe that's why. Is the hack working with Q without problems?

joozy, 2016-12-12:
I tested the hack with Q but at some point that camera refused to boot...constant yellow, I think. Did not attempt to reflash. Reverted to a different hack. 

Not conclusive that it does not work with Q, could have been something I did wrong.

R is out.

ygator, 2017-01-08:
The setting that corresponds to the Night Vision IR Light in the app is wrHdThreadKB in /etc/ui.conf.
2=Off 3=On

irovder, 2017-01-08:
I edited the file, changed the param from 1 to 2, rebooted but IR light is still on.

ygator, 2017-01-08:
@irovder Did you check that the setting is still 2 after reboot?  When I make changes I normally do the following:
1. /home/killapp.sh
2. kill -9 log_server dispatch
3. Make changes
4. sync; sync; reboot

What firmware version are you on?  I guess it could be different.  You can always check.  Telnet in and change it in the app and see what changes in /etc/ui.conf.  That is how I got the various settings.

irovder, 2017-01-08:
Thanks for your answer - yes the setting is still 2 after reboot; using Q FW - app shows no IR off option so that could be the problem. Maybe I should try another FW but don't know which one works with this hack (China HW).

ygator, 2017-01-08:
I also have two china cameras which are on 1.8.5.1M_201512011815, but they do not have IR.

airdrummingfool, 2017-01-10:
In my testing (firmware 1.8.5.1M_201512011815), a modified `ui.conf` has to be written to `mtd2` in order to have any effect (though I've only played with a couple settings).

Here's how I write to it (adapted from `update.sh`) - do this at your own risk!:
```sh
sync
TMPCONF=$(sed  '/conf/!d' /proc/mtd |awk -F':' '{print $1}')
/home/flash_eraseall /dev/$TMPCONF
dd if=/etc/ui.conf of=/dev/$TMPCONF count=128 bs=1024
sleep 5 && reboot
```

ygator, 2017-01-10:
@airdrummingfool, I don't think that is really necessary.  I have tried the wrHdThreadKB (IR) on my US camera and it works.  Also ddnsSIndex and ptz_hflip works on my China cameras.  Have not tried anything else.

airdrummingfool, 2017-01-11:
@ygator thanks for letting me know. I must not have been using `sync` to flush the disk buffer or something like that.

staresAtFlames, 2017-10-18:
@irovder  When you place the command to turn off the IR in the check motion script, how exactly do you place it in? Could you possibly share the contents of your check_motion.sh?

When I submit the command directly it works, but I'm having trouble putting it in the loop and getting it to work. I'm just using VI to edit the file, but I could pull the SD card out to edit on computer, too.

Clem-, 2017-10-18:
Hi, 
Did you try running the command with its full path like: 
‘/home/led_ctl -boff -yoff &’

staresAtFlames, 2017-10-18:
I did, I placed the line "/home/sendMq 0x10 1 0x1141 0" into the loop in check_motion.sh.

jelbo, 2019-01-13:
Has anyone gotten this to work? Like @josephdonati, the command `/home/sendMq 0x10 1 0x1141 0` works, but putting it in `\test\check_motion.sh` does not. How I have set it up:

```
cd /home/hd1/record/
while [ 1 -eq 1 ] 
  do
    /home/sendMq 0x10 1 0x1141 0
    motion_file=$(find . -type f -name "*.mp4" -mmin -1 | tail -1)
    echo "M="$motion_file
    echo $motion_file | sed "s/.\//record\//" > /home/hd1/test/http/motion
    sleep 30
done
```

I've also edited `/etc/ui.conf` from `wrHdThreadKB=1` to `wrHdThreadKB=2` and I've even done the steps  @airdrummingfool suggested, using dd.

Then I've tried to turn off the IR option using the Yi Home app. I downloaded an .apk from version 2.10.19.1_20171213, which worked. I upgraded the firmware to 1.8.7-something and then the option was there.

Going back to the hacked firmware and the IR LEDs go back on..

jcconnell, 2019-05-14:
I'm running this firmware on a 22US: https://github.com/npetrovski/yi-hack-easy

I was able to horizontally flip my image by:
```
vi /etc/ui.conf
sync
reboot
```

Working great now.