fritz-smh/yi-hack #85: Camera doesn't answer after a while

jimbo72, 2016-09-16:
After a random time (a couple of minutes, hours, or days when lucky) the camera stops responding.
No more video stream, no telnet, no response to ping.

I would like to known if others have the same issue ?

How to debug this ?
Is there an automatic reconnection to the wifi or not ?
Is there a way to have a cron that checks network connection and reconnect ? or reboot ?
Can you give me some clue ?
Thanks in advance

Scatman007, 2016-11-17:
since of couple of weeks I am experiencing the same issue.
I have to restart the camera to have it work again.

caseyfw, 2017-01-18:
I have the same problem. Oddly enough the camera will continue to record motion and still responds to changes in lighting by toggling the IR filter. I suspect it's simply dropped the wifi connection and refuses to reconnect.

One solution might be to add a cron that checks for a connection and attempts to re-establish one if it doesn't exist. I'll look into options.

stomicron, 2017-02-05:
I too was losing the feed on some of my cameras every few days. I can't say if it was this issue or #131, as I never bothered pinging (router still showed all cameras as connected FWIW), but @airdrummingfool's commit #108 seems to have done the trick for me.