fritz-smh/yi-hack #86: Is there a way to remove this hack once installed?

epinapala, 2016-09-18:
Hi,

I would like to know if it is possible to go back to official firmware once yi-hack is installed. If so would the renamed system file(as mentioned [here](https://github.com/fritz-smh/yi-hack#warning--changes-done-on-the-original-camera-filesystem) ) be reverted too ?

hermitcrabslab, 2016-09-19:
If you want the timeout sound back:
Edit equip_test.sh to look like this:
```
#!/bin/sh
######################################################
# Xiaomi Yi hack restore
######################################################

### Rename back the timeout sound file ...
[ -f /home/timeout.g726.OFF ] && mv /home/timeout.g726.OFF /home/timeout.g726

sync
```
Then reboot the camera so the script runs. After remove the stuff from sd card and you are all original.

epinapala, 2016-10-04:
Thanks for the reply. Also once hack is installed, does the camera continue to work with mi home apps?

Shuunen, 2016-10-04:
On my side, the My Yi Home Camera android app does not access the cam anymore.

fritz-smh, 2016-10-04:
@epinapala @Shuunen true, once installed, you can't use the official app anymore because the app use the cloud feature which is deactivated by the hack.

@hermitcrabslab thanks for the restore proposal
I added it in https://github.com/fritz-smh/yi-hack/blob/master/README.md
Closing the ticket

closed 2016-10-04 by fritz-smh

epinapala, 2016-10-05:
Awesome thanks alot folks for yiur answers!!