fritz-smh/yi-hack #87: Possible to get rid of the time on top left ?

hohob28, 2016-09-19:
Hi, 
Is it possible to remove the time displaying on the top left corner? Thank you.

Btw, I love the hack so much. It would be great if you can share the tutorial of editing the firmware.

dale3h, 2016-11-30:
I would also like to know if there is a way to disable the time from displaying.

hermitcrabslab, 2016-12-17:
Curious thing, I got another camera(EU version) and this one has no time/date display(never used with yi app)

ygator, 2016-12-31:
There may be a setting in /etc/ui.conf

I did mess around with the fonts and when I made the 3 iso*.bin font files in /home into zero length files and rebooted I got no timestamp on the high quality and two locations with junk on the low quality.

ygator, 2017-01-08:
@hermitcrabslab I just got a US version.  The reason there is no date/time display is that the camera runs in UTC.  The app converts to local time.

hermitcrabslab, 2017-01-08:
I never used any app with this version(EU).
Placed the stuff on sd card and connected to my nvr on firstboot ;)
I plan to keep it that way :P

williamstlr, 2018-01-25:
I was able to remove my timestamps by following @ygator's suggestion. Here's how I did it

`mkdir -p /tmp/hd1/test/backup
cp -a /home/iso* /tmp/hd1/test/backup/    #Make a copy of the files
> iso8859-1-16x32.bin    #Make the files "zero-length"
> iso8859-1-4x8.bin
> iso8859-1-8x16.bin
reboot`

After that the timestamps were gone on my high-quality streams and there was some junk in place of timestamps on my low quality streams.