fritz-smh/yi-hack #88: Take a single Picture

magadan, 2016-09-24:
Hello, do you if is possible to create a script to take only one single picture (not movie) ?

dmolner, 2016-10-24:
I need these too

bilinguliar, 2016-12-02:
I used "avconv" as a workaround to get frame from RTSP stream and convert it to png. But ability to get full size photo directly from sensor would be great.

dmolner, 2016-12-02:
I use ispy for take single image, but if camera could take picture him self and later upload by ftp we have a totally autonomous camera

resident12, 2017-09-09:
Someone found a workable solution to acquire still images with the Xiaomi Ant by means of a local script? Those movies are cumbersome to review what happened at the scene in retrospect over a multi day period.