fritz-smh/yi-hack #89: How to change wpa supplicant?

cuibap1212, 2016-10-06:
Hello,

In file wpa_supplicant.conf:

> ctrl_interface=/var/run/wpa_supplicant
> ap_scan=1
> network={
> ssid="your_wifi_ssid"
> scan_ssid=1
> proto=WPA RSN
> key_mgmt=WPA-PSK
> pairwise=CCMP TKIP
> group=CCMP TKIP
> psk="your_password"
> }

I want to change: WPA-SPK to WPA2-SPK and TKIP to AES 
How to change it?

peytonjol, 2016-10-12:
Try that :

proto=WPA2
key_mgmt=WPA2-PSK
pairwise=AES
group=CCMP

cuibap1212, 2016-10-13:
I have research:
**proto:**
- "RSN" = WPA(2)
- "WPA" = WPA(1)
**key_mgmt:**
- "RSN" = WPA(2)
- "WPA" = WPA(1)
**pairwise & group:**
- "CCMP" = AES cipher as part of WPA(2) standard.
- "TKIP" = TKIP cipher as part of WPA(1) standard.

peytonjol, 2016-10-13:
So is that working now ? I made some change on mine yesterday and i guess it works. Will put the config file if you want it.

tampiss, 2016-10-22:
Hi guys, do you have it please wotking with WPA2-PSK? Could you provide the config? Thank you, tampiss

mookishuki, 2016-10-24:
I'd like to join tampiss request.

I worked with the YI firmware version M for over 6 months. For some reason it stopped working (connecting to wifi). Could not get it to connect to wifi on original version with the app. I stumbled upon this project and still no luck (LED turning blue for short period and then blinking orange). I have WPA2 AES protected LAN.

I tried:
1. editing wpa_supplicant.conf with ssid and psk and leaving the rest default
2. changing proto key_mgmt , pairwise and group to RSN and CCMP as described earlier
3. in yi-hack.cfg I have to change the ip of the gateway & nameserver to 10.100.102.1 (router)
4. my router does not allow me to assign static ip's, so I also tried another thing,
 writing ' udhcpc -b -i ra0 ' in equip_test.sh instead
ifconfig ra0 $(get_config IP) netmask $(get_config NETMASK)
route add default gw $(get_config GATEWAY)

what am I doing wrong?
thank you

mustekito, 2016-10-25:
I have a similar problem. The camera connects to wifi, and telnet, rtsp, etc, works fine. The "LED" turning blue for short period and then it blinks orange. **BUT the videos are not saving**, the motion detection does not work...

The log does not show any problem.
```
Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Oct 25  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Oct 25  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Oct 25  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Oct 25  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Oct 25  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = yes
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=0  (0 is ok)
Wifi configuration answer: 
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:26:17:FA  
              inet addr:192.168.0.99  Bcast:192.168.0.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe26:17fa/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:496 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:130101 (127.0 KiB)  TX bytes:300 (300.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.europe.pool.ntp.org':
Previous datetime is Thu Jan  1 01:00:50 CET 1970
Done
New datetime is Thu Jan  1 01:01:20 CET 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1784 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1811 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1918 root      2660 S <  ./rtspsvrM
Processes after startup :
  PID USER       VSZ STAT COMMAND
[... 53 lines cut ...]
  987 root      1164 S    ./peripheral
  988 root      1144 S <  ./dispatch
  989 root      1068 S    ./exnet
 1122 root      3136 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1784 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1811 root      1408 S    ./serverM 80
 1829 root      3360 S    ./record_event
 1830 root     14728 S    ./mp4record 60
 1859 root      1716 S    /usr/sbin/crond -b
 1908 root       904 S    /home/led_ctl -boff -yfast
 1909 root     94448 S    ./rmm
 1918 root     27252 S N  ./rtspsvrM
 2014 root      1708 R    ps
Storage status :
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.5M      2.7M    868.0K  76% /
tmpfs                    17.2M     16.0K     17.2M   0% /dev
/dev/mtdblock5            8.9M      8.0M    916.0K  90% /home
tmpfs                    32.0M     44.0K     32.0M   0% /tmp
/dev/hd1                 29.3G      8.7M     29.3G   0% /tmp/hd1
tmpfs                   512.0K     16.0K    496.0K   3% /home/mmap_tmpfs
tmpfs                    16.0M         0     16.0M   0% /home/tmpfs
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                 29.3G      8.7M     29.3G   0% /tmp/hd1/test/http/record
```

antidata, 2016-11-08:
For WPA2 use:
```

proto=WPA2
key_mgmt=WPA-PSK
pairwise=CCMP
group=CCMP
```

peytonjol, 2016-11-10:
I used that for more than 1 month with WPA2 AES
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP

mookishuki, 2016-11-20:
Still no luck :( . This is my log (at least the relevant part):
`Starting to log...`
`Start telnet server...`
`MTK 7601`
`Hardware version informations : `
`    hwplatform=2`
`Software version informations : `
`    [sys]`
`    version=1.8.5.1M_201512011815`
`    [kernel]`
`    version=3-0.8`
`    provider=18e`
`    devtype=18e`
`Firmware letter is : 'M'`
`The RTSP server binary version which will be used is the 'M'`
`The HTTP server binary version which will be used is the 'M'`
`Check for some files size...`
`    -rwxr-xr-x    1 root     root        144885 Sep  9  2016 /home/hd1/test/http/serverJ`
`    -rwxr-xr-x    1 root     root        144912 Sep  9  2016 /home/hd1/test/http/serverM`
`    -rwxr-xr-x    1 root     root        121932 Sep  9  2016 /home/hd1/test/rtspsvrI`
`    -rwxr-xr-x    1 root     root         39572 Sep  9  2016 /home/hd1/test/rtspsvrK`
`    -rwxr-xr-x    1 root     root         39580 Sep  9  2016 /home/hd1/test/rtspsvrM`
`The blue led is currently blinking`
`Debug mode = no`
`Check for wifi configuration file...*`
`/home/wpa_supplicant.conf`
`Start wifi configuration...`
`Status for wifi configuration=255  (0 is ok)`
`Wifi configuration answer: Successfully initialized wpa_supplicant`
`rfkill: Cannot open RFKILL control device`
`Could not set interface ra0 flags (UP): Operation not permitted`
`WEXT: Could not set interface 'ra0' UP`
`ra0: Failed to initialize driver interface`
`Do network configuration 1/2 (ip and gateway)`
`Done`
`Configuration is :`
`    lo        Link encap:Local Loopback  `
`              inet addr:127.0.0.1  Mask:255.0.0.0`
`              inet6 addr: ::1/128 Scope:Host`
`              UP LOOPBACK RUNNING  MTU:16436  Metric:1`
`              RX packets:0 errors:0 dropped:0 overruns:0 frame:0`
`              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0`
`              collisions:0 txqueuelen:0 `
`              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)`
`    `
`Do network configuration 2/2 (DNS)`
`Done`

This is my wpa_supplicant.conf:
`ctrl_interface=/var/run/wpa_supplicant`
`ap_scan=1`
`network={`
`ssid="THEWIRELESSNAME"`
`scan_ssid=1`
`proto=WPA RSN`
`key_mgmt=WPA-PSK`
`pairwise=CCMP TKIP`
`group=CCMP TKIP`
`psk="THEPASSWORD"`
`}`
and this is my yi-hack.cfg:
`# Don't forget to also fill the file wpa_supplicant.conf for the wifi configuration`
`IP=10.100.102.5`
`NETMASK=255.255.255.0`
`GATEWAY=10.100.102.1`
`NAMESERVER=10.100.102.1`

Anyone has ideas why ra0 cannot be set-up? Am I missing something?

mookishuki, 2016-11-27:
OK, I have more info. I opened up the camera and connected to the serial port.
Lots of info appears during boot but I'll post the things that I think are more important:
1. Do I have some SD card issues?
```
cp: can't create '/etc/Wireless/RT2870STA/RT2870STA.dat': No space left on device
rm: can't remove '/home/mmap_tmpfs/mmap.info': No such file or directory
/home/hd1/test/equip_test.sh: line 90: can't create /etc/TZ: No space left on device
mv: can't create '/usr/share/udhcpc/default.script': No space left on device
rm: can't remove '/etc/resolv.conf': No such file or directory
ln: /etc/resolv.conf: No space left on device
tmpfs on /tmp type tmpfs (rw,relatime,size=32768k)
/dev/hd1 on /tmp/hd1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=cp437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
```

2. Did my wireless module die on me?
```
RtmpOSFileOpen(): Error 2 opening /etc/Wireless/RT2870STA/RT2870STA.dat
Open file "/etc/Wireless/RT2870STA/RT2870STA.dat" failed!
1. Phy Mode = 6
ERROR!!! RTMPReadParametersHook failed, Status[=0x00000001]
Bulk In Failed. Status=-2, BIIdx=0x0, BIRIdx=0x0, actual_length= 0x0
unlink cmd rsp urb
!!! rt28xx init fail !!!
```

3. trying to see manually if ra0 is up and trying to bring it up:
```
# ip link show ra0
ra0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
# ip link set ra0 up
RtmpOSFileOpen(): Error 2 opening /etc/Wireless/RT2870STA/RT2870STA.dat
Open file "/etc/Wireless/RT2870STA/RT2870STA.dat" failed!
1. Phy Mode = 6
ERROR!!! RTMPReadParametersHook failed, Status[=0x00000001]
Bulk In Failed. Status=-2, BIIdx=0x0, BIRIdx=0x0, actual_length= 0x0
unlink cmd rsp urb
CMDTHREAD_RESET_BULK_IN: Cannot do bulk in because flags(0x1080042) on !
!!! rt28xx init fail !!!
ip: SIOCSIFFLAGS: Operation not permitted
```

camera version is 22CN.
Any ideas?

Bugi824, 2022-10-06:
> Still no luck :( . This is my log (at least the relevant part): `Starting to log...` `Start telnet server...` `MTK 7601` `Hardware version informations :` `hwplatform=2` `Software version [...]

It is too late, but if anyone have the same problem, keep your WiFi SSID and password in " " :-D