fritz-smh/yi-hack #93: How to set user/pass for rtsp ?

sagilo, 2016-11-01:
I don't like Xiaomi holding my videos but leaving the rtsp out open is not a good solution too.
Of course I can close the port for external access but then I can't see the camera online.
Is there a way for settings user/pass on the rtsp server ? 
Cheers.

raspserver, 2016-11-10:
I also have this requirement to have user Id & pw on rtsp server. This hack version currently mentioned cannot be implemented. Just wish to understand more whether it is technically not possible or a feature to be implement in future.
Is there any major security risk impact if there is no user id & password protection on the video stream ? Assuming someone have to hack into the wifi network first which is pw protected before can connect to the open rtsp server ? Anyone can advise does that make sense there is already some level of protection to the video stream ?

raspserver, 2016-12-04:
Anyone able to share any idea to set user id & password for RTSP video streaming ?

sagilo, 2016-12-04:
It is probably not possible.
I enabled VPN in my local network and canceled the rtsp port expose to the outside.

raspserver, 2016-12-04:
May I understand why ? Is it RSTP natively does not support or there is a lot of effort to re-code the hack ?

sagilo, 2016-12-04:
I think the hack only expose an available rtsp feature which already exists in the camera code. (this is why it is not available in newer versions), but it's only my opinion by observing the hack code.

raspserver, 2016-12-04:
I hope someone able to include RSTP user & pw authentication to be more complete hack some day....

ishalem, 2017-01-21:
Hi
is there any update on if and when it will be available? security is very important in order to use the camera from external network
also, did anyone find a workaround?