fritz-smh/yi-hack #94: Starting problem

bengyskhan, 2016-11-03:
hi !
I plug my Yi with ur hack, it s starting: Yellow blink -> blue blink -> blue led on and after ... yellow blink all time ...
thx for ur help.

this is its log : 
`Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Oct  4  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Oct  4  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Oct  4  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Oct  4  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Oct  4  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=0  (0 is ok)
Wifi configuration answer: 
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:52:90:F7  
              inet addr:192.168.0.121  Bcast:192.168.0.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe52:90f7/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:0 (0.0 B)  TX bytes:300 (300.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.uk.pool.ntp.org':
Previous datetime is Thu Jan  1 00:00:50 GMT 1970
Done
New datetime is Thu Jan  1 00:01:20 GMT 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1765 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1803 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1895 root     27252 R <  ./rtspsvrM
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  620 root         0 SWN  [jffs2_gcd_mtd4]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  680 root      1704 S    telnetd
  686 root       984 S    /home/log_server
  816 root         0 SW   [RtmpTimerTask]
  817 root         0 SW   [RtmpMlmeTask]
  818 root         0 SW   [RtmpCmdQTask]
  819 root         0 SW   [RtmpWscTask]
  949 root         0 SW   [hidog]
  987 root      1164 S    ./peripheral
  988 root      1144 S <  ./dispatch
  989 root      1060 S    ./exnet
 1114 root      3128 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1765 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1803 root      1408 S    ./serverM 80
 1821 root      3360 S    ./record_event
 1822 root     14728 S    ./mp4record 60
 1840 root      1716 S    /usr/sbin/crond -b
 1889 root       904 S    /home/led_ctl -boff -yfast
 1890 root     94444 S    ./rmm
 1895 root     27252 S N  ./rtspsvrM
 1995 root      1708 R    ps
Storage status :
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.5M      2.7M    844.0K  76% /
tmpfs                    17.2M     16.0K     17.2M   0% /dev
/dev/mtdblock5            8.9M      8.0M    916.0K  90% /home
tmpfs                    32.0M     44.0K     32.0M   0% /tmp
/dev/hd1                 62.5G      9.4M     62.5G   0% /tmp/hd1
tmpfs                   512.0K     16.0K    496.0K   3% /home/mmap_tmpfs
tmpfs                    16.0M         0     16.0M   0% /home/tmpfs
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                 62.5G      9.4M     62.5G   0% /tmp/hd1/test/http/record
`

raspserver, 2016-11-05:
Hi 
bengyskhan ,

I have the same problem as you are facing that the camera starting: Yellow blink -> blue blink -> blue led on and after few seconds yellow blink all time.

Have you manage to solve this ?

Syone, 2016-11-05:
Hi, I've got the same problem too...

Syone, 2016-11-05:
it's maybe your ip configuration, I solve this by editing the IP, GATEWAY and NAMESERVER on yi-hack.cfg

raspserver, 2016-11-05:
I'm still unable to solve it. The camera still show yellow color. My yi-hack.cfg is:-
IP=192.168.0.121
NETMASK=255.255.255.0
GATEWAY=192.168.0.254
NAMESERVER=192.168.0.254

Do you get sold blue color in your camera ?

joozy, 2016-11-05:
bengyskhan, your log shows that you were not able to reach the ntp server, previous datetime and new datetime showing 1970.

Maybe - try using uk.pool.ntp.org.

I was partly successful with this hack, in that it failed to boot after two days of use. It's odd but I stopped trying to figure out why and reverted to a simpler hack...does not provide English voice instructions/feedback and manually telnet to correct time.

Syone, 2016-11-06:
@raspserver do you a router? if so, you need to change IP and GATEWAY configuration

kevinm78, 2016-11-08:
i have the same problem, Yellow blink -> blue blink -> blue led on and after ... yellow blink all time ...

i'm using a 12CN version, anyone can help me ? 

my log : 
Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Oct  4  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Oct  4  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Oct  4  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Oct  4  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Oct  4  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = yes
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=0  (0 is ok)
Wifi configuration answer: 
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:63:E0:81  
              inet addr:192.168.1.125  Bcast:192.168.1.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe63:e081/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:240 errors:0 dropped:0 overruns:0 frame:0
              TX packets:1 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:54215 (52.9 KiB)  TX bytes:200 (200.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.uk.pool.ntp.org':
Previous datetime is Thu Jan  1 01:00:37 GMT 1970
Done
New datetime is Thu Jan  1 01:00:38 GMT 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1204 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1242 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1335 root      2660 S <  ./rtspsvrM
Processes after startup :
[... 55 lines cut ...]
  988 root      1144 S <  ./dispatch
  989 root      1060 S    ./exnet
 1125 root      3128 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1204 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1242 root      1408 S    ./serverM 80
 1260 root      3360 S    ./record_event
 1261 root     14728 S    ./mp4record 60
 1279 root      1716 S    /usr/sbin/crond -b
 1328 root       904 S    /home/led_ctl -boff -yfast
 1329 root     94456 S    ./rmm
 1335 root     27252 S N  ./rtspsvrM
 1445 root      1708 R    ps
Storage status :
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.5M      2.6M    872.0K  76% /
tmpfs                    17.2M     16.0K     17.2M   0% /dev
/dev/mtdblock5            8.9M      8.0M    908.0K  90% /home
tmpfs                    32.0M     44.0K     32.0M   0% /tmp
/dev/hd1                  1.9G      8.3M      1.9G   0% /tmp/hd1
tmpfs                   512.0K     16.0K    496.0K   3% /home/mmap_tmpfs
tmpfs                    16.0M         0     16.0M   0% /home/tmpfs
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                  1.9G      8.3M      1.9G   0% /tmp/hd1/test/http/record

thanks a lot for your help

kevinm78, 2016-11-10:
using the custom firware from http://xiaoyi.querex.be/
i succeed to execute all the https://diy.2pmc.net/solved-xiaomi-xiao-yi-ant-home-camera-can-used-china/
using the Telnet connection. but that mean i can't use your firmware ? i replace the test folder by your it restart :(

bengyskhan, 2016-11-14:
hi all
thank for ur answers.
I tried with a new IP configuration and tried an other NTP server but nothing ... always this f$%#*!% yellow blinked light

joozy, 2016-11-14:
bengyskhan,

would you be keen to try a different firmware that could get your camera
going again?

bengyskhan, 2016-11-14:
I tried with Q firmware but nothing ...

joozy, 2016-11-14:
You have the older Chinese camera?

bengyskhan, 2016-11-14:
yes I have older this camera on Aliexpress.
but the Q firmware had install before the update

bengyskhan, 2016-11-14:
which is the best firmware to success this hack?

joozy, 2016-11-14:
Download from here, saved the file/folder to your sdcard - https://www.dropbox.com/sh/wfjl6xs1ca6c42m/AADyIp-drbmQTTle7A-4FB9Sa?dl=0

joozy, 2016-11-14:
need to telnet to camera to change timezone.
need to muck around to get the correct offset.
will still have chinese instruction audio.
playback timeline is wonky...not working right.

joozy, 2016-11-14:
feedback after testing, if you decide proceed.

bengyskhan, 2016-11-14:
where to find timezone parameter? 
and others parameters like Wifi SID and IP ?

bengyskhan, 2016-11-14:
and it was impossible to load your Q version.
the yellow light always blinked : S
I rollbacked this upgrate ... My Yi cam is now in M firmware

bengyskhan, 2016-11-14:
I used Kevinm78 s solution.
It s well but there is no http live streaming  :( only RTSP

bengyskhan, 2016-11-14:
IT S WORK!!! :)
So I download the good starting Firmware: 1.8.5.1M_201512011815 here (http://xiaoyi.querex.be/firmwares/)
I installed it with the SD card like this instruction http://xiaoyi.querex.be/instructions/
and turn on the cam until blue light.

After turn off the cam
and use the Fritz s hack https://github.com/fritz-smh/yi-hack/archive/master.zip like its instruction.
and turn on the cam and wonderfull! it s work.

kevinm78, 2016-11-15:
thanks bengyskhan  i'll try too. keep you informed :)

kevinm78, 2016-11-17:
Hi bengyskhan  ,

did you use the Home file from the hack ? or the one from querex ? i try what you said but it fail for me 
I installed it with the SD card like this instruction http://xiaoyi.querex.be/instructions/
 and turn on the cam until blue light.
=> i don't have the blue light the cam talk chinese saying something about wifi :( 

and the other step 
After turn off the cam
 and use the Fritz s hack https://github.com/fritz-smh/yi-hack/archive/master.zip following its instruction.
=> after the blue bliinking => back to yellow blink :( 

i'll try with another firmeware maybe we have different cam ? i have a 12CN

kevinm78, 2016-11-18:
back for news : it's ok for me until i try the hack .. i'm now on the M version. started ok with Yi home app. then when i put the test folder on SD card managing all parameter ok => back to yellow blink :(

kevinm78, 2016-11-18:
my new log is : 

Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Oct  4  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Oct  4  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Jan 21  2015 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Dec  1  2015 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Dec  8  2015 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = yes
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=0  (0 is ok)
Wifi configuration answer: 
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr B0:D5:9D:63:E0:81  
              inet addr:192.168.1.32  Bcast:192.168.1.255  Mask:255.255.255.0
              inet6 addr: fe80::b2d5:9dff:fe63:e081/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:9 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:2448 (2.3 KiB)  TX bytes:200 (200.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.uk.pool.ntp.org':
Previous datetime is Fri Nov 18 21:40:20 GMT 2016
Done
New datetime is Fri Nov 18 21:40:50 GMT 2016
Start blue led on
Start ftp server...
Check for ftp process : 
 1771 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1798 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1903 root      2660 S <  ./rtspsvrM
Processes after startup :
  PID USER       VSZ STAT COMMAND
    1 root      1708 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW   [rcu_kthread]
    7 root         0 SW<  [khelper]
    8 root         0 SW   [kworker/u:1]
  143 root         0 SW   [sync_supers]
  145 root         0 SW   [bdi-default]
  146 root         0 SW<  [kintegrityd]
  148 root         0 SW<  [kblockd]
  161 root         0 SW   [khubd]
  174 root         0 SW<  [cfg80211]
  175 root         0 SW   [kworker/0:1]
  257 root         0 SW<  [rpciod]
  260 root         0 SW   [kswapd0]
  314 root         0 SW   [fsnotify_mark]
  322 root         0 SW<  [nfsiod]
  331 root         0 SW<  [crypto]
  405 root         0 SW   [mtdblock0]
  410 root         0 SW   [mtdblock1]
  415 root         0 SW   [mtdblock2]
  420 root         0 SW   [mtdblock3]
  425 root         0 SW   [mtdblock4]
  430 root         0 SW   [mtdblock5]
  435 root         0 SW   [mtdblock6]
  438 root         0 SW   [romblock0]
  441 root         0 SW   [romblock1]
  444 root         0 SW   [romblock2]
  447 root         0 SW   [romblock3]
  450 root         0 SW   [romblock4]
  453 root         0 SW   [romblock5]
  456 root         0 SW   [romblock6]
  591 root         0 SW<  [kpsmoused]
  600 root         0 SW   [kworker/u:2]
  616 root         0 SW   [mmcqd/0]
  620 root         0 SWN  [jffs2_gcd_mtd4]
  624 root      1712 S    /bin/sh /etc/init.d/rcS
  636 root       872 S <  udevd --daemon
  642 root      1704 S    /bin/sh /etc/init.d/S80network
  656 root         0 SWN  [jffs2_gcd_mtd5]
  657 root      1708 S    /bin/sh ./init.sh
  664 root         0 SW   [flush-mtd-unmap]
  675 root      1712 S    /bin/sh /home/hd1/test/equip_test.sh
  677 root         0 SW   [flush-179:0]
  680 root      1704 S    telnetd
  686 root       984 S    /home/log_server
  816 root         0 SW   [RtmpTimerTask]
  817 root         0 SW   [RtmpMlmeTask]
  818 root         0 SW   [RtmpCmdQTask]
  819 root         0 SW   [RtmpWscTask]
  949 root         0 SW   [hidog]
  987 root      1164 S    ./peripheral
  988 root      1144 S <  ./dispatch
  989 root      1060 S    ./exnet
 1120 root      3128 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1771 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1798 root      1408 S    ./serverM 80
 1816 root      3392 S    ./record_event
 1817 root     22988 S    ./mp4record 60
 1846 root      1716 S    /usr/sbin/crond -b
 1895 root       904 S    /home/led_ctl -boff -yfast
 1896 root     94440 S    ./rmm
 1903 root     27252 S N  ./rtspsvrM
 2006 root      1708 R    ps
Storage status :
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.5M      2.6M    884.0K  75% /
tmpfs                    17.2M     16.0K     17.2M   0% /dev
/dev/mtdblock5            8.9M      8.1M    760.0K  92% /home
tmpfs                    32.0M     44.0K     32.0M   0% /tmp
/dev/hd1                  1.9G     37.0M      1.9G   2% /tmp/hd1
tmpfs                   512.0K     16.0K    496.0K   3% /home/mmap_tmpfs
tmpfs                    16.0M      4.0K     16.0M   0% /home/tmpfs
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                  1.9G     37.0M      1.9G   2% /tmp/hd1/test/http/record

kevinm78, 2016-11-18:
I DID it :) it work, was a problem of Gateway :s too many time loose for this ... thanks a lot for all guys :)

Jab413, 2016-12-04:
What do you mean gateway problem? Did you use the one from ipconfig in windows? I am also having the same issue. No matter what I do

VaporKing, 2016-12-23:
Hi

After spending hours on this GitHub Project and other pages I managed to hack two Yi Antscam 720p with CN serial number. As I can tell, a lot of other people are having also problems. So I wanted to share my experiance on how I managed to get the camera running without the Yi Home App and with RTSP access:

First of all: I have three of these little cameras. The first one i bought was also a CN version, but I was able to hack the camera by simply following thes steps in this Project and using the hack from fritz. So I ordered an other two cameras from an other seller on AliExpress. I thought well, if it worket with the first camera, it will work with the new cameras too. It turned out, both the new camears are also CN versions for arount 30 Dollar. But when I tryied to hack the new cameras with the steps and firmware from this project, I experianced the following:

After resetting the camera and flashing the hacked firmware via the SD card, the camera LED was first blinking orange, then steady orange, then blue blinking and then steady blue. After a few seconds the blue steady LED started blinking again. So the camera disconnected from my WiFi after a few seconds. No mather what I tryied with the two new cameras and the hack from this project, it would not work. 

So I started searching the Net for other solutions and I found a workarround for my two cameras. But a different one for each of the two. First I thought, wow, I managed to hack the first one, so the same hack will work with the second one. Well, it didn't. So here are my two ways that worked for me:

Camera one with CN Firmware:

-Download the firmware for CN version / 1.8.5.1M_201512011815 from http://xiaoyi.querex.be/firmwares/

-Download the hacked russian version from here:
https://www.dropbox.com/sh/4s2r8l5ajqim64o/AAA83mRz6Vgcb9cUj92rtw53a?dl=0

-put the firmware file 1.8.5.1M (not the hacked one! The firstone you downloaded) on a new formated sd card
--------------------------------------------
-download the yi home app from app store
-in the app, choose Europe as your location
-create an account
-ad a new camera 
-be sure to be logged in to the SSID you wan to connect the camera to
-put in the WiFi password for your SSID
-go one step furhter to create the QR code
-------------------------------------------------
-Unplug the camera
-Reconnect the power
-Reset the camera via the reset button on the back
-Unplug the power
-insert the SD card with 1.8.5.1M firmware
-reconnect the power
-wait untill properly flashed
-Unplug the camera
-Take out SD card
--------------------------------------
-on your computer, format the SD card again
-put the the two files from the second download from Dropbox on to the SD Card
-insert SD card back in to the camera

-NOW THIS IS IMPORTANT!!!: Go tho the place where you will deploy the camera! The Hack only works once, witch means, with the first powering on after inserting the SD Card with the hacked firmware. If you unplug the camera after flashing the hacked firmware, and then power it again, the camera will ask you again for the QR code and promt you with Invalid QR code or will disconnect from the WiFI after 40 minutes with the prompt Connection Time-Out or somethin like that. So insert the SD card with the hacked firmware and go to the place where you will deploy the camera. Take your Cellphone with you with the QR Code, then:

-reconnect the power
-wait untill some strange music is comming from the camera
-After the music, the camera will talk you in english and chinese
-When the camera is telling you, ready to connect, face the QR code from the Yi Home App to the camera. It will tell you, QR code successfully scanned.
-After this it shoult tell you, successfully connected or something like that.
-You should now be able to see the camera and its IP  in your network with a network scanner app like FING. You will not be able to connect to the camera via the Yi Home APP. The only use for the App is to give the WiFi credentials to the camera via QR Code.
-You should be able to connect to the camera via IP in your browser and via RTSP in your local network
----------------------------------------
The above steps worked for me with one of the cameras. With the second one, exactly the same steps,  I had the problem, that even after the first boot after flashing the hacked firmware, it was telling me invalid QR Code or it disconnected after about 40 minutes telling me something about a Time-Out. So with the second camera, I did nearly the same thing, exept I tried different original older firmwares before flashing the hacked version:

So if you have the same problem, repeat the steps from above from step one! Everytime it is not working, reset the camera, flash different old firmware, and then flash the hacked version..... Go on, until it works. For me, with the second camera firmware version 1.8.5.1B_201513211614 worked fine.

Hopefully this works for some other people and I could save you some evenings trying to connect your cameras ;-)

And remember! With the hacked firmware, be sure to be at the right place and power it on only once. Now that it's working for me, I have not tried what happens, when I disconnect the power again after a few days. I'm only glad it is working now. I will see what happens at the next power loss in our town ;-)

Let me know, if it worked for you...

Joeribl, 2016-12-30:
I have the same problem with a CN12. After a long struggle i decided to connect the camera to a TTL Serial adapter. Then at least i can see what happens.

Unfortunately i still have no idea what is going on. If i perform a manual DHCP request this is Acknowledged by the router. The ip is set and resolv.conf is updated. Routing table looks ok.

However i cannot ping anything and so it seems there is some sort of network problem.

If anybody has some idea what to check, let me know.

vosmont, 2017-01-16:
Hello,

I had this problem too. I've added a sleep command, to let the system really get its IP (in the logs there was just an IP v6).
Edit  /home/hd1/test/equip_test.sh 

`
log "Do network configuration 1/2 (ip and gateway)"
##ifconfig ra0 192.168.1.121 netmask 255.255.255.0
##route add default gw 192.168.1.254
#ifconfig ra0 $(get_config IP) netmask $(get_config NETMASK)
#route add default gw $(get_config GATEWAY)
#log "Done"
sleep 10
log "Do network configuration (DHCP)"
udhcpc -b -i ra0 >> ${LOG_FILE}
log "Done"
`

Joeribl, 2017-01-17:
Jeez, that's it. I can ping the gateway now and reach the HTTP server. Excellent. So maybe this fix should be applied in git....