fritz-smh/yi-hack #95: Mijia 360 support?

Shadowface, 2016-11-09:
Probably not the right place to ask, but will you add Mijia 360 support?

There's another project ongoing that adds basic 360 mods, but it has been abandoned.
Especially RTSP support, as I want to use it to monitor my store in realtime.

Thanks. =)

TommyChausson, 2016-12-14:
Take a look at it : https://github.com/niclet/xiaomi_hack

Shadowface, 2016-12-14:
I'm aware of that project, and he has unfortunately abandoned it. (For now at least)

Shame really, because the camera sensor on the 360 is excellent, but without RTSP and continuous recording, it's not very useful for my needs.. :S

crirus, 2017-01-31:
Hello

I just received my Mijia 360. Is there any chance to have a look into hacking that camera too?

Thank you

MichaelSchemmer, 2017-02-27:
+1 on hacking the camera! A working RTSP would be amazing! I did use some of the config of the Yi-Hack Project and got RTSP working, but it would only stream for a while, die and then restart... It just ran RTSP in a continuous loop...

ess87, 2017-06-05:
Hello,
Any news for run RTSP on this camera?
Without that the camera is not very usefull...

jyxjjj, 2019-12-06:
Does this says like https://www.mi.com/camera-360/specs ? If not i will open a new issue.