fritz-smh/yi-hack #97: Issue with wifi connexion

goodspeed1, 2016-11-21:
Hi guys,

I have an issue with my new camera. It's a 12CN version-
I complete wpa_supplicant.conf file and yi-hack.cfg but the camera is not connected to my local network...

First - light sequence : Red - blue blinking - blue - red blinking

>> wpa_supplicant.conf
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="MySSID"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="MyPass"
}

>> yi-hack.cfg
IP=192.168.20.220
NETMASK=255.255.255.0
GATEWAY=192.168.20.254
NAMESERVER=192.168.20.254

Finally -
Protocol : WPA2-PSK/TKIP
EAPOL : version 1

>>log.txt
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=255  (0 is ok)
Wifi configuration answer: Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
Could not read interface ra0 flags: No such device
WEXT: Could not set interface 'ra0' UP
ra0: Failed to initialize driver interface
Do network configuration 1/2 (ip and gateway)

What I missed ??
Thanks in advance !

robertbeal, 2016-11-22:
I'm suffering exactly the same issue. Mine has been working without problem for months. Today I unplugged it and later plugged it back in and got this problem.

My logs look like this:

```
Starting to log...
Hardware version informations : 
    hwplatform=2
Software version informations : 
    [sys]
    version=1.8.5.1M_201512011815
    [kernel]
    version=3-0.8
    provider=18e
    devtype=18e
Firmware letter is : 'M'
The RTSP server binary version which will be used is the 'M'
The HTTP server binary version which will be used is the 'M'
Check for some files size...
    -rwxr-xr-x    1 root     root        144885 Nov 22  2016 /home/hd1/test/http/serverJ
    -rwxr-xr-x    1 root     root        144912 Nov 22  2016 /home/hd1/test/http/serverM
    -rwxr-xr-x    1 root     root        121932 Nov 22  2016 /home/hd1/test/rtspsvrI
    -rwxr-xr-x    1 root     root         39572 Nov 22  2016 /home/hd1/test/rtspsvrK
    -rwxr-xr-x    1 root     root         39580 Nov 22  2016 /home/hd1/test/rtspsvrM
The blue led is currently blinking
Debug mode = no
Check for wifi configuration file...*
/home/wpa_supplicant.conf
Start wifi configuration...
Status for wifi configuration=0  (0 is ok)
Wifi configuration answer: 
Do network configuration 1/2 (ip and gateway)
Done
Configuration is :
    lo        Link encap:Local Loopback  
              inet addr:127.0.0.1  Mask:255.0.0.0
              inet6 addr: ::1/128 Scope:Host
              UP LOOPBACK RUNNING  MTU:16436  Metric:1
              RX packets:0 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0 
              RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
    
    ra0       Link encap:Ethernet  HWaddr 00:6C:FD:16:B8:B4  
              inet addr:192.168.1.199  Bcast:192.168.1.255  Mask:255.255.255.0
              inet6 addr: fe80::26c:fdff:fe16:b8b4/64 Scope:Link
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:167 errors:0 dropped:0 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000 
              RX bytes:50457 (49.2 KiB)  TX bytes:400 (400.0 B)
    
Do network configuration 2/2 (DNS)
Done
Get time from a NTP server...
But first, test the NTP server '0.uk.pool.ntp.org':
Previous datetime is Thu Jan  1 00:00:50 GMT 1970
Done
New datetime is Thu Jan  1 00:01:20 GMT 1970
Start blue led on
Start ftp server...
Check for ftp process : 
 1773 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
Start http server : serverM...
Check for http server process : 
 1811 root      1408 S    ./serverM 80
Start rtsp server : rtspsvrM...
Check for rtsp process : 
 1923 root      2660 S <  ./rtspsvrM
Processes after startup :
[... 54 lines cut ...]
  987 root      1164 S    ./peripheral
  988 root      1144 S <  ./dispatch
  989 root      1068 S    ./exnet
 1122 root      3128 S    /home/wpa_supplicant -B -i ra0 -c /home/wpa_supplica
 1773 root      1716 S    tcpsvd -vE 0.0.0.0 21 ftpd -w /
 1811 root      1408 S    ./serverM 80
 1829 root      3360 S    ./record_event
 1830 root     14728 S    ./mp4record 60
 1859 root      1716 S    /usr/sbin/crond -b
 1913 root       904 S    /home/led_ctl -boff -yfast
 1914 root     94440 S    ./rmm
 1923 root     27252 S N  ./rtspsvrM
 2014 root      1708 R    ps
Storage status :
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.5M      2.7M    840.0K  77% /
tmpfs                    17.2M     16.0K     17.2M   0% /dev
/dev/mtdblock5            8.9M      8.1M    748.0K  92% /home
tmpfs                    32.0M     44.0K     32.0M   0% /tmp
/dev/hd1                 14.8G      8.5M     14.8G   0% /tmp/hd1
tmpfs                   512.0K     16.0K    496.0K   3% /home/mmap_tmpfs
tmpfs                    16.0M         0     16.0M   0% /home/tmpfs
tmpfs                    16.0M      3.0M     13.0M  19% /home/jrview
/dev/hd1                 14.8G      8.5M     14.8G   0% /tmp/hd1/test/http/record
```

goodspeed1, 2016-11-22:
I have : Status for wifi configuration=255 (0 is ok)
You have 0 as result, so this is no exactly the same issue

umasse, 2017-01-14:
I'm having the same issue:
Status for wifi configuration=255 (0 is ok)
Did you find a solution?
I really want to make my camera independent of Xiaomi and with RTSP enabled.

TommyChausson, 2017-01-14:
Can you try the hack with this fork : https://github.com/hermitcrabslab/yi-hack ?
It's a wifi chipset issue because we can read `Could not read interface ra0`
the fork will solve your problem.  

Right?

umasse, 2017-01-14:
@TommyChausson Thank you so much!!! That fork works great!!!!

juan22a91, 2017-06-07:
@TommyChausson I have this same issue and still cannot solve it. Any other suggestion?

wilb, 2019-05-04:
Same problem on a CN12 of mine - fork works great.