fritz-smh/yi-hack #100: Re-enable cloud

csu, 2016-11-28:
Is there a way to re-enable the official cloud functionality while keeping the HTTP/FTP/RTSP of the custom firmware? Which part of `equip_test.sh` prevents the cloud functionality from starting?

notdone, 2016-12-13:
Yes, i also just want the index page of HTTP server like this hack when i surf to http://ip cam. 'Cause fw of my cam 1861 cant downgrade to fw 1851. So how i mod the index page of HTTP server? Pls

death2all110, 2017-01-06:
@fritz-smh I would like to know this as well. Would it be possible to re-enable the cloud functionality, while keeping the HTTP/FTP portion of the custom firmware?

death2all110, 2017-01-09:
@csu @notdone I found this here: https://github.com/fritz-smh/yi-hack/issues/26 Looks like if you add back 

```./p2psever &
./cloud & ```

after `./mp4record 60 &` on [Line 363](https://github.com/fritz-smh/yi-hack/blob/master/sd/test/equip_test.sh#L363)

death2all110, 2017-01-14:
Update: Did this on my camera at home. Couple issues. 

1. I noticed my camera frequently disconnects from wifi and needs power cycled to bring back online. 

2. Alerts do not work through the Yi app with this. Not sure why.

csu, 2017-01-14:
Thanks for testing it out, @death2all110. Breaking the official app alerts is a deal breaker for me :(

death2all110, 2017-01-14:
@csu Same here. 2 of my cameras are back to stock, I have a 3rd here that I'm tinkering with to see if I can get alert functionality back with the custom fw

death2all110, 2017-01-14:
@CSU Great news! Got it figured out. Here is an excerpt of my equip_test.sh

```
...
# fix logo
cp -f /home/hd1/test/*.bmp /home/
mv /home/recv.726 /home/recv_X.726

if [ -f "/home/notfirst" ]; then         
if [ -f "/home/wpa_supplicant.conf" ]; then         
echo "wifi connectting"
/home/monitor_wifi &
cd /home
else                              
echo "1" > /tmp/waitdump
echo "1" > /tmp/announce_success  
/home/monitor_dump && rm /tmp/waitdump && /home/monitor_wifi &                 
cd /home
fi 
else
echo "first start"
fi 

# do rmm stuff now rather than later
cd /home
./rmm &

### Launch record event
cd /home
./record_event &
./mp4record 60 &
#./p2pserver &
./cloud &
/home/watch_process &

### Some configuration

himm 0x20050068 0x327c2c
#himm 0x20050068 0x0032562c
...
```

I moved the rmm script execute above the record_event and cloud stuff, and commented out the 
`/home/rmm "/home/hd1/test/voice/success.g726" 1` line below all that.

But I believe the key to it all is the monitor_wifi stuff that is above the rmm and cloud stuff. When checking logs over telnet it was failing with some cloud stuff until I added that back in. Also, the ./p2pserver can be commented out/removed as the cloud script or monitor_wifi seems to call it on their own.

I hope this all makes sense and isn't too confusing!

death2all110, 2017-01-16:
@csu, @notdone 

I decided to fork @fritz-smh work on this and added the cloud toggle into my own fork.

You can find the commit [here](https://github.com/death2all110/yi-hack/commit/89549759e5fb4167c95efd96d45ee66ddb1033cf)

And you can find my fork [here](https://github.com/death2all110/yi-hack/)