fritz-smh/yi-hack #110: I think HTTP can use cgi-bin

gaeco0, 2016-12-12:
# I found httpd(busybox) and tested that runs cgi-bin on other port(in my case 88).

# It runs by config file(standard busyboxhttpd's)

# if "Yi" can run CGI-Bin, maybe It could be customized(ex. 2way audio)

#Here is My httpd.conf
#==========================================
# H means home path
# A:* allow All
# I means Index file Name
# .ext:MIME TYPE
# and .cgi does not need MIMETYPE Configuration

H:/home/hd1/
A:*
I:index.html

.htm:text/html
.html:text/html
#==========================================
# sorry for my bad English

gaeco0, 2016-12-14:
I built the php for xiaoyi ants(base board : hi3518) and it is static binary(need no libraries)
I want to upload the binaries(arm-php, arm-php-cgi)
Could you join me this project?

ygator, 2016-12-16:
Nice find.