fritz-smh/yi-hack #111: Night Vision

PhilFjr, 2016-12-12:
Je poste en français ,si il faut le mettre en anglais  ,j'essayerai.
Le hack fonctionne bien ,mais : en vision nocturne je n'ai pas de flux vidéo ,j'ai juste une image fixe et l'heure bloquée au démarrage du flux .Quelqu'un a t'il eu se problème ?
Caméra avec numéro commençant par 12cn (chinoise)
VLC v.2.2.4
Flux : rtsp://192.168.1.19:554/ch0_0.h264
ou rtsp://192.168.1.120:554/ch0_1.h264

Merci et bravo pour le travail
Philippe

joozy, 2016-12-12:
Quelle est la version de votre firmware?

(Google Translate - What version is your firmware?)

PhilFjr, 2016-12-13:
Selon le hack le firmware serait le M.
Etant newbie , je ne sais pas comment trouver le firmware exact 

Retour commande 'uname -a'
Linux (none) 3.0.8 #1 Wed Apr 30 16:56:49 CST 2014 armv5tejl GNU/Linux

nl20900, 2016-12-19:
Bonjour, Je suis le même problème, par hasard vous avez résolu? Merci (google traducteur)

PhilFjr, 2016-12-20:
@nl20900 Pas encore, je vais mettre en stand-by jusqu'à la fin de l'année